lvgl/lvgl #73: Driver register param and update flag

Zaltora, 2018-01-14:
I write a driver for ssd1306/SH1106
I want to know if it can be a good idea to add a custom parameters for the users ?
To add for example a custom buffer that will be used by draw functions instead of hard-coded one like ST7565.
It is like add multi-screen support. What can be a good implementation otherwise ?
example of changes:
```
typedef struct _disp_drv_t {
    /*Write the internal buffer (VDB) to the display. 'lv_flush_ready()' has to be called when finished*/
    void (*disp_flush)(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p, void* parameters);

    /*Fill an area with a color on the display*/
    void (*disp_fill)(int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t color, void* parameters);

    /*Write pixel map (e.g. image) to the display*/
    void (*disp_map)(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p, void* parameters);

#if USE_LV_GPU
    /*Blend two memories using opacity (GPU only)*/
    void (*mem_blend)(lv_color_t * dest, const lv_color_t * src, uint32_t length, lv_opa_t opa, void* parameters);

    /*Fill a memory with a color (GPU only)*/
    void (*mem_fill)(lv_color_t * dest, uint32_t length, lv_color_t color, void* parameters);
#endif

   /* Custom driver  parameters */
   void* parameters;
} lv_disp_drv_t;
```
---
For small screen monochrome, i used a VDB smaller. Instead of byte, each bit represent a pixel. I disable VDB and advanced features from the library.
When drawing function are called (lv_task_handler), they update the buffer. (like ST7565).
In my loop, i got 2 functions:
  * lv_task_handler(); 
  * driver_update_screen();

I want to know if i need to update the screen, lvgl got a flag to know if the screen need be redraw ?
It is possible to add this flag  when` lv_task_handler()` is called ? 
to do this for example:
```
 if (lv_task_handler())
 { 
   driver_update_screen();
 }
```
Bonus: 
In the future, it possible to implement a VDB that use bit instead of byte for monochrome display with LV_COLOR_DEPTH to 1 ? (Difficult to implement i guess ?)

kisvegabor, 2018-01-15:
Hi,
the prototype of `disp_flush`, `disp_fill`etc.  is strict to ensure compatibility with other drivers. If you modify them lvgl also need to be modified which is reasonably not preferred. 

It's better to set a custom buffer for driver with a function like: `ssd1306_set_buf(my_buf)`. Unfortunately it's very difficult to support multi-displays this way. To support multiple displays you can do the following:
1. Let's say you have 2 displays, both with 128x64 resolution 
2. Set `LV_HOR_RES 256` and `LV_VER_RES 64`
3. In your `disp_flush` you can call an appropriate drive according to the area

Maybe I misunderstood something by why not call `driver_update_screen()` in `disp_flush/_fill/_map` ?

As surprisingly lot of people use LittlevGL with monochrome displays I will look after how to use "bit based" VDB.

Zaltora, 2018-01-15:
If we got a VDB buffer with same size of the screen, It is ok to update buffer in  `disp_flush()`. Because it will be called once. Other settings mean called it more than once. For this screen size, is not really the best. Use a VDB buffer 8 times larger either to save RAM. ( That why "bit based" VDB will be a great features :) )

For example, when you want draw a line from 0,0 to  0,64 . You need update 70% of the screen in the simple way  or call multiple time positioning cursor (Again additional transaction). Already tested with different libs, the performance difference is not huge and depend what you update.

If we don't use VDB, `disp_map/_fill ` will call many times `driver_update_screen()` and you will redraw an area already updated again and again. The performances will be very low.
With my test, i see that map/fill can be called between 10 and 20 times (more ?) when call `task_handler()`. When you use I2C protocols, it is like update 2 screen or more ^^.

In general, for small screens I2C/SPI, a VDB with same pixel size as screen is used for performance issue.

kisvegabor, 2018-01-15:
If you draw a (0;0) (0;64) line then it's just a 1 pixel heigh and 64 pixel wid. LittlevGL update only this small area. Or the 70% comes from a display specific usage?

Yes, it absolutely recommended to use VDB because of the communication overheads and slow pixel redrawing. As you might saw in ST7565 driver I used a "bit based" frame buffer: https://github.com/littlevgl/lv_drivers/blob/master/display/ST7565.c#L83 I copy pixels there from a small VDB to the frame buffer and update the screen from this small frame buffer.

Zaltora, 2018-01-15:
It is display specific usage, not LittlevGL problem. Look page 25,
[SSD1306 datasheet](https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf)
PAGE System. My example was bad because vertical line is easy to draw, horizontal line need draw a square area  ( 8*line_width), vertical line need call cursor multiple time.
It's easier to support the SH1106 when update full buffer too.
It's the same screen but with some subtlety (no scrolling, different SPI3 protocol,...)

kisvegabor, 2018-01-15:
I see. And what about the Small VDB and bit based frame buffer?

Zaltora, 2018-01-15:
Small VDB mean call multiple time buffer update. Bit based frame buffer solve the problem for one  utilization case.
Maybe better that redraw flag need to be implemented in the driver himself. with a function like this
`driver_need_redraw();`
`disp_flush/_fill/_map ` will set flag redraw.
Example:
```
lv_task_handler();
 if ( driver_need_redraw())
 { 
   driver_update_screen();
 }
```

kisvegabor, 2018-01-15:
Yes, but then seems better to create a redraw task: `lv_task_create(my_redaw, 50, LV_TASK_PRIO_MID, NULL)` . Then surly there won't be issues because of the concurrent access to the shared resource (the frame buffer). (The task handler is is not preemptiv)

Zaltora, 2018-01-15:
I don'k know yet the task API from lvgl.
The function will be called periodically. That mean, it will update the screen always or the handler task will not call my_redaw, when not needed ?

kisvegabor, 2018-01-15:
It will be called periodically but you can return at the first line like: `if(redraw_needed == 0) return;`

kisvegabor, 2018-02-05:
The discussion continues in `lv_drivers` repository

closed 2018-02-05 by kisvegabor