lvgl/lvgl #76: state of button in list after focusing

hotpeperoncino, 2018-01-16:
Hi. after focusing list, the state of first button of the list is not LV_BTN_STATE_PR.
The fragment will show the problem.

if the list has multiple buttons, we can use arrow key to move focus, but it does not work well.

# i use master branch

#include <stdio.h>
#include "lvgl/lvgl.h"
void onclick(){}

int app_init(lv_indev_t *keypad)
{
  lv_theme_t *th;
  th = lv_theme_mono_init(210, NULL);
  lv_theme_set_current(th);

  lv_obj_t *scr = lv_obj_create(NULL, NULL);
  lv_obj_t *list = lv_list_create(scr, NULL);
  lv_obj_set_size(list, 200, 200);
  lv_obj_align(list, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);

  lv_obj_t *btn = lv_list_add(list, SYMBOL_FILE, "Back", onclick);
  lv_list_add(list, SYMBOL_FILE, "Test", onclick);

  lv_obj_t *group = lv_group_create();
  lv_indev_set_group(keypad, group);

  printf("%d\n", lv_btn_get_state(btn));
                                                                                                                                    

  lv_scr_load(scr);
  lv_group_focus_obj(list);

  printf("%d\n", lv_btn_get_state(btn)); // expect the state is PR
  //if (lv_btn_get_state(btn) == LV_BTN_STATE_REL) {  // I expect that the state is PR.                                                            
  //    lv_btn_set_state(btn, LV_BTN_STATE_PR);                                                                                                    
  //    printf("%d\n", lv_btn_get_state(btn));                                                                                                     
  //  }     
}

hotpeperoncino, 2018-01-17:
No buttons in list has LV_BTN_STATE_PR state.

kisvegabor, 2018-01-17:
Hi,
Thank you for the feedback!
I tested you code and found two problem: one with your code and one with the library:
1. You forget to add `list` to the group
2. I found bugs in the code and fixed them

My test code:

```c

lv_res_t onclick(lv_obj_t *list_btn)
{
    const char* txt = lv_list_get_btn_text(list_btn);
    printf("%s\n", txt);
    return LV_RES_OK;
}

...

    lv_theme_t *th;
    th = lv_theme_mono_init(210, NULL);
    lv_theme_set_current(th);

    lv_obj_t *scr = lv_obj_create(NULL, NULL);
    lv_obj_t *list = lv_list_create(scr, NULL);
    lv_obj_set_size(list, 200, 200);
    lv_obj_align(list, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);

    lv_obj_t *btn = lv_list_add(list, SYMBOL_FILE, "Back", onclick);
    lv_list_add(list, SYMBOL_FILE, "Test", onclick);
    lv_list_add(list, SYMBOL_FILE, "Test2", onclick);
    lv_list_add(list, SYMBOL_FILE, "Test3", onclick);

    lv_group_t *group = lv_group_create();
    lv_indev_set_group(keypad, group);

    lv_group_add_obj(group, list);               /*Add the list to the group*/

    printf("%d\n", lv_btn_get_state(btn));

    lv_scr_load(scr);
    lv_group_focus_obj(list);

    printf("%d\n", lv_btn_get_state(btn)); // Due to the earlier fixes TGL_REL is the expected state
```

hotpeperoncino, 2018-01-17:
Thank you. when i rewrote my actual code to short program, i forgot calling lv_group_add_obj.

kisvegabor, 2018-01-19:
Hi,
Was it working well?

kisvegabor, 2018-01-23:
I suppose this issue also solved with updating lvgl.

closed 2018-01-23 by kisvegabor