lvgl/lvgl #77: big font lib store

helezh, 2018-01-17:
gui and font can stored in diff device,because many east asia language has huge sigmbols,it has the same size,you can make a api or private date pointe for others to ext the function,thanks!

helezh, 2018-01-17:
i like the style in 4th,it has desktop，5th is canced?

kisvegabor, 2018-01-17:
You mean store in external flash?

Yes the desktop is cancelled because it wasn't practical and well usable. Maybe if a similar but better concept is raised it can be added again.

helezh, 2018-01-17:
yes，big font lib stored in ext flash,ascii can buildin

kisvegabor, 2018-01-17:
Good idea! Thank you!
I add it the todo list and will implement in the next release!

xiaoleizii, 2018-01-18:
I wrote a GUI to achieve a similar function, the idea is very simple, only need to write a bitmap driver to get the data. Add an if statement to lv_font_get_bitmap () in lv_font.c to determine the range of Unicode encoding that needs external read. If the current character is within this range, the dot matrix data of the word is directly read from the external memory into the buffer And return to the first address of the cache can.

kisvegabor, 2018-01-19:
Please continue the discussion about font system improvements  here:
https://github.com/littlevgl/lvgl/issues/78

closed 2018-01-19 by kisvegabor