lvgl/lvgl #79: Button + List(two buttons) with keyboard

hotpeperoncino, 2018-01-18:
Hi.
I wrote the following program.
The screen has two objects (button and list(two buttons) );
We can change focus each other by TAB key.
But when focus stays at list, we cannot change focus buttons in list.

I expect that 
1. TAB key changes focus between button and list.
2. When focusing list, arrow key changes focus between "abc" and "def" list item button.

1 is ok, but 2 is not ok.
                                     
#include <stdio.h>
#include "lvgl/lvgl.h"
int onclick(lv_obj_t *o){return LV_RES_OK;}
#define SWIDTH 200
#define SHEIGHT  200
#define XMARGIN 5
#define YMARGIN 5
int app_init(lv_indev_t *inddrv)
{
  lv_theme_t *th;
  th = lv_theme_mono_init(210, NULL);
  lv_theme_set_current(th);

  lv_obj_t *scr = lv_obj_create(NULL, NULL);
  lv_group_t *group = lv_group_create();

  lv_obj_t *btn1 = lv_btn_create(scr, NULL);
  lv_obj_t *label1 = lv_label_create(btn1, NULL);
  lv_label_set_text(label1, "Back");
  lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, onclick);

  /*Crate the list*/
  lv_obj_t * list2 = lv_list_create(scr, NULL);
  lv_obj_set_size(list2, SWIDTH, SHEIGHT);
  lv_obj_align(list2, NULL, LV_ALIGN_IN_TOP_LEFT, XMARGIN, YMARGIN);;
  lv_list_set_sb_mode(list2, LV_SB_MODE_AUTO);

  int count =0;
  lv_list_add(list2, SYMBOL_FILE, "abc", onclick);
  lv_list_add(list2, SYMBOL_FILE, "def", onclick);

  lv_obj_set_size(btn1, 180, 40);
  lv_obj_set_pos(btn1, 5, 5);
  lv_obj_set_size(list2, 180, 100);
  lv_obj_set_pos(list2, 5, 45);

  lv_group_add_obj(group, btn1);
  lv_group_add_obj(group, list2);
  lv_indev_set_group(inddrv, group);
  lv_scr_load(scr);
  return LV_RES_OK; /*Return OK because the list is not deleted*/
}

kisvegabor, 2018-01-19:
Hi,

it worked for me fine. The problem could be in your keyboard driver.
You should the control keys to `LV_GROUP_KEY_...`

See this example:
https://github.com/littlevgl/lv_drivers/blob/master/indev/keyboard.c#L88

hotpeperoncino, 2018-01-19:
I use pc_emulator keyboard driver . I 'll check it!

xiaoleizii, 2018-01-22:
I test the code and have the same problem.
I can only use LV_GROUP_KEY_NEXT and LV_GROUP_KEY_PREV to switch btn and list, but i can not use the LV_GROUP_KEY_UP LV_GROUP_KEY_DOWN LV_GROUP_KEY_LEFTT LV_GROUP_KEY_RIGHT to switch the selected item in list.
I test my driver, it send key value use lv_group_send_data() but nothing happened after that.
In indev_proc_task(), i only find LV_GROUP_KEY_NEXT and LV_GROUP_KEY_PREV to switch focused object, the other key value use lv_group_send_data() to send.
In act->signal_func(), it send LV_SIGNAL_CONTROLL.
In lv_list.c i find that all the LV_SIGNAL_CONTROLL signal need have a btn in the list is LV_BTN_STATE_PR state. But when the GUI switch focus to the list, no button in the list is set to  LV_BTN_STATE_PR state,so all key value goes wrong!
I think this is the main problem why i cannot use up/down/enter key to control focus the item in the list.

kisvegabor, 2018-01-22:
As I said in a previous comment I updated the lubeary too. Please update lvgl. It should work then.

xiaoleizii, 2018-01-23:
Yes. I test the newest code,it works good.

kisvegabor, 2018-01-23:
Great!
You welcome!

closed 2018-01-23 by kisvegabor