lvgl/lvgl #80: 【Suggestion】adjust the btn position of tabview

qwert1213131, 2018-01-20:
The following contents are done with the translation tool. Please forgive me where the expression is not clear.
Can the BTN of tabview be adjusted to other locations, such as left, bottom, and right;
Whether the size of the BTN can be adjusted, sometimes the character is not fully displayed
[image]

kisvegabor, 2018-01-20:
Now only top position is possible.

I'm not sure I understand well. Do you mean the height of the BTN area is sometimes small and the text on the buttons are not fully visible?

qwert1213131, 2018-01-20:
hi , 

1. put the btn to the bottom position,  is it possible?

2.if there are more BTNs on the tabview, text isn't fully visible

thank you

kisvegabor, 2018-01-20:
Hi,
1. Now bottom position is not possible, but it's no so difficult to implement in the next release.
2. I see. And you mean put the BTNs to right or left could give a lot of space for the texts?

qwert1213131, 2018-01-22:
Hi，
     Yes, indeed.
     There is a similar situation in the EMWIN of segger company, as follows：
[image]
[image]
[image]

kisvegabor, 2018-01-23:
Thanks for the images.
In my opinion it would be better to keeps text horizontal when they are on the left or right. Because this way we really earn some extra space.

qwert1213131, 2018-01-23:
yes，the horizontal text is better option.
Top and bottom alignment can be used as an option.
Thank you !

kisvegabor, 2018-01-23:
Thank you for the idea!
I will add it in v5.1

kisvegabor, 2018-03-09:
There are some news in #147 
Let's continue there!

closed 2018-03-09 by kisvegabor