lvgl/lvgl #81: 收集一些关于littlevGL的论坛帖子讨论 Some forum discussions on LittlevGL

qwert1213131, 2018-01-23:
有很多的人开始对这个GUI感兴趣了，感谢作者的无私奉献和努力。
特收集一些帖子，也方便大家为自己的板子做对应的学习研究。
喜欢的可以star这个项目一下并做相应的代码修改提交。

There are a lot of people who are interested in this GUI, thanks to the author's selfless dedication and effort.
Special collection of a number of posts, it is also convenient for your own board to do the corresponding study.
If you like it，you can star this project and make the corresponding code modification to submit.

http://bbs.armfly.com/read.php?tid=29381&page=e
http://forum.armfly.com/forum.php?mod=viewthread&tid=82301&highlight=LITTLE
http://www.stmcu.org/module/forum/forum.php?mod=viewthread&tid=614534&fromuid=394920
https://debugdump.com/t_687.html
http://www.openedv.com/forum.php?mod=viewthread&tid=264861&highlight=LITTLE

kisvegabor, 2018-01-23:
Hi,
Thank you for sharing this post! 
It's very good to see that people are discussing about LittlevGL and trying it on their own hardware! :)

kisvegabor, 2018-03-12:
I close this issue to keep list of issues clean.

Thank you again for sharing these links! :)

closed 2018-03-12 by kisvegabor