lvgl/lvgl #82: Alternative draw on error

Zaltora, 2018-01-23:
Idea:
Add a return error (int) to GPU and flush function.
When a error occured (-1) on advanced draw, lvgl will swap to basic draw (map and fill ).

if map and fill fail too, the handler return an error to inform app user (or an other way).

kisvegabor, 2018-01-24:
In my opinion if you properly configure your hardware ad-hoc errors can not be occurred. Or what do you think which circumstances  can cause error?

Zaltora, 2018-01-24:
GPU already used with other interfaces for examples. ( esp32 got 2 core )

kisvegabor, 2018-01-24:
So do you suppose GPU as a shared resource?

Zaltora, 2018-01-24:
Yeah. Can be an option ? Or you think it is useless ?
I like the idea of integrating an error feedback for the user if needed.

kisvegabor, 2018-01-24:
For me a general error screen with a simple text if a critical error occurs seems more useful. For example a black screen with red error text.
It can be triggered from a lot of sources like:
- Out of memory
- GPU error
- Parameter error in a function 
- etc.

User should enable or disable this feature to lower overheads.

What do you think?

hreichert, 2018-01-24:
Good idea. The error text should be customizable (maybe a #define).

kisvegabor, 2018-01-24:
What about a function like:
`lv_error("my error text")`

or with printf like parameters:
`lv_error("Parameter 2 was too big: %d", param2)`

Then you can easily raise an error from application code too.

Zaltora, 2018-01-24:
Send a global IHM error to the screen is a good idea. (hoping the screen still work ). I agree
I suggest a function to retrieve this error too for user application. ( the last error occurred for example like libc)
It will allow user to know if we need reset the screen or send the information to others devices to know if the device screen work. Usefull with IoT project.

kisvegabor, 2018-01-24:
Indeed. 
Even a callback can be assigned by the user. It seems to me much simpler because user don't need to check the global flag.

Zaltora, 2018-03-20:
I suggest to change prototype "void fill/map/flush()" to "int fill/map/flush()" to retrieve possible transmission error.
I forget a case that the I2C/SPI or other transmission can fail if the resource is busy or the device didn't answer properly.
i had added  transmission error in drivers but this features is optional and can be remove with a define set to 0 (by default)

kisvegabor, 2018-03-20:
It sounds useful!
The only problem is that you can't see the return value of fill/map/flush functions because they are called internally.
Or do you mean LittelvGL should call the error function if these functions return with error?

Zaltora, 2018-03-20:
Exactly, lvgl need call the error callback function set by user if it exist.
Then lvgl will inform as parameter of this function the error type:
-    Out of memory
-    GPU error
-    Parameter error in a function
-    Transmission (EIO)
-    busy (EBUSY)
-    etc.

kisvegabor, 2018-03-21:
Good idea! Thank you!

It seems to me an `lv_error_t` enum should introduced. Let's collect the possible errors. So so far we have:
* LV_ERROR_OK
* LV_ERROR_NO_MEM
* LV_ERROR_PARAM
* LV_ERROR_GPU
* LV_ERROR_IO (Busy and Transmisson together)

Please tell your remarks and error ideas.

Zaltora, 2018-03-22:
use error system is a user choice. If user won't use it for performance reason, he can disable it. He can decide to ignore just a small part too. e.g for "LV_ERROR_IO", i got a macro like this one that i will use for each driver:
```c
#if SSD1306_TRANSMISSION_CHECK
#define verify_send(fn) do { int err; if((err = fn)) return err; } while (0)
#else
#define verify_send(fn) (fn)
#endif
```

kisvegabor, 2018-07-07:
Hi,

I added the `lv_log` module to `dev-5.2` branch. In `lv_conf.h` choose from these log level options:
- LV_LOG_INFO
- LV_LOG_WARN
- LV_LOG_ERROR
- LV_LOG_USER

You can register a callback with `lv_log_register_print` and add log like: 
`lv_log_add(LV_LOG_LEVEL_INFO, __FILE__, __LINE__, "log text...");`

Your callback can be not only a "print". It can reset the MCU on `LV_LOG_LEVEL_ERROR` or write the warnings to the screen etc.

Zaltora, 2018-07-11:
Great, i got a problem with lvgl and know it will help !

When power-up, i load a picture on my screen. Sometimes (i don't now why) this picture is not loaded.
this picture is on a flash and i use virtual filesystem from lvgl.
You think i will retrieve  this kind of error ? IT will help me to draw again if this happen.

kisvegabor, 2018-07-11:
I hope so!

However, the `log_add`s still not placed in LiittlevGL, you can do it to test the issue. (which is a strange issue, by the way) Please, let me know what was the solution to it.

kisvegabor, 2018-07-14:
As using `lv_log` you have the opportunity to add an "alternative draw on error" feature I close this issue.

Thank you for thinking together :)

closed 2018-07-14 by kisvegabor