lvgl/lvgl #83: Input pointer drag on empty container causes invalidation/refresh

upbeat27, 2018-01-25:
Easiest way to see and replicate this issue is to run the theme demo.
Go to Tab 3 - everything on this page fits on screen, so there is no vertical scrollbar.
If you drag in any direction in the empty area to the right of the floating windows (in the area where there are no objects to interact with) the result is essentially a refresh of the entire screen.
You can see this on Tab 2 as well, if you drag in the area between the graph/text/keyboard.
The expected behavior would be no areas redrawn, since no objects were interacted with/nothing actually changed on screen.

kisvegabor, 2018-01-26:
Thank you for reporting the issue!
It's not trivial how to fix it but I have an idea! When the scrollable area is dragged and re-positioned but it can not moved there (would be out of parent) it sets its position back. However meanwhile it was invalidated the areas during positioning. 
So the drag algorithm should know if a new area is invalidated but there were no movement then remove this new area.

kisvegabor, 2018-02-24:
Hi,

I fixed this bug `dev-5.1`. Can you test it?

upbeat27, 2018-02-27:
Tried it out with `dev-51` and it looks fixed! Closing this issue.

closed 2018-02-27 by upbeat27

kisvegabor, 2018-02-27:
Thank you very much!