lvgl/lvgl #84: [Feature Request] 4-bit color support

upbeat27, 2018-01-25:
There are a number of OLED display drivers that support 16 shades of gray.
For example, the SSD1322: [https://www.newhavendisplay.com/app_notes/SSD1322.pdf](url)
In this case the VDB can store 2 pixels per byte, saving RAM and also allowing for DMA transfers in the flush callback.
One caveat would be that the invalidation would always have to adjust the x1/x2 +/- 1 pixel since it isn't possible to send less than 2 pixels to the display.
I might be able to add this functionality myself and issue a pull request - but it probably wouldn't be any time soon, and I thought it would be worth it to at least bring this feature/idea into discussion.

Zaltora, 2018-01-26:
It can be a work combination with monochrome display that can store 8 pixel by byte.  (1 bit color support).

kisvegabor, 2018-01-26:
Hi,
It seems useful. It would be implemented with Zaltora's  suggestion. 
At first look the color create, and draw function are affected at most. 

You are right with rounding up the values. The only drawback is it can require additional drawing because the object's background might be invalidated too. However on monochrome or grayscale displays there are no really "heavy" drawing tasks so it doesn't need much extra time.

upbeat27, 2018-02-02:
FYI I have an OLED driven by SSD1322 working right now.

I have yet to totally implement 4 bit color though - right now I am just using 8 bit color mode, and themed everything such that the resulting value I pass to the OLED is 4 bit.

I also forgot that the driver actually can only address a minimum of 4 pixels at a time, not 2. So it forces the invalidated area to be stretched a bit more than I previously had mentioned.

Right now in my testing I just have it forcing x1 and x2 to make it work. But I think there should be some option to specify the minimum # of addressable pixels in lv_conf.h.
It would be similar to the **LV_INV_FULL_ROW** and **LV_INV_FULL_COL** from #66.
Something like **LV_INV_MIN_WIDTH** and **LV_INV_MIN_HEIGHT**  that would both be 1 by default. If either is not equal to 1, the code to restrict x/y to correct boundaries would be added in. Full row/column could be achieved by setting them to the screen width/height. SSD1322 would work by setting **LV_INV_MIN_WIDTH** to 4.

Does this make sense?  It is a step in the right direction for support of display controllers like the SSD1322. Actual 4 bit color support will require more work, of course.

kisvegabor, 2018-07-14:
`lv_refr_set_round_cb` is added in `v5.1` to enable the rounding of the invalidated areas. 

In #300 there is a conversation about a general solution for supporting individual color formats in VDB. It will be a solution for 4-bit VDB too. So please share your further thought about this topic there.

closed 2018-07-14 by kisvegabor