lvgl/lvgl #87: [Suggestion] Make GPU blend/fill functions 2D instead of 1D

upbeat27, 2018-01-26:
Currently the blend and fill functions use only a "length" parameter which allows a transfer of at most 1 line with hardware acceleration. LittlevGL loops over the height internally, calling the GPU function for each vertical line.

My suggestion would be to change these functions to be more like the flush function - accepting x/y coordinates (2D) instead of length (1D). That way 2D DMA can be used to fill/blend the entire area, instead of 1 line at a time (which sort of defeats the purpose of the GPU functionality).

This would change the API, but it wouldn't be hard to adapt to this new behavior. If someone needs it to work like the current 1D way, they can still loop over the height themselves within the custom blend/fill instead of LittlevGL assuming this responsibility.

I don't have an idea of how much this would reduce CPU usage, but my guess is it would be enough to make it a worthwhile change.

kisvegabor, 2018-01-29:
Good idea, but there is one thing: When bled/fill is called it need to bled/fill a part of VDB. If it was extended to 2D a lot of arguments would be needed to pass. For example:
```c
my_fill(dest, dest_x1, dest_y1, dest_x2, dest_y2, color);
my_blend(dest, dest_x1, dest_y1, dest_x2, dest_y2, src, src_x1, src_y1, src_y1, src_y2)
```

When I designed these interface functions the goal was the simplicity and easy portability even at the cost of minor performance loss. 
However I'm open to any idea and I'm convincable.

upbeat27, 2018-01-29:
I guess it does end up making the implementation of the interface functions more complicated, since you have to deal with offsets and such that aren't a factor when dealing with a single line.

kisvegabor, 2018-01-30:
Yes, however e.g. blending of a 100 px wide image line-by-line won't use the real power of GPU because of its the initialization overhead. 
So I will examine you idea deeper! Maybe replacing x1, y1 ,x2 ,y1 coordinates with an `lv_area_t * src_area` would help to keep the interface clear. And some good examples can explain how to use the GPU line-by-line or the whole area at once.

upbeat27, 2018-01-30:
That sounds good, I was going to suggest passing in lv_area_t * instead of x/y but I wasn't sure if you were purely against a lot of arguments, or just against having to handle more stuff within the function.

In the end, if someone is using the GPU functions they can probably handle the additional complexity of the 2D vs 1D. And like you said, an example of both should clarify things.

kisvegabor, 2018-01-31:
Okay, it sounds good! But unfortunately it can be added only in v6.0 because it changes API.

BenArtes, 2018-03-06:
I looked into implementing this myself and noticed a potential issue that might give some guidance:

`lv_disp_mem_fill' is used to fill a rectangle with a specific color.  `lv_vfill` has a static buffer the size of a single row that it fills with the desired color and passes, row by row, to lv_disp_mem_fill.  If the fill functions were to be come 2S< this buffer would have to become the same size as the screen (or maybe just the same size as VDB?), drastically increasing memory.

This seemed silly to me but I couldn't seem to find a way around it.

I'm developing on an STM32F4 and have DMA and the DMA2D available to me, but it seems the limiting factor is that DMA2D can't DMA_R2M while blending... Is that correct?  It appears that I can DMA the Color R2M to a correctly sized buffer then DMA_M2M_BLEND from that buffer, but I still need to waste all that memory even though I only want to blend against a single color.

Is there a better way to implement this?

kisvegabor, 2018-03-07:
> I'm developing on an STM32F4 and have DMA and the DMA2D available to me, but it seems the limiting factor is that DMA2D can't DMA_R2M while blending... Is that correct?

Yes, you are correct. I developed the mem_blend/fill with STM32F4 Discovery and noticed there is no DMA_R2M with alpha option. So I thought it's not so common and would be better to implement this feature in an other way. Hence the one line buffer came to the picture.

So, yes you are right. When working with areas it would be more practical to pass to `mem_blend` a same sized area as the destination area but it requires a big color buffer not only one line.

So let's summarize:
- mem_fill: works well with no alpha on areas  
- mem_blend: works well to blend too areas
- mem_fill: fails with opacity on areas

What if we leave it to user. He can implement `mem_fill` as he want. If filling an area with no opacity just simply fill with GPU. If filling with opacity he also can prepare a buffer, fill it with a color, and blend it in the fill function.
But maybe it's too comlicated and makes porting difficult. What do you think?

cutephoton, 2018-07-16:
I'm not familiar with lvgl... just exploring, as an option. I was reading the backlog and I believe I recognize the R2M issue back from when I was writing a toolkit on STM32.

One option for STM32 may be to use the CLUT. No need to fill or initialize a buffer. Just point to a valid memory region of sufficient size and fill the CLUT table with the same color. The color always decodes regardless of the input. Of course, it comes at the expense of pointless reads of memory. For small regions, a software renderer may perform better... (Admittedly, this relies on the developer being comfortable with such silliness, but there may be clever mitigation to make it look less like an awful hack.)

It's been a while since I worked on DMA2D so my memory is a bit fuzzy, but a garbage-in-proper-output approach would be highly satisfying to implement. ha ha ha. Sadly, I avoided the corner case entirely.

kisvegabor, 2018-07-16:
If I understood well:
1. Fill the whole CLUT (256 colors?) with e.g. ARGB(0.3,1,0,0)
2. Using this CLUT on any memory will result ARGB(0.3,1,0,0) because all possible values (0..255) are decoded as the same color.
3. And somehow blend this memory (which has indexed color format) to the destination.

Is it correct?

BesitzeRuf, 2018-07-16:
Hi guys,
the idea to change function the header of blend function to something like "ex_mem_blend(lv_area_t * src_area, uint8_ src_opa, lv_area_t * dest_area, uint8_t dest_opa)... But the implementation should be defined by user I think. If DMA2D interrupt is used, the rendering can be fast too. Maybe it is better just implement own design functions, which will be optimized for DMA2D usage, but for me current implementation is unbelievably fast ( even for 800x480 display resolution)...

cutephoton, 2018-07-16:
@kisvegabor - I think you can use 4-bit CLUT so 16... but yeah, that's more or less it. You can also get a bit more clever with the flash memory such that you can render patterns.

@BesitzeRuf - First rule of optimization: Don't. :) STM32s are pretty fast. For many applications, the CLUT workaround is probably unnecessary.

kisvegabor, 2018-07-17:
@cutephoton I see, thank you. Please, let us know if you tried to use LittlevGL's `mem_blend` with CLUT. I'm very curious because ST's official view is that this feature (R2M with alpha) is not supported: https://community.st.com/thread/48453-chrom-art-transparent-rectangles

@BesitzeRuf  Good to hear that! :) Can you make some test with the [Benchmark application](https://github.com/littlevgl/lv_examples/tree/master/lv_apps/benchmark)  with and without DMA2D?

BesitzeRuf, 2018-07-17:
@kisvegabor Yes, I will try to prepare the clean demo for stm32f429-disc1 board. Probably till end of this week.

BesitzeRuf, 2018-07-25:
Hi all,

I upload benchmark project in which lcd driver uses only DMA2D. [Here is the repository](lv_refr_set_onScreenRefreshed_cb). Here is and example how to rotate the screen and ofcourse in this case we cannot to flush whole area by DMA2D, but we can to do this line by line and the perfomance doesn't decreased to much. I haven't example without DMA2D, but the **@kisvegabor's** [project](@kisvegabor ) can be used to compare the results. 

With rotated screen (optimization level -O3 and disabled AA) I still have these results:

Nothing is enabled - 9.6 ms
Enabled wallpaper  - 12.1 ms
Enabled Shadow - 20.1 ms
Enabled opacity - 10.5 ms
Enabled wallpaper + recolor - 34.s ms
Everything is enabled -  44 ms

Some notes.... 
In "[ili9341_drv.h](https://github.com/BesitzeRuf/32F429IDISCOVERY_LVGL_BENCHMARK/blob/master/inc/ili9341_drv.h)"  dy define LCD_ENABLE_VSYNC you can enable/disable vertical synchronization feature, which swaps between 2 buffers to prevent flickering effect.  To have it, I also modified files "refr.c" and "refr.h". But it is experimental thing...

kisvegabor, 2018-07-25:
@BesitzeRuf 
Thank you very much! These are impressive results! It worse case we still have ~20 FPS, but 40-50 FPS is possible with more life-like settings.

Are you sure `LCD_ENABLE_VSYNC` swaps two buffers? Not just waits for a VSYNC to synchronise the refreshing?

The first two links are broken (404 error) but the last one to ili9341 is working. So I suppose your project is here: https://github.com/BesitzeRuf/32F429IDISCOVERY_LVGL_BENCHMARK
Your ILI9341 and DMA2D driver seems very detailed. Congratulation! :)

FYI, to disable DMA2D on these modificatinos are required:
```c
disp_drv.mem_blend = NULL; /*Blend two color array using opacity*/
disp_drv.mem_fill = NULL; /*Fill a memory array with a color*/
```

I will also test is soon!

BesitzeRuf, 2018-07-25:
@kisvegabor Yes, link should be https://github.com/BesitzeRuf/32F429IDISCOVERY_LVGL_BENCHMARK (fixed). 

To synchronize refreshing I am using line interrupt of LTDC. It is almost the same as pooling VSYNC. Before I can swap buffers, I need to do it in the right moment and after buffers are swapped, all refreshed areas should be duplicated to the rendering buffer (in simple way just copy whole rendered screen), before next screen refreshing. 

By setting mem_xx  functions to NULL you will not disable DMA2D in my code. I don't remember, but maybe functions "**lv_hal_lcd_map**", "**lv_hal_lcd_fill**" may not work properly. The point use DMA2D in every function is to free DMA master and compare the performance. It is just an experiment :-)

kisvegabor, 2018-07-27:
Thank you for the explanation. 

> in my lv_conf.h (double buffer was disabled by this)

Does it means `lv_VDB_DOUBLE`? In the updated table the left and right values refer this?

BesitzeRuf, 2018-07-27:
In my case the screen I rotated and benchmark example looks a little bit different because of the alignment of buttons. If the screen scrolled to the top, you should get approximately same values as well as from the left side. Values from the right side are related to the screen scrolled to bottom.

kisvegabor, 2018-07-29:
I tested the project in [lv_projects](https://github.com/littlevgl/lv_projects/tree/master/stm32f429_discovery_no_os) with the following settigns:
- Frame buffer in the internal RAM and  `LV_VDB_SIZE         (40 * LV_HOR_RES)` and with or without GPU
- Frame buffer in the external RAM  and `LV_VDB_SIZE         (160 * LV_HOR_RES)` and with or without GPU

- 

| Config   | Nothing  | Wallpaper      | Shadow| Opacity | Wallpaper + recolor | Everithing |
|---------------------------------------|-------------|-------------------|------------|-----------|-----------------------|---------------|
| No GPU, Int FB, Int VDB, 1 VDB |

[image]

It's clear the @BesitzeRuf's implementation is much faster then the curently used in the STM32F429 Discovery project.

kisvegabor, 2018-07-29:
I tested the project in [lv_projects](https://github.com/littlevgl/lv_projects/tree/master/stm32f429_discovery_no_os) with the following settigns:
- Frame buffer in the internal RAM and  `LV_VDB_SIZE         (40 * LV_HOR_RES)` and with or without GPU
- Frame buffer in the external RAM  and `LV_VDB_SIZE         (160 * LV_HOR_RES)` and with or without GPU

- 

| Config   | Nothing  | Wallpaper      | Shadow| Opacity | Wallpaper + recolor | Everithing |
|---------------------------------------|-------------|-------------------|------------|-----------|-----------------------|---------------|
| No GPU, Int FB, Int VDB, 1 VDB |
| No GPU, Int FB, Int VDB, 1 VDB |
| No GPU, Int FB, Int VDB, 1 VDB |
| No GPU, Int FB, Int VDB, 1 VDB |
| No GPU, Int FB, Int VDB, 1 VDB |
| No GPU, Int FB, Int VDB, 1 VDB |
| No GPU, Int FB, Int VDB, 1 VDB |
[image]

It's clear the @BesitzeRuf's implementation is much faster then the curently used in the STM32F429 Discovery project.

kisvegabor, 2018-07-29:
@BesitzeRuf Thank you for the explanation!

I tested the project in [lv_projects](https://github.com/littlevgl/lv_projects/tree/master/stm32f429_discovery_no_os) in the following cases:

A) Internal Frame buffer, No GPU, 1 VDB in internal RAM (LV_VDB_SIZE = 32 lines)
B) External Frame buffer, No GPU, 1 VDB in internal RAM (LV_VDB_SIZE = 160 lines from now )
C) External Frame buffer, GPU, 1 VDB in internal RAM
D) External Frame buffer, GPU, 2 VDB in internal RAM
E) BesitzeRuf's results

The result are in **millisecond** unit.

| Settigns                   |  A          |      B  (ext FB)     |      C (GPU)     |    D (2 VDB)       |   E (BesitzeRuf)  |  
|---------------------------|------------|-------------|-----------|-----------|-----------|
Nothing                     |     19     |     12      |     11     |     9      |    10
Wallpaper                 |     20     |     14      |     13     |     12    |    12
Shadow                    |     55    |      32      |    28      |     27   |     19
Opacity                     |    26     |     19      |    13      |     12   |     11
Wallpaper + recolor  |    43     |     36      |    35      |     33   |     30
Everithing                 |    95     |     69      |    54      |    53    |     41

[image]

**Conclusion**
1. A bigger VDB (B) was mush more effect then the GPU (C) 
2. BesitzeRuf's driver is faster then the current in the project.

**Question**
Why has the GPU so small effect? Because it's 1D?

BesitzeRuf, 2018-07-30:
@kisvegabor  The DMA2D has small effect because LittlevGL is not fully optimized to use DMA2D. The fastest performace you can gain (I think..) only when you will manipulate with memory areas, instead of drawing object pixel by pixel, line by line...For example just for once create a number of pictures of buttons, icons, fonts (as bitmap picture) and with DMA2D transfer these to right place. Please correct me, if I am wrong, but actual image format is not prepared to use directly by DMA2D. Each line of the image (with alfa byte) is prepared in 
> void lv_draw_img(const lv_area_t * coords, const lv_area_t * mask, const lv_style_t * style, const void * src) 
instead to configure some parameter of DMA2D and just transfer it.... And many many (but little) things like these.. 

.. Fewthougths about image recolor.. With DMA2D we can prepare additional temporary buffer with size equal IMG_WIDTH x NUMBER_OF_LINES, filled by the color. After blend destination image by temporary buffer (one or more times, till whole image is recolored)... 

But... you library is still awesome, maybe here will be some "ports" with DMA2D optimization...

@cutephoton Yes, true. RTOS can be used as solution, for example just call "lv_task_handler()" in while loop.

cutephoton, 2018-08-01:
@BesitzeRuf This could work but using lv_task_handler may be ill-advised. A task could manipulate the data structures in a manner that would yield unexpected results. It could end up being counter productive (significantly slower) if not written carefully.

---

I noticed something interesting. I took time to refactor the ST provided BSP code to work with CubeMX generated code with FreeRTOS. For the 469i-DISCO board, the BSP package seems to runs the LCD in video mode with burst. There is a different mode of operation called adapted command mode. This mode relies on an LCD screen with a frame buffer (GRAM). You can perform full or partial updates to the framebuffer too. OTM8009A has GRAM. Sufficed to say, this frees up the bus, reduces contention and memory consumption. DMA2D performance might improve.

BesitzeRuf, 2018-08-01:
@cutephoton Working with GRAM of LCD will free just the memory space from other place (SRAM, SDRAM... other external ram), but the performance will be decreased, because the best results you can get with 2 VDB buffers and these buffers should be placed only in internal SRAM of MCU ( I tested all possible configurations :-) )

kisvegabor, 2018-08-01:
**1. Small effect of GPU and RTOS benefit**
@cutephoton I agree, it would be an interesting measurement to see how much time does it take to finish a DMA2D operation.

@BesitzeRuf Actually the images can be passed to DMA2D. It happens in the `blend` function. But now, for simplicity, the images are passed line-by-line instead of a rectangle. Speaking about an RTOS environment and yielding the task while polling for DMA2D would be better to have a 2D blend function because it would mean only 1 context switch per image.

**2. Image recolor**
The image can be recolored with an overlaying opaque rectangle only if there are no gaps on the image (not chrome keyed or no alpha byte) because only really visible pixels should be recolored.

**3. Shadow**
Usually, the shadow is small, e.g. < 10 px. Initializing the DMA2D for 10 px might take more time than calculating these few number of pixels by software.  

**4. Buffer objects**
A lot of things would be much easier if some kind of "temporal graphics buffer" could be used. It would be much easier to avoid drawing pixels twice or would be much easier to make blur (shadow). However, the whole library tries to be very effective with regard to memory usage and not using extra temporal buffers for graphics. However, I have an idea for it! See below!

**5. Better GPU support**
Maybe you heard that v6.0 will have multi-display support. You will be able to register more displays and unique `display_flush` functions and each display can have its own resolution. But the another displays would be the layers of the LCD controller. You could add e.g. a wallpaper to one and just objects to the other. This way the wallpaper needs to be drawn only once and the LCD controller will merge the layers by hardware. Or another example: you can add a big grey opaque rectangle to an upper layer to show that the current content is inactive. The opacity can be controlled by the layer attributes.

What do you think?

BesitzeRuf, 2018-08-05:
@kisvegabor  For small shadows (or maybe any other manipulation with pixels), yes it will be faster, I tried to sat for example, to create font like this: [image]

I thought (never tested, now I making some other tests) we will need to have just pixel map in format A4/A8 with contains values of alpha channels of shadowed pixels ( 0 - transparetnt, 7 - 50% transparent, 15 - 100% dark), then set this memory map as source for both layers of DMA2D (white color as background, and shadows for foreground) but with different configurations and blending options... For background I hope it is possible to replace source alpha channel by constant (in our case for example to 100%).. for foreground set the color in color register (for this picture it should be black).. Maybe it will be slow, but it should free mcu...

By the way, one thing (maybe crazy) which I want to test, if it is possible to transfer DMA2D configuration (by reference manual without CLUT table, it should be around 80 bytes) by DMA :D... So then we will need in interrupts just just turn on DMA2D... 

Things you had described in **5. Better GPU support** should really improve flexibility of your library. I will wait for it.

kisvegabor, 2018-08-09:
**About fonts**
Actually, now the bpp (bit-per-pixel) means the alpha value of the pixel of a character. There are 4 options: A1, A2, A4, A8 which is the same as 1 bpp, 2 bpp, 4 bpp, 8 bpp respectively. However, I also didn't test a weird font like this. What's the name of the font in the image? Or can you attach it's TTF file?

**Transfer DMA2D config wit DMA**
It really sounds crazy. :D I'm curious about your results!

kisvegabor, 2019-03-02:
For `v6.0` I re-considered the use of 2D GPU functions but found several obstacles. The most complicated one is to blend an arbitrary area of an image to a buffer with arbitrary size considering a "mask" area. Requesting the user to implement such a hard-to-imagine function is not seems reasonable to me. So finally I'd like to leave the original 1D function. I'm still thinking about how to improve the interface.

Anyway, thank you very much for the conversation, measurements, tests, and contribution here!

closed 2019-03-02 by kisvegabor

turoksama, 2019-03-21:
@kisvegabor 
I suggest to leave obviously special 2D GPU operations such as rotate, move, fill and etc as __weak functions, let users to deal with GPU if needed because of some SOCs are rich supported with it.
What do you think?

kisvegabor, 2019-03-22:
AFAIK `__weak` is not supported by every compiler. Instead, callbacks can be registered in the display driver.

I examined the GPU topic further and found that the current 1D fill can be improved to 2D fill. To others (rotate, transform etc) are out-of-scope now because they can't be easily attached to LittlevGL drawing interface.

turoksama, 2019-03-22:
Hmm, Well. just for discussion, 2D GPU features from an ARM9 soc: NUC972
[image]

kisvegabor, 2019-03-22:
Thank you for the list!

I've just found a good opportunity to make use of these advanced GPU functions. [Here](https://github.com/littlevgl/lvgl/issues/684#issuecomment-470191818) were talking about image rotate support. Now it works with the canvas object. It means an image can be rotated and stored in a RAM buffer. We are planning to improve the canvas with other features too (e.g. zoom). 

So what I can imagine is adding an interface to the canvas to use GPU accelerated transformation. As LittelvGL's drawing algorithms are highly optimized they have some limitations regarding to flexibility. But the case of the canvas object is still much more simple because you just have 2 buffers to make on opereation on them. 

I'll mention it in the mentioned topic.

turoksama, 2019-03-23:
That's amazing. As you can see, bresenhan line drawing is supported within the advanced 2d gpu, so, the library may be a lot faster with these accelerations.Well, anti-aliasing is not supported with this soc.
I've successfully achieved 3D object manipulation with canvas, so, it is very flexible.
ST published there new soc STM32MP1xx which contains a Application core and Cortex-M4 core, amazingly embeded a 3D GPU. For overhead future works, maybe we should take a snapshot with it.

kisvegabor, 2019-03-24:
> I've successfully achieved 3D object manipulation with canvas, so, it is very flexible.

Awesome! Is this project public? If yes it could be a nice reference to use advanced GPU with LittlevGL.

> ST published there new soc STM32MP1xx which contains a Application core and Cortex-M4 core, amazingly embeded a 3D GPU

That will be the future. MCU-s with a 3D engine. It's good to be prepared for it

turoksama, 2019-03-24:
Unfortunately it does not use GPU rendering, only in software rendering due to MCU platform.
It is not fully functional yet(stucked with sscanf in ARM platform), and I would love to share when it is ready in the future updates.
If STM32MP1 used, besides licensing, I think OpenGL might be a good choice when 3D GPU engine exists.

kisvegabor, 2019-03-24:
> It is not fully functional yet(stucked with sscanf in ARM platform), and I would love to share when it is ready in the future updates.
If STM32MP1 used, besides licensing, I think OpenGL might be a good choice when 3D GPU engine exists.

Sound good! Please, let me know if you have some news.