lvgl/lvgl #88: Porting ADS7846 Touchscreen

avafinger, 2018-01-29:
Hi,

I am porting to TFT 2.8 (320x240) with ADS7846 touch which is capacitive and wonder which coordinates should be passed to:
**bool my_input_read(lv_indev_data_t *data)** .

Although ADS7846 is capacitive i get ABS_X and ABS_Y values ranging from 290 to 4000, just like on a  resistive touch. I was expecting ABSOLUTE values like 0 ~ 320 and 0 ~ 240 but get these relative values.

Can you share some thoughts if it is OK to pass those values and if not any idea what should be done in these case?

qwert1213131, 2018-01-29:
https://github.com/littlevgl/lv_drivers/blob/master/indev/XPT2046.c
You can refer to the web site above。

kisvegabor, 2018-01-29:
Hi,

you need to scale the values to match with the screen resolution.

avafinger, 2018-01-29:
@qwert1213131 @kisvegabor 
Thanks for the reply, will do that and post the resilts.

avafinger, 2018-01-31:
I was about to test the implementation but hit a wall, i cannot find where **lv_indev_register** .
Can you please point where it is?

qwert1213131, 2018-01-31:
Hi avafinger ,
please refer the link:
https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/0_porting/lv_tutorial_porting.c

avafinger, 2018-01-31:
@qwert1213131 

Thanks for pointing, i used as reference the **Porting Documentation** there is a typo there:

Input device interface
To set up an input device an lv_indev_drv_t variable has to be initialized:

```
lv_indev_drv_t indev_drv;
lv_indev_drv_init(&indev_drv);     /*Basic initialization*/
indev_drv.type = ...               /*See below.*/
indev_drv.read = ...               /*See below.*/
lv_indev_register(&indev_drv);     /*Register the driver in LittlevGL*/

```

kisvegabor, 2018-01-31:
Thank you, I fixed it on the webpage!

avafinger, 2018-01-31:
@kisvegabor , @qwert1213131 

Ok, i got it working using XPT2046 as a base but need some adjustments. I will try to explain briefly, see if you can help. So this is the first attempt and i don't know much about lvgl (by the way awesome work)
I use demo_create(); to check how touch is working and i don't know where i got that ads7846 was capacitive... anyway, is resistive. I am using events to grab the touch points, and when i touch the panel i receive tons of events and i do normalize x and y as in XPT2046, but as an example if i touch (Write tab) and the letter 'w', the letter 'w' is highlighted (pushed) but letter 'a' and 's' are 'pushed' in a sequence, it does not happen all the time but very frequently, say 80% of the time. The logic i am using is that:
* check if there is a touch event
* if there is, grab the x and y
* normalize x and y
I think the problem arises when i receive a dozen events for a single touch, and obviously the resistive touch is giving me different location and these new touch points are being passed to lvgl.

```
/**
 * Get the current position and state of the touchpad
 * @param data store the read data here
 * @return false: because no ore data to be read
 */
bool ads7846_read(lv_indev_data_t * data)
{ 
    static int16_t last_x = 0;
    static int16_t last_y = 0;
    bool valid = true;
    
    int16_t x = 0;
    int16_t y = 0;
    
    int touch = get_touch_event(&x, &y);

    if (touch) {
        
        /* --- Normalize Data --- */
        ads7846_corr(&x, &y);
        ads7846_avg(&x, &y);
        
        last_x = x;
        last_y = y;
    } else {
        x = last_x;
        y = last_y;
        avg_last = 0;
        valid = false;
    }
    
    data->point.x = x;
    data->point.y = y;
    data->state = valid == false ? LV_INDEV_STATE_REL : LV_INDEV_STATE_PR;

    return valid;
}

```
I haven't tried anything yet, so i am just describing it to hear the best approach to solve this.

I have some question about the demo_create(), if i touch a letter on the keyboard (Write tab) it should write the letter to the Text area?

kisvegabor, 2018-01-31:
I think your issue happens because when you begin to release the touch pad in the very last moment it is still touched but you read false values.

I suggest to set `LV_INDEV_POINT_MARKER` to 3. It will put red point where the screen touched.

> I have some question about the demo_create(), if i touch a letter on the keyboard (Write tab) it should write the letter to the Text area?

Yes, it should. Does it not write the letters?

avafinger, 2018-01-31:
After several attempts i got one letter written to text area, but 95% it highlights two or three letters and does not write anything or the 'first' letter to the text area, the one that should.

kisvegabor, 2018-01-31:
Have you tried the LV_INDEV_POINT_MARKER ? 

You should not create the demo to try the touch pad on an empty screen.

avafinger, 2018-01-31:
Not yet, i am revising the get_event and will pass x and y only when touch is up. I report soon my findings, thank you.

avafinger, 2018-01-31:
@kisvegabor 

I got it right this time, maybe on missed thing, on [List tab] i have parcial list, should i be able to scrool down? I've noticed i can only do touch and not drag.

Can you please confirm if i should be able to scroll down a drag or mark some part of the text on the Text Area where there is a string added?

avafinger, 2018-01-31:
OK, i see i should be able to drag and scroll down the list, but when i touch a button inside the list it quickly shows a push down and up, no time enough to start a long press. I will enable LV_INDEV_POINT_MARKER and where is the touch point.

kisvegabor, 2018-02-01:
It seems the TP data is noisy. LV_INDEV_POINT_MARKER should reveal it!

avafinger, 2018-02-01:
yes, TP data is noisy but TP works nice with X11. I think reading TP registers for X and Y is best solution but i don't know how to do it in user space. The idea was to read /dev/input/event* just like HAL so no matter what kind of TP is on the panel but i get crazy TP data, something is like getting only X, or only Y from the events and sometimes just a PUSH event without X or Y. My first approach worked but i lost dragging or **long push** cause i discarded many events and returned RELEASE when in fact should be still a PRESS. Second approach was to read every single event, that works and i have touch and dragging working but kind of slow, like in slow motion, not that slow but very noticeable. So, reading one by one event is slow, reading as many events as possible is to fast. Maybe i should implement some kind of queue, but that i was expecting to be done by the kernel /dev/input/event*. The problem relies on how i get TP data. Any thoughts?

kisvegabor, 2018-02-02:
Can create a log about the events? Like: 
1. lv_read started
2. x read
3. y read
4. press read
5. release read

If you attache a log like this than we can think together.

avafinger, 2018-02-02:
Ok, i will provide a log but it looks similar to the one provided above. Just recorded a short video to show how it looks like, i think i will need to calibrate the values on the edges, forget about quality of video. Thanks for your patience.

https://drive.google.com/open?id=1qQGtMG6ijRP6FKhA-xh2cLulTuLdEkzb

avafinger, 2018-02-02:
I added a big circular buffer and get events from there, it is now much more smooth.
Just noticed in Text Window that cursor is off when text is getting more characters, i think you need to revise how cursor position is been calculated .

https://drive.google.com/open?id=1mUsQ37zqgz27zaJPC3y8b1biraNF732A

kisvegabor, 2018-02-04:
I see. Because the screen refresh is fast I suppose the programs stays too long in your `indev_read` function.
The code of `indev_read` and the mention would be useful to examine your issue.

avafinger, 2018-02-05:
@kisvegabor 

Thanks for your tip. I buffered the TS events in a dumb circular buffer and got better results. So i read a chunk of events and then get one by one from this buffer.

Here is the log, just for the record. **valid: 1** means i get ABS_X and ABS_Y and return **LV_INDEV_STATE_PR**, when no events or no more **ABS_PRESSURE** i return **LV_INDEV_STATE_REL**

read_events() reads all available events into the circular buffer and read_event gets the events one by one from that buffer instead of reading /dev/eventX one by one.

There must be a better way of doing this, but it works. Anyways i have another LCD here with capacitive TS and this approach might work with any TS i would think.

I will clean up the code, test on this capacitive TS and try to PR my changes.
You may close the issue if you wish. And looking for your new version, maybe with the cursor position fixed?

Keep the good work and Thanks

```
--------------- fd_ads7846_touch: 4
read_events() #: 14 - complete read: 1 - read_max: 144 - next free slot: 14
ABS_X: 3421 - normalized: 266
ABS_Y: 843 - normalized: 44
-> pos(x=266,y=44)
read_event valid (x,y): 1
read_events() #: 15 - complete read: 1 - read_max: 130 - next free slot: 29
ABS_X: 3435 - normalized: 266
ABS_Y: 836 - normalized: 44
-> pos(x=266,y=44)
read_event valid (x,y): 1
read_events() #: 16 - complete read: 1 - read_max: 115 - next free slot: 45
-> pos(x=266,y=44)
read_event valid (x,y): 0
read_events() #: 3 - complete read: 1 - read_max: 99 - next free slot: 48
ABS_X: 3437 - normalized: 267
-> pos(x=267,y=44)
read_event valid (x,y): 1
ABS_X: 3434 - normalized: 267
ABS_Y: 841 - normalized: 44
-> pos(x=267,y=44)
read_event valid (x,y): 1
ABS_X: 3439 - normalized: 267
ABS_Y: 845 - normalized: 44
-> pos(x=267,y=44)
read_event valid (x,y): 1
ABS_X: 3445 - normalized: 267
ABS_Y: 841 - normalized: 44
-> pos(x=267,y=44)
read_event valid (x,y): 1
ABS_X: 3431 - normalized: 266
ABS_Y: 838 - normalized: 44
ABS_X: 3442 - normalized: 266
ABS_Y: 834 - normalized: 44
-> pos(x=266,y=44)
read_event valid (x,y): 1
ABS_X: 3441 - normalized: 266
ABS_Y: 829 - normalized: 43
-> pos(x=266,y=43)
read_event valid (x,y): 1
ABS_X: 3450 - normalized: 267
ABS_Y: 834 - normalized: 43
-> pos(x=267,y=43)
read_event valid (x,y): 1
ABS_X: 3453 - normalized: 267
ABS_Y: 829 - normalized: 43
-> pos(x=267,y=43)
read_event valid (x,y): 1
-> pos(x=267,y=43)
read_event valid (x,y): 0
read_events() #: 17 - complete read: 1 - read_max: 144 - next free slot: 17
ABS_X: 2282 - normalized: 174
ABS_Y: 664 - normalized: 33
-> pos(x=174,y=33)
[... 2512 lines cut ...]
ABS_X: 2182 - normalized: 168
ABS_Y: 2995 - normalized: 172
-> pos(x=168,y=172)
read_event valid (x,y): 1
ABS_X: 2172 - normalized: 167
ABS_Y: 2923 - normalized: 171
-> pos(x=167,y=171)
read_event valid (x,y): 1
ABS_X: 2197 - normalized: 167
ABS_Y: 2885 - normalized: 170
-> pos(x=167,y=170)
read_event valid (x,y): 1
ABS_X: 2203 - normalized: 166
ABS_Y: 2868 - normalized: 168
-> pos(x=166,y=168)
read_event valid (x,y): 1
ABS_X: 2220 - normalized: 167
ABS_Y: 2955 - normalized: 168
-> pos(x=167,y=168)
read_event valid (x,y): 1
ABS_X: 2221 - normalized: 168
ABS_Y: 2374 - normalized: 160
-> pos(x=168,y=160)
read_event valid
```

kisvegabor, 2018-02-05:
Hi,
okay, I'm happy to hear that it is working better now.

Just a question: are you using the buffering feature of input drivers (returning true to indicate there are more data be read) I'm asking just because there is bug with it which will be fixed soon.

avafinger, 2018-02-05:
I think so. I just do a:

Initialization:

```
    ads7846_init();
    lv_indev_drv_init(&indev_drv);     /*Basic initialization*/
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read = ads7846_read; 
    lv_indev_drv_register(&indev_drv);     /*Register the driver in LittlevGL*/
```

And **lv_indev_read** calls **ads7846_read** which returns **valid = true** when i get  **ABS_PRESSURE** and there are more events buffered and **valid = false** when there are no more events to pop from my circular buffer.

If ads7846_read is spending too much time is cause i normalize ABS_X and ABS_Y **separately** since i don't always get a pair, some times only ABS_X, some times only ABS_Y.

Your question is interesting, i can only return a  **LV_INDEV_STATE_REL** on next call (i think 50 ms later) when there are no more Touch events. I think this explains why the slowness feeling but i am not sure.

I thought i had posted some Event Data log to explain my point but could not find it above. Attached is the event log just for the record.
[ev.txt](https://github.com/littlevgl/lvgl/files/1694869/ev.txt)

avafinger, 2018-02-05:
@kisvegabor ,

FYI i have ported the same logic to a capacitive touch GT9xx_ts (1024x600) and it works, so i should call this **hal_touch**.c (https://drive.google.com/open?id=1OlZb3l3E7zQI91bM_eFUdBmTLpwxXZyC).
16 bit and 8 bit int size in a loop  seems to be expensive for 64 bit arch, i will try to find a better way to normalize / calibrate X and Y points, maybe like X11 does. I will try to find a way to implement it if i can.

A couple of observations and some requests if you allow me to do it in this post.

* you can see the slider loose focus when dragging it and drags the screen together, i think it is a bug.
* i could not test rolling the listbox, but i think this driver is in better shape than ads7846.

Your Gui has great potencial, so i would kindly request if possible:
* Drag and place icons on screen just like Android (seems to be complex and i can't help in this area)
* have an edit text, when touch the editable area it would drop the keypad, scrolled if needed

kisvegabor, 2018-02-06:
Wow, very nice! I've never seen LittlevGL on such a large TFT :) Thank you for the video.

> 16 bit and 8 bit int size in a loop seems to be expensive for 64 bit arch

Which loop do you mean?

**observations**
- It's not bug, it's a feature, or at least these is the expected behavior. But I see your point. Some objects should keep the input device during the press.  I have an idea how to do it. I will add it to the next release.
- I think you can't roll the list because it has to few elements on this large screen. You can add some elements manually in `demo_create()` 

**requests**
- Can explain it in more detail? Even in a new issue dedicated to it.
- If I understand you well, you can do it by adding an *action function* to the text area. to open the keyboard when it is clicked. Anyway the demo should work like this. I will add it.

avafinger, 2018-02-06:
> requests
> 
> Can explain it in more detail? Even in a new issue dedicated to it.

Ok, will do.

> If I understand you well, you can do it by adding an action function to the text area. to open the keyboard when it is clicked. Anyway the demo should work like this. I will add it

Yes, i think it was intended to work like that but once i close the keyboard i cannot open it anymore if i want to type more text, i think would be nice if you could just click in the text area and the keyboard would re-appears, i don't know if i missed how to re-open the keyboard?

Think of this situation, i have a dialog with the input:
user: [ _________ ]
pasw: [ ________ ]
I click in the user edit text area and keyboard pops up below or the user text and the edit text area scrolls up to adjust the keyboard below the text area and after the user finishes typing the keyboard is dismissed , same for the password text.

8 bit and 16 bit in size in a **for loop**, is nice when dealing with 16 bit or 32 bit architecture with memory size constraint, but i think has a penalty in 64 bit arch. 
**uint8_t i;** and you have int16_t * x and int16_t * y, i would just use **int** and let the compiler do the optimizations, sure you save a lot of memory with int16_t size but in expense of speed, but **take it just as my observation**, i can change the loop on the file here and see if it improves, no problem. Changing x and y size needs a re-write of library.
I have to normalize x an y individually cause i don't get always both values. These are the routines that needs to be optimized some how.

```
**********************
 *   STATIC FUNCTIONS
 **********************/
static void ads7846_corr_x(int16_t * x)
{
    if((*x) > ADS7846_X_MIN) (*x) -= ADS7846_X_MIN;
    else (*x) = 0;
    

    (*x) = (uint32_t) ((uint32_t)(*x) * ADS7846_HOR_RES) /
                             (ADS7846_X_MAX - ADS7846_X_MIN);

    
#if ADS7846_X_INV != 0
    (*x) =  ADS7846_HOR_RES - (*x);
#endif
   

}

static void ads7846_avg_x(int16_t * x)
{
    /*Shift out the oldest data*/
    uint8_t i;
    for(i = ADS7846_AVG - 1; i > 0 ; i--) {
        avg_buf_x[i] = avg_buf_x[i - 1];
    }

    /*Insert the new point*/
    avg_buf_x[0] = *x;
    if(avg_last_x < ADS7846_AVG) avg_last_x++;
    
    /*Sum the x and y coordinates*/
    int32_t x_sum = 0;
    for(i = 0; i < avg_last_x ; i++) {
        x_sum += avg_buf_x[i];
    }

    /*Normalize the sums*/
    (*x) = (int32_t)x_sum / avg_last_x;
}

static void ads7846_corr_y(int16_t * y)
{
    
    
    if((*y) > ADS7846_Y_MIN) (*y) -= ADS7846_Y_MIN;
    else (*y) = 0;

    (*y) = (uint32_t) ((uint32_t)(*y) * ADS7846_VER_RES) /
                             (ADS7846_Y_MAX - ADS7846_Y_MIN);
   
    
#if ADS7846_Y_INV != 0
    (*y) =  ADS7846_VER_RES - (*y);
#endif
    

}

static void ads7846_avg_y(int16_t * y)
{
    /*Shift out the oldest data*/
    uint8_t i;
    for(i = ADS7846_AVG - 1; i > 0 ; i--) {
        avg_buf_y[i] = avg_buf_y[i - 1];
    }

    /*Insert the new point*/
    avg_buf_y[0] = *y;
    if(avg_last_y < ADS7846_AVG) avg_last_y++;
    
    /*Sum the x and y coordinates*/
    int32_t y_sum = 0;
    for(i = 0; i < avg_last_y ; i++) {
        y_sum += avg_buf_y[i];
    }

    /*Normalize the sums*/
    (*y) = (int32_t)y_sum / avg_last_y;
}

```

avafinger, 2018-02-06:
Forgot to comment:

> It's not bug, it's a feature, or at least these is the expected behavior. But I see your point. Some objects 
> should keep the input device during the press. I have an idea how to do it. I will add it to the next release.

It is a nice feature, but the intended action was to move the ellipse (in slider) only, at some point it understood it should scroll the page just as if i had removed the finger from the slider and touched the page which did not actually happen, in a small screen these does not happen.

kisvegabor, 2018-02-07:
Thank you for the "keyboard show" the suggestion. I will modify the *demo* to work like this!

8 bit, 16 bit: I consider an important thing to keep the library's API (and the library) well defined to make it truly platform independent. The changing `int` size can cause different behavior on different architectures.  However in your functions  (e.g. indev functions) you can use `int` internally.

kisvegabor, 2018-03-09:
Hi,
I added the mentioned things in v5.1: 
- The demo was update to open/close the keyboard on click
- A new functionality were added to prevent losing of press for some object types: `lv_obj_set_protect(obj,LV_PROTECT_PRESS_LOST`

As the last topics are not related to the original and the functionalities are added in v5.1 I close this issue.

If you have any other questions or ideas just open a new issue for them!

closed 2018-03-09 by kisvegabor

avafinger, 2018-03-09:
@kisvegabor 

**Great news, thank you!**
Yes, i have a few questions and will ask in the pertinent new issue.

Just a few comments, i have changed the source files to include <headers.h> from a different location so i can have my make to build on any platform, by just typing **make lvgl**. I have made Makefile not dependent on anything. Unfortunately, this does not follow your rules and I cannot PR my little changes that will break your way of building lvgl. But i can contribute to a show demo with different display types and spread the news until I catch up with your dev way and do real contributions. 

Keep the good work.

kisvegabor, 2018-03-12:
If you think your changes can make LittelvGL more general (or easier to build on various platforms) please share your thought! The current Makefiles are designed to build with NuttX too. Although we should keep this compatibility, I'm open to make updates in this field.