lvgl/lvgl #90: Found a small problem

chenshijianworkgit, 2018-01-29:
Hi kisvegabor,

1.Lv_win_set_btn_size function repeated statement in lv_win.h file
2.in Lv_sw.h file switch depends on USE_LV_SLIDER, but it has been written as USE_LV_SW
3.In the lv_test_list_1 function, the name of the button scrolls incompletely
4.lv_test_label_2 function, label3 display mode is LV_LABEL_LONG_ROLL, the scroll shows the contents of incomplete

In addition, I have a question：
In the lv_test_line_1 function, line3 is copied from line2, and then call lv_line_set_y_invert (line3, true) function does not achieve the image reversal function, try to enter the parameters as false or true are the same.

kisvegabor, 2018-01-30:
Hi chenshijian,

Thank you for the issue report. I checked them and pushed the updates into the `beta` branch.

However I tested  y_invert in lv_test_line_1 and it works for me. `lv_line_set_y_invert(line3, true);` and `lv_line_set_y_invert(line3, false);` gives me different result. (the red line is vertically mirrored)

chenshijianworkgit, 2018-01-31:
Hi kisvegabor,

Thank you for solving the problem so efficiently
I tested lv_test_line_1 function again is no problem, the previous it may be the cause of eclipse did not compile completed。

In addition, I have a question：
lv_test_label_2 function is used like this?
In the lv_test_label_2 function I added lv_label_set_anim_speed (label3,100) after the lv_label_set_long_mode (label3, LV_LABEL_LONG_ROLL) line but it does not seem to get faster.

kisvegabor, 2018-01-31:
Your welcome!

True, it wasn't working. I also fixed it in `beta`

chenshijianworkgit, 2018-01-31:
Perfect solution

closed 2018-02-03 by kisvegabor