lvgl/lvgl #92: Font icon / ASCII

Zaltora, 2018-01-30:
Hi, i am not sure but after test, we can't use for example `USE_LV_FONT_DEJAVU_20_SUP`  in same time that as `USE_LV_FONT_SYMBOL_20_FEEDBACK`. Create an "#error"  in lv_conf.h to prevent user can be useful ? 

Otherwise, i suggest to add some generic icon to feedback (if it is possible).
-USB 
-Data transfert
-Ethernet
-Localization (Modernize GPS  icon ?)
-Message

 I don't know icon max number with ASCII character ? if some people got idea of generic icon for electronic/embedded device that can be good to add in lvgl default icon. post :)

kisvegabor, 2018-01-30:
Hi,

good idea, however it's true only if LV_UTF8 is disabled. It can be tested for built in fonts but user can add fonts with any range later.

I used this fonts: http://fontawesome.io/icons/

The icons should came with font to ensure easy glyph generation.

The bottle neck of adding new fonts is to store them with UTF8 = 0. It this case the available space is very limited. The current layout is:

| Category | Range | Size | Used|
|--------------|----------|---------|-------|
| Basic |  0xC0..0xCF | 16 | 15 |
| Feedback |  0xD0..0xDF | 16 | 15  |
| File |  0xE0..0xFF | 32 | 20 |

Maybe it worth to join all categories into one?

Zaltora, 2018-01-31:
I got an idea. We can create config on memory chunk and attribute type icon on it in lv_conf.h. user will select what icon he need. That mean we can add many icon, final user will select the icon he need.
A chunk can got 8 icon sized. 
macro will define offset on icon.
e.g
```
#define LV_CHUNK_ASCII_1  ICON_MEDIA
#define LV_CHUNK_ASCII_2  ICON_FEEDBACK
....
```
What did you think?

kisvegabor, 2018-01-31:
Now this mapping is in `lv_symbol_defs.h`. You mean somehow make this configuration to public and allow user pick and place icons? In other words enable icons individually?

Zaltora, 2018-02-01:
yes. you are right.
But instead of enable 1 icon by one icon. Enable a chunk of 8 icon to be easy to configure.
That means with current memory usage on ASCII. That will get 8 memory chunk with 8 icon each.
Example of 8 segment type:
```
ICON_FEEDBACK_LINK
ICON_FEEDBACK_RADIO
ICON_FILE_1
ICON_FILE_2
ICON_MEDIA
....
```
Work with a chunk of 8 icon is good to you ? or you prefer smaller  chunk ?
On UTF-8, i am not sure that this feature is needed and all icon can be enable.

kisvegabor, 2018-02-01:
So we should have a general symbol font (or more with different sizes) with all the symbols and enable parts of it with defines? 
If yes, then creating chunks with more fonts may lead to the same problem as we have  right now. If the chunks are not fully filled then we loose same valuable place.

And if we remove a part from a font's bitmap array then the library will not know where the characters starts. Look at for example `symbol_10_basic_map` (in symbol_10.c) If we remove e.g. the 5th symbol the the 6th will start at the place of the 5th. It seems very difficult write `..._basic_map` for every configuration option.

However if we find a solution  this feature might be useful  with UTF-8 as well to save flash memory by disabling the unused symbols.
 
But is it sure that the symbols need to be grouped somehow? It is only 64 characters. Not so much compared to other "textual" fonts.

Zaltora, 2018-02-05:
I will take a look to see how it is work and do some experiment ( if got time on this ). The font system don't help, that true. Need to think.

I had suggest chunk to be easier ^^ but if is not, it is not necessary.
For example, you got 300 icon in lvgl ( that means 38 chunk of 8 icons except the last)
My idea was to found a way to user to redirect the 8 chunk available with an offset in the good position.
Use chunk of 8 was just to not configure icon one by one with this hypotheses: 8 chunk is enough for the final user application. ( need debate on it )

So the main problem is how to select the icon i want without add icon i don't want with a limited spaces and how organize data icon to achieve this.  
well well.

kisvegabor, 2018-02-05:
True, It would be great to enable user to pick a few icons from a large pool but I don't know what is the right way to do it.

kisvegabor, 2018-02-05:
I think the most forward-looking way is to merge the symbol fonts into one font because it is more general and can might be easier to develop it in the future.

Do you agree?

Zaltora, 2018-02-08:
I  have not seen enough of the icons to advise you. ^^
But if you think, i will be better then to develop custom selection icons by user.
I am in !

kisvegabor, 2018-02-08:
The user have the opportunity to define create his own font icons. So maybe it's not so important make the built-in icons so cusomisable

By the way, the new font converter will be a fancy user friendly online tool on littlevgl.com (instead of the current Python in terminal method)

Zaltora, 2018-02-09:
the tool need be friendly ^^. Otherwise, this features still got a sense. the are the many useful icon. user will add maybe custom, but only few icon. 

Can maybe imagine the new web tool get  selection of all icon from the website ?

kisvegabor, 2018-02-09:
> Can maybe imagine the new web tool get selection of all icon from the website ?

Very good idea!

kisvegabor, 2018-03-09:
Finally I added some links to help user selecting the right font on the font converter tool: 
https://littlevgl.com/ttf-font-to-c-array

closed 2018-03-09 by kisvegabor