lvgl/lvgl #93: button to change screen

Zaltora, 2018-01-30:
I am new with lvgl API and i try to understand the KEYPAD system. I need help.
My hardware: OLED monochrome + 4 buttons.
Button config wanted:  ESC/LEFT/RIGHT/OK

my first thought with lvgl would be to use a button to change the contents of a label (if it is possible ?)
I see some API to retrieve cursor position but not for retrieve button pressed/released ? ( it is exist one ?)
I want to know if it is possible when a do an action, to change screen content ?
I try do not use complex object because the screen is small (128x64 ).

This is the example i test to try the lvgl behaviour:
```c
bool button_read(lv_indev_data_t * data)
{
    //update_adc();
    /*Store the collected data*/
    data->key = LV_GROUP_KEY_ENTER ; 
    data->state = left_button_down ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;
    return false;
}

int screen (void)
{
    //custom input
    lv_indev_drv_t indev_drv2;
    lv_indev_drv_init(&indev_drv2);      
    indev_drv2.type = LV_INDEV_TYPE_KEYPAD;
    indev_drv2.read = button_read;      
    lv_indev_drv_register(&indev_drv2);

    lv_obj_t * scr = lv_obj_create(NULL, NULL);
    lv_scr_load(scr);
//SCREEN 1
    //lv_obj_t * win = lv_win_create(lv_scr_act(), NULL);
    /*Create a Label on the currently active screen*/
    lv_obj_t * label1 =  lv_label_create(lv_scr_act(), NULL);
    lv_obj_t * label2 =  lv_label_create(lv_scr_act(), NULL);

    char buf[32];
    sprintf(buf, "%s%s", SYMBOL_WIFI,SYMBOL_BATTERY_FULL);
    lv_label_set_text(label1,buf);
    lv_obj_align(label1, scr, LV_ALIGN_IN_TOP_RIGHT, 0, 0);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label2, 128, 20);
    lv_label_set_text(label2, "mytext_3h_nope.txt");
    lv_obj_align(label2, scr, LV_ALIGN_IN_BOTTOM_MID, 0, 0);

////SCREEN 2

    lv_obj_t * scr2 = lv_obj_create(NULL, NULL);
    //lv_obj_t * win2 = lv_win_create(scr2, NULL);
    lv_obj_t * bar1 = lv_bar_create(scr2, NULL);
    lv_obj_set_size(bar1, 100, 20);
    lv_obj_align(bar1, scr2, LV_ALIGN_IN_BOTTOM_MID, 0, 0);
    lv_bar_set_value(bar1, 70);
    /*Create a label right to the bar*/
    lv_obj_t * bar1_label = lv_label_create(scr2, NULL);
    lv_label_set_text(bar1_label, "Progress");
    lv_obj_align(bar1_label, scr2, LV_ALIGN_IN_TOP_LEFT, 0, 0);

//group settings
    lv_group_t *group = lv_group_create();
    lv_group_add_obj(group, scr);
    lv_group_add_obj(group, scr2);
}
```

What missing to swap between 2 screen with the button.
In my app, when i click on OK, it will load a file. I need to change screen when i put OK.
If i use next/prev, the label need change and draw the next/prev file name. (need list to do it ?)

lvgl permit to retrieve button action to permit "user app" to do some treatments ?
Or can i register a callback function on button event ?
What i am wrong with my code template ? 

Thank you

kisvegabor, 2018-01-30:
Hi,

Your code example looks good!

You can create a button ad a callback like this:
```c
static lv_res_t btn_ok_clicked(lv_obj_t *btn)
{
    lv_obj_t * label = lv_obj_get_child(btn, NULL); /*The label is the only child*/
    lv_label_set_text(label, "Clicked");
    return LV_RES_OK;   /*The button is not deleted*/
}

...
    /*Create an OK button*/
    lv_obj_t * btn_ok = lv_btn_create(lv_scr_act(), NULL);
    lv_btn_set_action(btn_ok, LV_BTN_ACTION_CLICK, btn_ok_clicked);
    lv_group_add_obj(group, btn_ok);

    /*Add a Label*/
    lv_obj_t * btn_ok_label = lv_label_create(btn_ok, NULL);
    lv_label_set_text(btn_ok_label, "Ok");
```

So when you click the button it will call `btn_ok_clicked`. In this function you can do anything you want, e.g. load a new screen: `lv_scr_load(other_screen)`

You can navigate among the object in a group with LV_GROUP_KEY_PREV/NEXT. And "press" them with LV_GROUP_KEY_ENTER.

LV_GROUP_KEY_LEFT/RIGHT/UP/DOWN are used to set values for example on a slider. UP/RIGHT and LEFT/DOWN usually does the same. (Excent e.g. Text area where they navigate the cursor in 2D)

Just a side note: 
This line: `sprintf(buf, "%s%s", SYMBOL_WIFI,SYMBOL_BATTERY_FULL);`
Can be replaced by: `lv_label_set_text(label1, SYMBOL_WIFI SYMBOL_BATTERY_FULL)` because adjacent strings are automatically concatenate by the compiler. It is also a valid string: `"100%" SYMBOL_BATTERY_FULL`

qwert1213131, 2018-01-31:
lv_scr_load(other_screen) 
That's cool!
I create a task to switch 2 scrrens!
[image]

Zaltora, 2018-01-31:
Create a task from lvgl ?  how you did it ?
It is possible to create a button without draw it ? using style i guess ?

qwert1213131, 2018-01-31:
Hi zaltora ，
Yes.i use the lv'task function to switch screens periodly.
You can use the io interrupt to handle the four hardware buttons😊

kisvegabor, 2018-02-01:
@Zaltora  
As @qwert1213131 said `lv_task` can be used to call a function periodically:
```c
lv_task_create(my_func, 200, LV_TASK_PRIO_MID, NULL ); //Call  my_func in every 200 ms
```

Zaltora, 2018-02-01:
Tell me if i am right,
You manage yourself the button with IRQ to swap between windows  (if it is not a task ) ? you don't use lvgl internal button system management ? 
I want know if it is possible in lvgl to do this without need external management of theses pin. I write a library to manage my four button, i can use it. I wondered if I could do my GUI without.
My idea is to create a virtual button ( style transparent) , with the button i got the callback to do action.

### Feature suggest:  
(I don't no if it is work already on lvgl)
- Button reading by IRQ instead of check the pin in each loop ( enable/disable in lv_conf.h )
example: a button flag can be register by user (format uint8_t )
- Group read button (to read a external GPIO extender for example once)

kisvegabor, 2018-02-01:
Your idea is correct. For your project it is good solution to create transparent buttons (or not transparent while testing) and write something similar to a touch pad driver. 
For example: when the ESC button is pressed you should send a coordinate of the corresponding `lv_btn` object (e.g. the center coordinate).

The `lv_group` system is designed to traverse through a lot of objects and set their values. It seems it is not you need right now.

Zaltora, 2018-02-01:
Okay, that is what I thought. OKay then :)
You think he can be a feature to lvgl to add "generic" callback on physical keypad button action ?  
group read button and irq are improvement suggestion (not related to my issue). seems good features to you ?

kisvegabor, 2018-02-02:
Great :)

What I thought to add a new type of input devices, e.g. LV_INPUT_TYPE_BTN with the following fields:
```c
drv.btn_state = &my_btn_flags;
drv.btn_map = my_btn_map;

...

const lv_point_t my_btn_map[] = {
 {20, 30},   //Assign this point to my_btn_flags & 0x001;
 {50, 30},   //Assign this point to my_btn_flags & 0x002;
 {80, 30},   //Assign this point to my_btn_flags & 0x004;
...
}

```

And then your only task is to refresh the `btn_state` in the read function. Is it would be quite simple, isn't it?

Zaltora, 2018-02-05:
I can't see how will work your system. You assign coordinate to button, then when you push button, it is will trig object on this coordinate ?

Just a suggestion/question about group. it is difficult/usefull to add some callback function on group ?
E.G:
I got a group with 3 object. When i push the button. It is trig a callback that will delete all object in the group. or call a windows directly or change a content from the object group ( title for button, string for label ... )

kisvegabor, 2018-02-05:
> I can't see how will work your system. You assign coordinate to button, then when you push button, it is will trig object on this coordinate ?

Exactly!

>  got a group with 3 object. When i push the button. It is trig a callback that will delete all object in the group. or call a windows directly or change a content from the object group ( title for button, string for label ... )

Do you mean a callback when the object comes in focus?

Zaltora, 2018-02-05:
I want do the inverse, use group mechanics. ( i guess) 
A physical button linked with objects (whatever the position). This physical button, when trig, will changes screen object ( size / text / delete ) with a callback.
To link this physical button, using group system.

e.g: i navigate like the group example from lvgl between object. When i put external button keypad,
it will trig a callback with the obj type or id parameter:  lv_group_cb( obj*/ id ).
In this callback, i can do some mutation  ( size / text / delete ) for the current obj position  or call global thing like swap screen.

kisvegabor, 2018-02-06:
Adding callback to group will be implemented in v5.1. See this:  #94

Zaltora, 2018-02-06:
AAAAAA OK, i did not understand things like that, sry xD

kisvegabor, 2018-02-06:
Which things? I don't understand :D

Zaltora, 2018-02-06:
when i read the 94 issue, what did not think they talk about group and callback group

kisvegabor, 2018-02-06:
Yes, finally we got to this idea there too :)

qwert1213131, 2018-02-06:
I have an idea.
If the screen is partitioned into different functional areas (the equivalent of hidden buttons), when the area touches the action or the physical key is pressed, it triggers a specific function.
I don't know whether it is feasible.
This is more suitable for small screen devices.
[image]

kisvegabor, 2018-02-06:
Hi,

For me the transparent button is more flexible and easier to test. Can you mention a few cases where your suggestion is more useful?

qwert1213131, 2018-02-06:
Hi,
For wearable devices (such as smart watches), their screens are relatively small, and there are not many kinds of widgets displayed. Therefore, dividing the screen into several areas will be easier for operation.
[image]
[image]
[image]

kisvegabor, 2018-02-06:
I see, bu this area can be transparent buttons.

qwert1213131, 2018-02-06:
Yes,Transparent buttons is a great method!
I will have a try,thanks

qwert1213131, 2018-02-06:
void tarnsp_btn_action(void)
		{
				printf("transp button clicked\n");
		}
		static void led_create(lv_obj_t *parent)
		{
		
		
				lv_obj_t * btn1=lv_btn_create(parent,NULL);
				lv_obj_align(btn1,parent,LV_ALIGN_CENTER,0,0);
				lv_obj_set_size(btn1,200,200);
				lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, tarnsp_btn_action);
				static lv_style_t style_btn;
				lv_style_copy(&style_btn, &lv_style_transp);
				lv_btn_set_style(btn1,LV_BTN_STYLE_REL,&style_btn);
				lv_btn_set_style(btn1,LV_BTN_STYLE_PR,&style_btn);
				lv_btn_set_style(btn1,LV_BTN_STYLE_TGL_REL,&style_btn);
				lv_btn_set_style(btn1,LV_BTN_STYLE_TGL_PR,&style_btn);
				lv_btn_set_style(btn1,LV_BTN_STYLE_INA,&style_btn);
		}

[image]

It worked！
cool

kisvegabor, 2018-02-07:
Nice! :)

Wit h interface mentioned [here](https://github.com/littlevgl/lvgl/issues/93#issuecomment-362562923) you could easily assign a physical button to an area on the screen.

EmrahAktas, 2018-02-22:
`#include "lvgl/lvgl.h"

#define REFR_TIME    1000

static void screenTransitionsTask(void * param);

static lv_task_t * refr_task;

static lv_obj_t *screenSynthetic;
static lv_obj_t *main_bg_p;
static lv_obj_t *synyhetic_bg_p;

lv_obj_t *screenCotton;
lv_obj_t *cotton_bg_p;

lv_obj_t *screenWool;
lv_obj_t *wool_bg_p;

LV_IMG_DECLARE(img_bg0);
LV_IMG_DECLARE(img_bg1);
LV_IMG_DECLARE(img_bg2);
LV_IMG_DECLARE(img_bg3);

void createDemoScreens(void){
	
	
	refr_task = lv_task_create(screenTransitionsTask, REFR_TIME, LV_TASK_PRIO_LOW, NULL);
	
	
	screenSynthetic = lv_obj_create(NULL,NULL);
	lv_scr_load(screenSynthetic);
	
	/*Secreen-1*/
	lv_img_create_file("main_bg", img_bg0);
	main_bg_p = lv_img_create(lv_scr_act(), NULL);
	lv_img_set_upscale(main_bg_p, true);
	lv_img_set_file(main_bg_p, "U:/main_bg");
	lv_obj_align(main_bg_p, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);
	
	lv_img_create_file("synyhetic_bg", img_bg1);
	synyhetic_bg_p = lv_img_create(lv_scr_act(), NULL);
	lv_img_set_upscale(synyhetic_bg_p, true);
	lv_img_set_file(synyhetic_bg_p, "U:/synyhetic_bg");
	lv_obj_align(synyhetic_bg_p, screenSynthetic, LV_ALIGN_IN_TOP_LEFT, 0, 0);

	/*Secreen-2*/
	screenCotton = lv_obj_create(NULL,NULL);
	
	lv_obj_align(main_bg_p, screenCotton, LV_ALIGN_IN_TOP_LEFT, 0, 0);

	lv_img_create_file("cotton_bg", img_bg2);
	cotton_bg_p = lv_img_create(lv_scr_act(), NULL);
	lv_img_set_upscale(cotton_bg_p, true);
	lv_img_set_file(cotton_bg_p, "U:/cotton_bg");

	lv_obj_align(cotton_bg_p, screenCotton, LV_ALIGN_IN_TOP_LEFT, 0, 0);

	/*Secreen-3*/
	screenWool = lv_obj_create(NULL,NULL);

	lv_obj_align(main_bg_p, screenWool, LV_ALIGN_IN_TOP_LEFT, 0, 0);

	lv_img_create_file("wool_bg", img_bg3);
	wool_bg_p = lv_img_create(lv_scr_act(), NULL);
	lv_img_set_upscale(wool_bg_p, true);
	lv_img_set_file(wool_bg_p, "U:/wool_bg");

	lv_obj_align(wool_bg_p, screenWool, LV_ALIGN_IN_TOP_LEFT, 0, 0);

	lv_group_t *group = lv_group_create();
	lv_group_add_obj(group, screenSynthetic);
	lv_group_add_obj(group, screenCotton);
	lv_group_add_obj(group, screenWool);
	
	screenTransitionsTask(NULL);
}
static void screenTransitionsTask(void * param){
	
lv_scr_load(screenSynthetic);

}`

Hey, using task I did not manage the change screens.
Any ideas.
Thanks in advance.

kisvegabor, 2018-02-22:
Hi,

I edited you post to better mark up the code. If the code is between  ` ```c   <code block> ``` ` tags then then will be highlighted with C syntax.

It seems you are already on `screenSynthetic`.
```c
	screenSynthetic = lv_obj_create(NULL,NULL);
	lv_scr_load(screenSynthetic);
```
Here is an other thing the notice: now you set `screenSynthetic` in every 100 ms. It has no sense.

What is your goal? How do you want to change between screens?

EmrahAktas, 2018-02-22:
This is because at the first stage screenWool appears on the screen.

I want to change screen in screenTransitionsTask repeatedly like  screenSynthetic->screenCotton->screenWool in task handler.

Zaltora, 2018-02-22:
one basic way:

```c
static uint8_t  i = 0
if(i==0)
    lv_scr_load(screenSynthetic);
else if(i==1)
    lv_scr_load(screenCotton);
else
    lv_scr_load(screenWool);
i++;
if (i == 2)  i = 0;
```
I am not sure if it is work. I mixed my code with yours

```c

static void ScreenPrepare(windows_name_t name)
{
    static lv_obj_t * scr = NULL; //screen handle
    static uint8_t oldname ;
    if(name == oldname)
    {
       return;
    }
    if(scr)
    {
        lv_obj_clean(scr);
        scr = NULL ;
    }
    scr = lv_obj_create(NULL, NULL);
    lv_scr_load(scr);
    oldname = name;
    return ;
}

static void ScreenIcon(void)
{
    ScreenPrepare(window_intro);

    static lv_obj_t * img_bin= NULL;
    img_bin = lv_img_create(lv_scr_act(), NULL);  /*Create an image object*/
    /* Set the image's file according to the current color depth
     * a blue flower picture*/
    lv_img_set_file(img_bin, "P:/img_logoCS2.bin");

    //lv_obj_align(img_bin, img_src, LV_ALIGN_OUT_RIGHT_TOP, 20, 0);     /*Align next to the s'ource image'*/
    lv_obj_set_click(img_bin, true);
    lv_obj_set_drag(img_bin, true);
    lv_obj_set_pos(img_bin, 0, 0);

    //lv_obj_del(img_bin);
    //img_bin = NULL;

}

static void ScreenFileList(void)
{

    ScreenPrepare(window_list);

    lv_obj_t * label1 =  lv_label_create(lv_scr_act(), NULL);
    label_list_file =  lv_label_create(lv_scr_act(), NULL);

    /*Modify the Label's text*/
    lv_label_set_text(label1,SYMBOL_WIFI SYMBOL_BATTERY_FULL);
    lv_obj_align(label1, lv_scr_act(), LV_ALIGN_IN_TOP_RIGHT, 0, 0);
    lv_label_set_long_mode(label_list_file, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label_list_file, 128, 20);
    lv_obj_align(label_list_file, lv_scr_act(), LV_ALIGN_IN_BOTTOM_MID, 0, 0);
    lv_label_set_text(label_list_file, xCurrentGUI.pcCurrentScript);

}

static void ScreenProgress(void)
{
    ScreenPrepare(window_work);

    //lv_obj_t * win2 = lv_win_create(scr2, NULL);
    lv_obj_t * bar1 = lv_bar_create(lv_scr_act(), NULL);
    lv_obj_set_size(bar1, 100, 20);
    lv_obj_align(bar1, lv_scr_act(), LV_ALIGN_IN_BOTTOM_MID, 0, 0);
    lv_bar_set_value(bar1, 70);
    /*Create a label right to the bar*/
    lv_obj_t * bar1_label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_long_mode(bar1_label, LV_LABEL_LONG_BREAK);
    lv_obj_set_size(bar1_label, 128, 20);
    lv_label_set_text(bar1_label, "Push long 'OK' to stop or valid");
    lv_obj_align(bar1_label, lv_scr_act(), LV_ALIGN_IN_TOP_LEFT, 0, 0);
}

static void screenTransitionsTask(void * param){
	
static uint8_t  i = 0
if(i==0)
    ScreenProgress();
else if(i==1)
    ScreenFileList();
else
    ScreenIcon();
i++;
if (i == 2)  i = 0;

}

```

EmrahAktas, 2018-02-22:
Thank you I have implemented your code to mine with minor changes and it works.

kisvegabor, 2018-02-22:
As this issue tend to become a little bit messy I close it.
If you need help with a specific topic, please open a new issue for it.

kisvegabor, 2018-02-25:
To the original topic: in `dev-5.1` branch I added `LV_INDEV_TYPE_BUTTON` as a new input device type. It's read function should look like:
```c
bool button_read(lv_indev_data_t * data)
{
    /*Save the lastly pressed/released button's ID*/
    data->btn = last_button;                  /*0 ... number of buttons-1*/
    data->state = is_pressed(last_button) ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;

    return false;
}
```

And you need to specify a point array where the points corresponds to a button. Pressing of a button will press the screen according the point array. 
point[0] in the array means  button_0  

To set this array:
```c
static lv_points_t points[] = {{20, 30},{ 50, 70},{100, 150}};    /*Points for 3 buttons*/
lv_indev_set_button_points(button_indev,  points);
```

Use `LV_INDEV_POINT_MARKER   1` in lv_conf.h to mark the pressed point on the screen.

Zaltora, 2018-03-01:
It is not clear for usage to me:
For now i use 4 button, and these button are read differently. I got a function to read 2 button in same time and an other to read the last 2 button one by one.
To manage the screen, i call function on event than can be: 
-one button pressed/released
-one button long pressed
-multiple button long pressed

If i understand your new input. I need assign a coordinate for the button, then if the widget button of screen got the button point, they will be linked together ?
It is a cool feature to combine multiple input.

kisvegabor, 2018-03-02:
You can think about the new interface type like Touchpad which can press only given points on the screen. (only one point at once) If you create a button object (lv_btn) on the screen and initialize one of your hardware buttons to press this lv_btn object.  On hardware button  press the button object will be pressed too. 

If you add actions to the button object (`lv_btn_set_action(btn, LV_BTN_ACTION_..., my_func1)`) then these action will can be triggered by the hardware button.

Of course you can use transparent buttons as well.

kisvegabor, 2018-03-09:
This feature is finally released in v5.1.
If you have any question about the new `LV_INDEV_TYPE_BUTTON` please open a new for it.
Thank you for the contribution and the great ideas! :)

closed 2018-03-09 by kisvegabor