lvgl/lvgl #94: [Bug] Focusing object in group within a page doesn't scroll page to show focused object

upbeat27, 2018-01-30:
To replicate:

1. Create a page
2. Create several objects, at least one of which is outside the viewable area of the screen
3. Add those objects to a group
4. Move focus to one of the objects in the group that is outside of the viewable area

Result: 
Page does not scroll and you can't see the focused object.

Proposed solution:
In lv_group_focus_next/lv_group_focus_prev, check if parent is a page. If so, call lv_page_focus.
Additional function lv_group_set_anim_time for use in the call to lv_page_focus.

This is a little bit tricky because the parent of the object will actually be the container that the page created, not the page itself. So it seems like it would have to traverse up through the parents until it hits a page object (or if it doesn't find one, do nothing).

I would implement this fix, but I am unclear at this point how to determine the type of an object.

kisvegabor, 2018-01-31:
As we discussed in #91 there is no way right now to determine the object's type.

Basically the object don't have any knowledge about the others. Except if it uses them (e.g. list uses page, button, label, and image). 

Okay, let's say we know the parent is a page (not surly directly, maybe only the third grand parent). What if later comes a new object type which is derived from page (let's call it lv_page2). It's similar to the original page but have some extra functionality. So this object also need to be notified when "group focus" happens. So in my opinion  something more general is required. Or do you think it is not problem? Maybe never comes an other page-like object.

upbeat27, 2018-01-31:
It does seem that something more general is required in that case.
Assuming no other object would have a similar scrolling nature might end up causing problems down the road.
I'm not sure what would be the solution to this then - I need to get more familiar with the internal workings of the code. You do agree though that the page should scroll itself to reveal any focused object though, correct?

BenQoo, 2018-02-01:
In current design, the scroll-to-focus action could be accomplished by calling `lv_page_focus`. I think it's not a bug. To make `lv_group_focus_next/prev` auto-scroll is not suitable for some situations.

For example, a page contains a list of options, every option consists of:

* `lv_btn` as radio button
* `lv_label` as title text.
* `lv_label` as description text.
* `lv_cont` be the parent of all above objects.

I would like to draw the focus indicator at radio buttons, so `lv_group` only add `lv_btn` objects.
However, when moving focus indicator, not only scroll to the radio button, but the texts must be viewed too.

To keep the pointer of `lv_page` is the simple way instead of traversing parents.

upbeat27, 2018-02-01:
I don't see how your example shows it isn't suitable to auto-scroll.
All lv_page_focus does is make sure the object is in view.
It won't do anything if the object is already in view.
I can't imagine a situation in which you would _not_ want the object that is focused to be in view, even if you also want other objects to be in view.

If focus next/prev doesn't auto scroll the page, where would it be handled? If it is outside of lvgl it needs to be done exactly once, _after_ the input device read has propagated through to change focus. There isn't a user callback to trigger after that point...

kisvegabor, 2018-02-01:
Calling `lv_page_focus` seems a good solution but unfortunately it isn't. You can load `lv_test_group_1()` to play with it. It uses `lv_win_focus()` (same as page focus) and it doesn't work well.  The focus is in late with one object. 

The summary @upbeat27  is correct:

> If focus next/prev doesn't auto scroll the page, where would it be handled? If it is outside of lvgl it needs to be done exactly once, after the input device read has propagated through to change focus. There isn't a user callback to trigger after that point...

Your best choice is to call the *focus* function when you send an *LV_GROUP_KEY_NEXT/PREV*. However at this point still the previous object is in the focus.

As @upbeat27  suggested adding an  "object focus callback" option to the group seems a good idea to me. It should be called when  the input device data (the key) is processed.

BenQoo, 2018-02-02:
> I don't see how your example shows it isn't suitable to auto-scroll. All lv_page_focus does is make sure the object is in view.

To focus a object which is added in `lv_group` can't make sure all expected objects in view.

> If focus next/prev doesn't auto scroll the page, where would it be handled? If it is outside of lvgl it needs to be done exactly once, after the input device read has propagated through to change focus. There isn't a user callback to trigger after that point...

It's correct, sorry I left it out of consideration.

@kisvegabor The "object focus callback" is a flexible way to handle focus action.

kisvegabor, 2018-02-24:
Hi,

I added this feature to `dev-5.1`. 
Set a callback function with `lv_group_set_focus_cb(focus_cb)`
The focus callback should look like:

```c
void focus_cb(lv_group_t * g)
{
  /*Do something*/
}
```

Can you test it?

kisvegabor, 2018-03-07:
Officially added to v5.1.

Thank you for the contribution!

closed 2018-03-07 by kisvegabor