lvgl/lvgl #96: lv_area_is_on function bug

chenshijianworkgit, 2018-01-31:
Hi kisvegabor,

I found the lv_area_is_on function implementation seems a bit of a problem, the comment right-top but the implementation is p.x = a2_p-> x1 should be p.x = a2_p-> x2;

And There are problems with the last two judgments, Because they are always fake：

((a2_p->y1 <= a1_p->y1 && a2_p->y1 >= a1_p->y2) ||       
     (a2_p->y2 <= a1_p->y1 && a2_p->y2 >= a1_p->y2) ||    
     (a2_p->y1 <= a1_p->y1 && a2_p->y2 >= a1_p->y2))

((a2_p->x1 <= a1_p->x1 && a2_p->x1 >= a1_p->x2) || 
       (a2_p->x2 <= a1_p->x1 && a2_p->x2 >= a1_p->x2) || 
       (a2_p->x1 <= a1_p->x1 && a2_p->x2 >= a1_p->x2))

kisvegabor, 2018-01-31:
You are right. Thank you. I don't how could this remain hidden so long because it is an essential function.
May I ask how did you find this bug? 

I think I need to re-think this function. Now it is too complicated. Have a good implemntation idea?
The goal is to give true if the 2 area overlaps each other.

upbeat27, 2018-01-31:
It looks like this is the simplest solution for checking it two rectangles overlap:
```
bool lv_area_is_on(const lv_area_t * a1_p, const lv_area_t * a2_p)
{
return (
(a1_p->x1 < a2_p->x2) && 
(a1_p->x2 > a2_p->x1) && 
(a1_p->y1 > a2_p->y2) && 
(a1_p->y2 < a2_p->y1));
}
```

Found here: [https://stackoverflow.com/questions/306316/determine-if-two-rectangles-overlap-each-other](url)

chenshijianworkgit, 2018-02-01:
[image]

If (zx == x && zy == y), then only one point intersects,Like Figure 1

chenshijianworkgit, 2018-02-01:
Reference program：

`#define ABS(x)  ((x)>0?(x):-(x)) 

bool lv_area_is_on(const lv_area_t * a1_p, const lv_area_t * a2_p)
{
    /*1.Calculate the horizontal and vertical distances between the center points of two rectangular areas。
     *  (Since the two equations are divided by 2, there is no addition here)*/
	int zx = ABS((a1_p->x1+a1_p->x2)-(a2_p->x1+a2_p->x2));
	int zy = ABS((a1_p->y1+a1_p->y2)-(a2_p->y1+a2_p->y2));
	
	/*2.Calculate the width and height of two rectangles
	 *(Since the two equations are divided by 2, there is no addition here)*/
	int x  = (a1_p->x2-a1_p->x1)+(a2_p->x2-a2_p->x1);
	int y  = (a1_p->y2-a1_p->y1)+(a2_p->y2-a2_p->y1);
    
    //3.Determine whether the two equations intersect
	if(zx <= x && zy <= y)
		return true;
	else
		return false;   /*no cover*/
}`

kisvegabor, 2018-02-01:
Thank you for the very deep work and explanation.
As this solution is much clearer then the current I will add this to the code!

kisvegabor, 2018-02-01:
@upbeat27
Thank you for the answer, but it doesn't give `true` if `a1` is large and fully includes `a2` (like *Case 2* on the above image)

upbeat27, 2018-02-01:
@kisvegabor I forgot to adjust the y axis for increasing top to bottom in the above example.
This fixes that and also handles when only the edges overlap.
You can confirm it works for all cases - it is also more simple.

```
bool lv_area_is_on(const lv_area_t * a1_p, const lv_area_t * a2_p)
{
return (
    (a1_p->x1 <= a2_p->x2) && 
    (a1_p->x2 >= a2_p->x1) && 
    (a1_p->y1 <= a2_p->y2) && 
    (a1_p->y2 >= a2_p->y1));
}
```

kisvegabor, 2018-02-01:
@upbeat27 
Looking again your suggestion it also seems correct. @chenshijianworkgit  do you see any problem with that?

chenshijianworkgit, 2018-02-02:
Can not see the method upbeat27 mention any problems, and more simple

kisvegabor, 2018-02-02:
Then I will add @upbeat27 's solution.
Thank you for both of you!

kisvegabor, 2018-03-07:
Fixed in `dev-5.1`.

closed 2018-03-07 by kisvegabor