lvgl/lvgl #97: [Bug?] Input device read doesn't actually process more data if callback returns true

upbeat27, 2018-02-02:
I guess this is a bug, most likely it was just forgotten (should be an easy fix).

User callback for input device read is supposed to be able to alert lvgl there are more input "events" to process. In `indev_proc_task` though, the returned value isn't checked.

I noticed this as I was trying to buffer encoder turn events (you can spin an encoder must faster than the input device read polling interval) and noticed it didn't seem to be working - looking at the code confirmed it.

kisvegabor, 2018-02-03:
Oh, yes its a forgotten thing...
Would you like to fix?

upbeat27, 2018-02-03:
I can probably fix it sometime next week.
It looks fairly straightforward, but if I miss something you can catch it.
I'd probably also wrap up this stuff into a static function while I'm at it:
```
/*Handle reset query if it happened in during processing*/
        if(i->proc.reset_query) {
            i->proc.act_obj = NULL;
            i->proc.last_obj = NULL;
            i->proc.drag_range_out = 0;
            i->proc.drag_in_prog = 0;
            i->proc.long_pr_sent = 0;
            i->proc.pr_timestamp = 0;
            i->proc.longpr_rep_timestamp = 0;
            i->proc.drag_sum.x = 0;
            i->proc.drag_sum.y = 0;
            i->proc.reset_query = 0;
        }
```
as it is an awful lot to repeat more than once.

kisvegabor, 2018-02-04:
Great! Thank you!
Adding a new function to reset those filed is a good idea!

kisvegabor, 2018-03-09:
I close this issue because it is fixed in v5.1.
Thank you for that!

closed 2018-03-09 by kisvegabor