lvgl/lvgl #98: lv_ll bug

chenshijianworkgit, 2018-02-03:
I found lv_ll_rem function to achieve a problem：

If the linked list has only two nodes, then delete the second node, then n_prev point to the first node, 
n_next will point to NULL, then call node_set_prev (ll_p, n_next, n_prev) will result in an error.

lv_ll_rem
	lv_ll_node_t* n_prev = lv_ll_get_prev(ll_p, node_p);
	lv_ll_node_t* n_next = lv_ll_get_next(ll_p, node_p);
	
	node_set_next(ll_p, n_prev, n_next);
	node_set_prev(ll_p, n_next, n_prev);

chenshijianworkgit, 2018-02-03:
I was wrong, there are judgments of the head and tail

chenshijianworkgit, 2018-02-03:
It is best to add a parameter legality judgment

void lv_ll_rem(lv_ll_t  * ll_p, void * node_p) ；       //if  node_p is NULL have a problem
void * lv_ll_get_next(lv_ll_t * ll_p, void * n_act)；   //if  n_actis NULL have a problem
void * lv_ll_get_prev(lv_ll_t * ll_p, void * n_act)；  //if  n_actis NULL have a problem
static void node_set_prev(lv_ll_t * ll_p, lv_ll_node_t* act, lv_ll_node_t* prev)；
static void node_set_next(lv_ll_t * ll_p, lv_ll_node_t* act, lv_ll_node_t* next)

chenshijianworkgit, 2018-02-03:
I find that all the linked list functions are used directly without parameter judgment.

kisvegabor, 2018-02-05:
The discussion continues in #101

closed 2018-02-05 by kisvegabor