lvgl/lvgl #99: Optimize lv_task function

chenshijianworkgit, 2018-02-04:
Hi,

I wanted to optimize its impulses when looking at the lv_task_handler function implementation, because the implementation made it a little puzzling.

The following is the code that I modified to help me see if there is any problem.

1.add lv_ll_ins_pos function

/**
 * Insert a new node in front of the n_act node
 * @param ll_p pointer to linked list
 * @param n_act pointer a node
 * @return pointer to the new head
 */
void * lv_ll_ins_pos(lv_ll_t * ll_p, void * n_act)
{
    lv_ll_node_t* n_new;
    lv_ll_node_t* n_prev;

    if(NULL == ll_p || NULL == n_act) {
        return NULL;
    }

    if(lv_ll_get_head(ll_p) == n_act) {
        n_new = lv_ll_ins_head(ll_p);
    }
    else {
        n_new = lv_mem_alloc(ll_p->n_size + LL_NODE_META_SIZE);
        if(n_new != NULL){
            n_prev = lv_ll_get_prev(ll_p, n_act);
            node_set_next(ll_p, n_prev, n_new);
            node_set_prev(ll_p, n_act, n_new);
        }
    }

    return n_new;
}

2.modify  lv_task_create function

/**
 * Create a new lv_task
 * @param task a function which is the task itself
 * @param period call period in ms unit
 * @param prio priority of the task (LV_TASK_PRIO_OFF means the task is stopped)
 * @param param free parameter
 * @return pointer to the new task
 */
lv_task_t* lv_task_create(void (*task) (void *), uint32_t period, lv_task_prio_t prio, void * param)
{
    lv_task_t* new_lv_task;
    lv_task_t* tmp;

    //Create task lists in order of priority from high to low
    tmp = lv_ll_get_head(&lv_task_ll);
    if(NULL == tmp){
        new_lv_task = lv_ll_ins_head(&lv_task_ll);
    }
    else{
        do{
            if(tmp->prio <= prio){
                new_lv_task = lv_ll_ins_pos(&lv_task_ll, tmp);
                break;
            }
        }while(NULL != (tmp = lv_ll_get_next(&lv_task_ll,tmp)));

        if(tmp == NULL){
            new_lv_task = lv_ll_ins_tail(&lv_task_ll);
        }
    }
    lv_mem_assert(new_lv_task);

    new_lv_task->period = period;
    new_lv_task->task = task;
    new_lv_task->prio = prio;
    new_lv_task->param = param;
    new_lv_task->once = 0;
    new_lv_task->last_run = lv_tick_get();

    return new_lv_task;
}

3. modify lv_task_handler function 

inline void LV_ATTRIBUTE_TASK_HANDLER lv_task_handler(void)
{
    static uint32_t idle_period_start = 0;
    static uint32_t busy_time = 0;
    uint32_t handler_start;
    lv_task_t* tmp;
    int endFlag;

    if(lv_task_run == false) return;
	handler_start = lv_tick_get();

    while(1){
        endFlag = 1;
        LL_READ(lv_task_ll,tmp){
            if(true == lv_task_exec(tmp,tmp->prio)){
                endFlag = 0;
                break;
            }
        }
        if(endFlag) {
            break;
        }
    }

    busy_time += lv_tick_elaps(handler_start);
    uint32_t idle_period_time = lv_tick_elaps(idle_period_start);
    if(idle_period_time >= IDLE_MEAS_PERIOD) {

        idle_last = (uint32_t)((uint32_t)busy_time * 100) / IDLE_MEAS_PERIOD;   /*Calculate the busy percentage*/
        idle_last = idle_last > 100 ? 0 : 100 - idle_last;                      /*But we need idle time*/
        busy_time = 0;
        idle_period_start = lv_tick_get();
    }
}

kisvegabor, 2018-02-04:
May I ask you to create pull request from the changes. Please, also include #98 changes. 
Just to be sure I won't miss anything

chenshijianworkgit, 2018-02-04:
How to submit changes, Here is my operation, the last error：

git clone https://github.com/littlevgl/lvgl.git
git branch -r
git branch
git branch beta
git checkout beta
git push origin beta

Username for 'https://github.com': 601714078@qq.com
Password for 'https://601714078@qq.com@github.com':
remote: Permission to littlevgl/lvgl.git denied to chenshijianworkgit.
fatal: unable to access 'https://github.com/littlevgl/lvgl.git/': The requested
URL returned error: 403

kisvegabor, 2018-02-04:
Pull request are not git but GitHub features. It's about you request me (in this case) to add your branch to lvgl.
Here you will fing how  to do it using GitHub: https://help.github.com/articles/about-pull-requests/

chenshijianworkgit, 2018-02-04:
The code has been submitted to you, and you have a time to look at it.

kisvegabor, 2018-02-24:
Finally I merged the the update in `dev-5.1`.
Can you check them?

chenshijianworkgit, 2018-02-25:
Yes, the New Year holiday, a few days did not see littlevgl library. 
Start working now and watch your update.

kisvegabor, 2018-02-25:
Okay, take your time, and Happy New Year! :)

chenshijianworkgit, 2018-02-28:
Happy New Year：

I'm sorry to reply you so late.

just read the update that you submit, the comment is very clear and easy to understand, thank you very much the perfect solution to my problem.

the lv_area_is_on function does not seem to be updated.

chenshijianworkgit, 2018-02-28:
Just read the lv_area_is_on function submitted in another issue, then this issue can be closed.

kisvegabor, 2018-02-28:
Thank you very much for your help and great ideas!

closed 2018-02-28 by kisvegabor