lvgl/lvgl #103: Rotation feature and more ?

Zaltora, 2018-02-05:
Can be interesting, isn't it ?
Maybe in same time with multi-display support because it is need to get variable height and width.

That we need i guess:
=> rotation input driver ( with gyroscope for example MPU6050 ) 
       to manage rotation with an input
=> a new function to link with driver or for user : lv_screen_rotation(uint16_t degree);
      this function can be call in a callback button.

I was thinking with "color inversion" features too. Because some screen got command to do this. Maybe useful with a button. To prevent user he push without need drawing on screen.

kisvegabor, 2018-02-05:
Hi,

Screen rotation with 90,18 or 270 degree is a good idea, already asked a lot of times. And yes it would be great to came with multi screen support.

About color inversion: it seems to me something that you can do in buttons's action callback if you want.

Zaltora, 2018-02-05:
It is possible to use a button without update it on the screen ?
i got my callback with color inversion but lvgl don't update the button style. ( save bus transaction )

kisvegabor, 2018-02-05:
You need to push the button to trigger its action.

Are you sure your *action* function is called?

Zaltora, 2018-02-05:
oups, missing details.
e.g : for button object in the screen, i push it. but instead  of see in the screen a button pushed, we see screen color inversion.

If i want to develop my question, here it is:
How screen color inversion can be useful with a GUI?
My first thought: trig button/action and do a visual reaction without redraw screen. how configure it in lvgl ?

qwert1213131, 2018-02-06:
Is it necessary to increase the rotation of the created objects?
just as lv_object_rotate(...)

kisvegabor, 2018-02-06:
@Zaltora  Maybe I misunderstood something, but if you assign an action to button (`lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, my_action)`) then in the action function you call your color inverter function (e.g. `color_invert_blinking_3_times()`) However he button need to be redrawn when you press and release it 

@qwert1213131  Rotating objects with 90, 180 or 270 degree can be done easily by modifying your disp_flush function. You can swap which direction you write the data into the frame buffer.

qwert1213131, 2018-02-06:
@kisvegabor 
Is it possible to rotate any degree of object  without modifying the underlying driver?
Such as ：add a rotation function to the object。

kisvegabor, 2018-02-06:
It would be very complicated to implement. And IMHO it's not important for embedded GUIs. 

However a rotation only for images would be useful and  easier to implement.

qwert1213131, 2018-02-06:
I do agree with you!
thanks

kisvegabor, 2018-02-07:
I added the image rotate feature to the list of idea: https://github.com/littlevgl/lvgl/blob/master/docs/TODO_MINOR.md

Turning back t screen orientation change: If you can detect the orientation from (gyroscope or accelerometer) you can modify your  `indev_read()` function too swap coordinates. So you can do o user side easily too. Or have I messed something?

Zaltora, 2018-02-08:
@kisvegabor My point was just an "optimization" using screen HW features.

- Now: Button need be redraw when you touch the screen anyway. That means transfer a lot of data to the screen when user touch the button. the redraw ban be visible on low-end MCU.

- Idea feature: Disable redraw when touch a button and call HW screen color inverter instead. That means 0 data transfers to the screen, and no visible redraw for low-end MCU. Just global color inverter using screen command. User know he touch the button because screen react but without refresh lag of a redraw.

Zaltora, 2018-02-09:
So, i can replace draw_function for each object and do my own callback ?
That sound perfect.

kisvegabor, 2018-02-09:
Exactly!