lvgl/lvgl #106: [BUG] No way to tell which button in button map is "focused" when using KEYPAD indev

upbeat27, 2018-02-07:
Right now you can navigate a button map (or for example, the keyboard object) with the KEYPAD input device. The problem is that you can't tell which button in the button map (or keyboard) is currently going to be sent when you send **LV_GROUP_KEY_ENTER** - that is, no focus is shown.

There should be something similar to `lv_group_set_style_mod_cb` for the button map object to show which button is currently focused.

kisvegabor, 2018-02-08:
Hi,
I tested it with the `lv_test_group_1()` When the keyboard is in focus you can go to the next button with LV_GROUP_KEY_LEFT/RIGHT. The selected key will go to pressed state.

upbeat27, 2018-02-08:
Woops!
I was working with a modified mono theme and accidentally set the pressed and released state to the same style 😿 
Sorry about that.

closed 2018-02-08 by upbeat27

kisvegabor, 2018-02-09:
No problem :)