lvgl/lvgl #107: Vector type font support ?

lukecian, 2018-02-08:
such as freetype font lib integration interface for lvGL.

Pablo2048, 2018-02-08:
I'm not sure if this is what @lukecian mean, but what about mcufont ( https://github.com/mcufont/mcufont )...

kisvegabor, 2018-02-08:
Hi,

it's a very good idea to ensure built-in TTF rendering support. 

Currently I'm reworking the font handling of LittlevGL. With it you will be able to set a callback to get a letter's bitmap. It can be used to load bitmaps from external flash, or even to generate the font bitmap from a TTF when it is required. 
So it will be possible in v5.1 however an example will be required to show that option for the users.

kisvegabor, 2018-03-09:
v5.1  is released where you can add custom bitmap get functions to the fonts. It means you can add a  TTF rendered which generates the fonts real time in a buffer.

For example:
```c
lv_font_t lv_font_dejavu_20 = 
{    
    .unicode_first = 32,    /*First Unicode letter in this font*/
    .unicode_last = 126,    /*Last Unicode letter in this font*/
    .h_px = 20,             /*Font height in pixels*/
    .glyph_bitmap = NULL, /*Bitmap of glyphs. Set freely to identify the font */
    .glyph_dsc = NULL,       /*Description of glyphs. Set freely to identify to any description data */
    .unicode_list = NULL,   /*Every character in the font from 'unicode_first' to 'unicode_last'*/
    .get_bitmap = get_ttf_bitmap,    /*Function pointer to get glyph's bitmap*/
    .get_width = get_ttf_width,  /*Function pointer to get glyph's width*/
    .bpp = 1,               /*Bit per pixel*/
    .next_page = NULL,      /*Pointer to a font extension*/
};
```
 Mcufont as @Pablo2048  suggested could be a great library achieve this!

closed 2018-03-09 by kisvegabor

notnr, 2018-07-05:
@kisvegabor 
Hi，I like truetype font。I always load a TTF file to RAM，then create fonts with different size. emWin+freetype support to realtime create font from TTF file, and very easy to use. I want use littlevgl in my  product, but I don't know whether or not littlevgl support freetype lib.

kisvegabor, 2018-07-06:
Hi,

You can add free type to LittelvGL. To do this you need to manually set up an `lv_font_t` variable need to be initialized like above.

You need to write your own `get_ttf_bitmap` and `get_ttf_width` functions but it seems straightforward using Freetype.

notnr, 2018-07-07:
@kisvegabor 
Thank you very much

fridgerator, 2019-03-13:
@notnr  were you able to get this working?  if so, do you happen to have an example I could see?  Thanks!

puzrin, 2019-04-05:
> You can add free type to LittelvGL. To do this you need to manually set up an `lv_font_t` variable and initialize it like above.
> 
> You need to write your own `get_ttf_bitmap` and `get_ttf_width` functions but it seems straightforward using Freetype.

To use full power of foreign renderers, API should be string-based, not char-based.

With char-based API almost all useful features will be lost - kerning, subpixel smoothing and so on.

kisvegabor, 2019-04-05:
> To use full power of foreign renderers, API should be string-based, not char-based.

With char-based API almost all useful features will be lost - kerning, subpixel smoothing and so on.

Good observation. I was not planning to rework this in this release, but it can be reasonable update later.