lvgl/lvgl #109: idle/shutdown control

Zaltora, 2018-02-10:
I am not sure if this feature exist. My idea is to add a function like this:

```c
//Set a idle control callback to lvgl
lv_idle_set_callback(  sleep_function ) ;

#define LV_SLEEP_TIMEOUT 300  //in second (0 disable)
#define LV_LONG_SLEEP_TIMEOUT  500  //in second  (0 disable)
#define LV_SHUTDOWN_TIMEOUT  600 //in second (0 disable)

//Lvgl screen status list
typedef enum {
   lv_start ;  // initialize screen from a shutdown
   lv_wake_up ;  // just tell screen to wake up + backlight
   lv_sleep ;   // sleep level 1 user (e.g: backlight 20%)
   lv_long_sleep;  // sleep level 2 user (e.g: idle screen + backlight 0%)
   lv_shutdown;  // shutdown screen 
} lv_display_mode_e ;

//custom user sleep function to control backlight or call specific driver command.
display_control( lv_display_mode_e etype)
{
   switch(etype)
   {
   case lv_start:
     //initialize screen here.
   break;
   case lv_sleep:
     // set backlight here  and/or screen command
   break;
   .....
   }
}
```
When the screen sleep or is off,  the first touch with an "indev" will call screen mode function above
and wake up the screen. The second interaction with screen will work as usual.

kisvegabor, 2018-02-12:
Hi,

You can use `lv_indev_get_inactive_time(my_indev)` the get the elapsed time since  last press.

Zaltora, 2018-02-17:
O well, i suppose i will implement the system with this and driver enable.
It is worth to implement a complete sleep system in lvgl ? If the structure is implemented, it's easier for the end user.
If my screen sleep for example, i need disable indev driver before enable it again ( for not that the GUI interacts when the screen is idle)

kisvegabor, 2018-02-17:
True, it will be easier for the user the handler the sleep with your suggested method.
A few things to consider:
1. The feature should the disabled to save RAM and Flash if unused
2. It should be related to a display not the full library (for the planned multi-display support)

Zaltora, 2018-02-18:
1) Totally agree
2) Multi-display support are not present yet but i was thinking like you.

Maybe wait multi-display feature is done to do this? or you got an idea to what var i can use to prepare this update.

kisvegabor, 2018-02-19:
Yes , I thing it would be great to release it with multi display update. 

I highly appreciate your help but I think we should add this feature when the basic multi-display environment is established.

DeonMarais64, 2018-10-15:
Hi,
I think it would be usefull to make provision for timeout from inactivity on active objects only (per input device) in addition to or instead of inactivity of events per se.

kisvegabor, 2018-10-15:
Hi,

So for example, if the user is clicking the background where there are no "active objects" then it shouldn't be counted as an activity?

DeonMarais64, 2018-10-15:
Yes, that is one possibility.  It could be an either or configuration, alternatively provide callbacks for both scnearios. IE Timeout on valid (absence of clicks on active objects only) or don't cares or both ?

kisvegabor, 2018-10-16:
I see, thanks! 
 
Can you mention a case where checking only the "active" objects is more useful than checking only the press in general?

DeonMarais64, 2018-10-16:
Hi Gabor,
Let's assume you wish an application to time out a login after a period of say 5 mins of no activity.
It would in some instances be desirable to do so if the logged in user is actually actve and making meaningfull clicks and not inadvertantly touching a TFT display by mistake or brusing past it. It is true that inadvertent events could take place on active areas but IMO it would never the less be a usefull option to have.

kisvegabor, 2018-10-16:
@DeonMarais64 Thank you for the explanation. Anyway, I agree, it's easy to have an option the select which way you prefer.

Bitblade, 2018-10-31:
I like the proposal as it will support a lot of different use cases. For example, we control the backlight in 255 steps, so we can gradually increase and decrease the brightness when going to sleep / waking up. We would probably use timers started from the callback function to implement this.

However, we would need one addition: 
We use devices like card readers. Activity on those will also wake up the screen in our system. A function to 'hint' lvgl on external activity would make sense. Another use case would be sensors that detect the presence of a user.

kisvegabor, 2018-11-02:
@Bitblade  It also sounds useful! FYI, to handle external events I'd create a dummy `LV_INDEV_TYPE_KEYPAD` input device and set/clear its `state` when something happens. Then you can use `lv_indev_get_inactive_time(NULL)` the get elapsed time since the last activity.

embeddedt, 2018-12-10:
@kisvegabor What would be the general API for this feature? I'm interested in implementing it.

kisvegabor, 2018-12-10:
@embeddedt  There were a lot of ideas here. So let's specify what we would like to achieve. The original suggestion of @Zaltora and/or the suggestion of @DeonMarais64 about the inactivity only with "active" objects?

DeonMarais64, 2018-12-10:
> 
> 
> @embeddedt There were a lot of ideas here. So let's specify what we would like to achieve. The original suggestion of @Zaltora and/or the suggestion of @DeonMarais64 about the inactivity only with "active" objects?

Perhaps we should also specify how it will be achieved, what I mean is the functionality could be achieved via query or callback functions. I have no view on which is preferable, either method could work in principle.

kisvegabor, 2018-12-12:
@DeonMarais64 Now, `lv_indev_get_inactive_time` works like a query method which can be extended with a define like `LV_INDEV_ACTIVITY_CLICKABLE` (or with a better name) to count the only the real click as an activity. However, @Zaltora suggested a callback method and two sleep timeouts (short: e.g. to lower backlight, long: turn off the display). 

I think we can combine the two methods: leave `lv_indev_get_inactive_time` with the `LV_INDEV_ACTIVITY_CLICKABLE` extension and a simple callback system which can use `lv_indev_get_inactive_time` under the hood.

What do you think?

kisvegabor, 2018-12-12:
Great! @embeddedt how does it sounds to you?

embeddedt, 2018-12-12:
@embeddedt It sounds OK. I would like this API to be able to handle another use case, however. I will do my best to explain it below.

A few days ago I ported LittlevGL to an e-reader with an eInk screen. Linux exposes a framebuffer but I also have to use a custom ioctl to actually flush the data to the eInk screen. For performance reasons this can only be done after LittlevGL stops drawing. Up till now I just used a flag variable in the driver that was set whenever one of the draw functions is called. I checked the flag on every iteration in the `while(1)` loop after calling `lv_task_handler` and flushed the data as necessary. Performance was excellent.

I'm wondering how such a feature can be integrated with `lv_indev_set_active_time`. If you are confused or want more details feel free to ask questions.

kisvegabor, 2018-12-13:
@embeddedt So if the whole screen needs to be updated you wait until all chunks are drawn to a buffer and only then flush the frame buffer to the display?

embeddedt, 2018-12-13:
@kisvegabor 

> So if the whole screen needs to be updated you wait until all chunks are drawn to a buffer and only then flush the frame buffer to the display?

That's a fairly accurate explanation. LittlevGL draws to the framebuffer just like any other Linux machine, but the hardware does not update the physical eInk screen with data from the framebuffer until I call the `ioctl`. This is because eInk screens are like a memory LCD: they can hold their state without power.

kisvegabor, 2018-12-13:
@embeddedt I think in this case you can use `lv_refr_set_monitor_cb`. It is called when all parts are refreshed. It tells you the number of refreshed pixels and the rendering time after all the invalidated areas are redrawn. So you can place the `ioctl` here.

embeddedt, 2018-12-13:
Thanks @kisvegabor! It's a much cleaner solution and works just as well!

kisvegabor, 2018-12-14:
@embeddedt Great! 
So still would you like to add this feature? I'm not pushing you, just asking :)

embeddedt, 2018-12-14:
@kisvegabor There is no need to add it :smile: . `lv_refr_set_monitor_cb` is the kind of setup I think the `ioctl` was designed for; my solution was more of a hack because I never saw that `lv_refr_set_monitor_cb` existed.

Also is there a minimum refresh rate for LittlevGL?

kisvegabor, 2018-12-16:
@embeddedt  There is no minimal refresh rate, but there is a maximal: `LV_REFR_PERIOD` in `lv_conf.h`. The refresh task will check with this period if there is something to do or not. 

If you want a minimal refresh rate (e.g. 1 sec) you can create an `lv_task` and call `lv_obj_invalidate(lv_scr_act())`

embeddedt, 2018-12-16:
@kisvegabor Oops, my question was worded ambiguously. What I meant was whether the design of LittlevGL requires a certain refresh rate in order for things to work properly (i.e. showing button toggles). 

I ask because most eInk screens cannot handle high refresh rates without some hacking, which I haven't gotten into yet. Therefore, I lowered `LV_REFR_PERIOD` in `lv_conf.h`. I then noticed that while button actions were still called, I could no longer see the visual toggle effect.

Is there a way to force a refresh even outside of `LV_REFR_PERIOD`?

kisvegabor, 2018-12-16:
@embeddedt As this issue is about "Idle/shutdown control", please open a new issue for these eInk related questions and let's continue there.

kisvegabor, 2019-03-02:
In `v6.0` I'm planning to add these:
- `lv_disp_get_inactive_time()`: get the inactive time from all `indev`s assigned to to a display (`v6.0` will have multi-display support)
- `lv_disp_trig_activity()`: externally add an activity event
- The inactivity will be counted by the display, not the individual `indev`s
- `lv_indev_get_inactive_time` will be removed

kisvegabor, 2019-04-04:
I've added the feature listed in the above post into `dev-6.0`.

Thank you very much for the ideas and the discussion.

closed 2019-04-04 by kisvegabor