lvgl/lvgl #110: [Question] How to handle key touch inside Touch callback (LV_INDEV_TYPE_POINTER)

avafinger, 2018-02-11:
Hi @kisvegabor ,

I have an LCD Panel with four KEYs, it usually send the key touch as key event, just like pointer touch event.
I would like to handle these Key touch, for example, KEY_HOME to fire some event or switch to a next button group or some thing like.

These LCD touch screens are usually made for Android and the button touch and panel/screen touch events come together.

If i understand **lvgl** (still learning), the **LV_INDEV_TYPE_POINTER** callback expects a (X,Y) Press or Release touch and do some processing.

I know there is a callback to handle Keypad or keyboard, but these key touches must be handled on the pointer callback due to limitations explained above.

How can i handle these KEY touches?

For reference, the touch driver in question is the **it7260_ts**, but you can consider that every LCD panel out there will drive like that, so it is the same way for **FT5X** and others.

Sample Data for ONE touch on screen and 4 button touch:

```
Event: time 1518386282.744546, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 26
Event: time 1518386282.744546, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 574
Event: time 1518386282.744546, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 65
Event: time 1518386282.744546, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 4
Event: time 1518386282.744546, -------------- SYN_REPORT ------------

Event: time 1518386283.848481, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1518386283.848481, -------------- SYN_REPORT ------------

Event: time 1518386285.728500, type 1 (EV_KEY), code 217 (KEY_SEARCH), value 1
Event: time 1518386285.728500, -------------- SYN_REPORT ------------

Event: time 1518386285.972513, type 1 (EV_KEY), code 217 (KEY_SEARCH), value 0
Event: time 1518386285.972513, -------------- SYN_REPORT ------------

Event: time 1518386286.936452, type 1 (EV_KEY), code 158 (KEY_BACK), value 1
Event: time 1518386286.936452, -------------- SYN_REPORT ------------

Event: time 1518386287.124452, type 1 (EV_KEY), code 158 (KEY_BACK), value 0
Event: time 1518386287.124452, -------------- SYN_REPORT ------------

Event: time 1518386288.132389, type 1 (EV_KEY), code 172 (KEY_HOMEPAGE), value 1
Event: time 1518386288.132389, -------------- SYN_REPORT ------------

Event: time 1518386288.348387, type 1 (EV_KEY), code 172 (KEY_HOMEPAGE), value 0
Event: time 1518386288.348387, -------------- SYN_REPORT ------------

Event: time 1518386289.232396, type 1 (EV_KEY), code 139 (KEY_MENU), value 1
Event: time 1518386289.232396, -------------- SYN_REPORT ------------

Event: time 1518386289.436390, type 1 (EV_KEY), code 139 (KEY_MENU), value 0
Event: time 1518386289.436390, -------------- SYN_REPORT ------------

```

kisvegabor, 2018-02-12:
Hi,

You can manage this only with two input devices: one for touch pad and an other for the buttons.  

I suggest you to process all the events in the keypad's  `indev_read` function and jut save the last x, y coordinates and events. And in touchpad's read function just use the stored values.

kisvegabor, 2018-03-09:
In `v5.1` you can assign a user data pointer to the input devices. This feature is by @CapnBry who have similar issue. Check the thread here: #115

closed 2018-03-09 by kisvegabor