lvgl/lvgl #111: Using DMA2D with LVGL STM32F746NG-DISCOVERY

KarolAbego, 2018-02-14:
According to this post -> https://littlevgl.com/blog/8/embedded-gui-on-stm32-discovery-board-with-littlev-graphics-library, I want to use instead of DMA, DMA2D controller. I already wrote a function which prototype look like this void lcd_draw(uint32_t pos_x, uint32_t pos_y, uint32_t size_x, uint32_t size_y, uint32_t *ptr_to_buffer). It works fine with e.g bitmaps.  Of course I'm using VDB_DOUBLE buffering mode. Is it possible to write function "ex_dips_flush", that will fit requirements of "lcd_draw" function?

kisvegabor, 2018-02-14:
For a `disp_flush` function which runs in the background you can use your `lcd_draw` but you need to inform the library when `lcd_draw` is ready (using `lv_flush_ready()`).
So you need a callback for example in a "DM2D ready interrupt".

Is it already working with LV_VDB_DOUBLE = 0? Don't forget that LV_VDB_DOUBLE = 0 is also double buffered just the parallel rendering and flushing is not supported.

KarolAbego, 2018-02-14:
Yes there is a flag to inform lv_flush_ready() function when drawing is finished. But I was hoping you to explain me how can I calculate the x y points and sizes of each rectangle in "ex_disp_flush". I'm basing on your stm32 discovery board example.

void ex_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
	/*Return if the area is out the screen*/
	if(x2 < 0) return;
	if(y2 < 0) return;
	if(x1 > TFT_HOR_RES - 1) return;
	if(y1 > TFT_VER_RES - 1) return;

	/*Truncate the area to the screen*/
	int32_t act_x1 = x1 < 0 ? 0 : x1;
	int32_t act_y1 = y1 < 0 ? 0 : y1;
	int32_t act_x2 = x2 > TFT_HOR_RES - 1 ? TFT_HOR_RES - 1 : x2;
	int32_t act_y2 = y2 > TFT_VER_RES - 1 ? TFT_VER_RES - 1 : y2;

	x1_flush = act_x1;
	y1_flush = act_y1;
	x2_flush = act_x2;
	y2_fill = act_y2;
	y_fill_act = act_y1;
	buf_to_flush = color_p;

	lcd_draw(?, ?, ?, ?, buf_to_flush);
	lv_flush_ready();

}
And I don't really know, but I really want to :-) - How to write the function ex_disp_flush in that way that all question marks that I indicated above in lcd_draw will fit the requirements. Because when I randomly chosen e.g lcd_draw(put x_1flush, y1_flush,act_x2, act_y2, buf_to_flush) it draws me something and animmates it but everything doesn't look like it should. I mean that I see moving rectangles from annimation example, which was working fine with lcd_draw_pixel.

kisvegabor, 2018-02-14:
I see. For me `lcd_draw(flush_x1, flush_y1, flush_x2 - flush_x1 + 1, flush_y2 - flush_y1 + 1, buf_to_flush)`

For size parameters I added +1 because for me the size of x1 = 3, x2 = 4 is 2.I think it is handled the same in DMA2D too.

KarolAbego, 2018-02-14:
Thank you it works much better, but it still corrupts objects. I attached the pictures to this post. Have you got any idea what I'm doing wrong. Also I attached the body of lcd_draw function and DMA2D configuration.
[functions and screens.zip](https://github.com/littlevgl/lvgl/files/1724088/functions.and.screens.zip)
Thank You again in advance.

kisvegabor, 2018-02-14:
It seems only the bottom part of the object is corrupted.

Firstly please use LV_VDB_DOUBLE = 0 to reduce the number of possible issues.
If it doesn't help increase LV_VDB_SIZE how it effects the corrupted area.

kisvegabor, 2018-02-14:
The most simple case to use settings like this:

#define LV_VDB_SIZE         (480*20)  /*20 lines for VDB*/
#define LV_VDB_ADR          (0)          		   /*Keep the VDB in the internal RAM*/

New suggestion: in lcd_draw(): add some delay (100 ms to be sure) before lv_flush_ready(). Now the library immediately thinks that flush is ready and the content of the VDB is not required anymore. So it starts to overdraw it.

KarolAbego, 2018-02-14:
It doesn't help at all. Even with 1 second delay beetween lcd_draw function and lv_flush_ready with LV_VDB_SIZE 480*20.

kisvegabor, 2018-02-14:
Okay, then try something more simple: just create a button and see how it is flushed

```c
    lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_t * label = lv_label_create(btn1, NULL);
```

kisvegabor, 2018-02-14:
Or a more basic test: without LittlevGL just try the lcd_draw this way:
```c
lv_color_t test_bitmap[100 * 50];
int i;
for(i = 0; i < 100* 50; i++) test_bitmap[i] = LV_COLOR_RED;

lcd_draw(50, 80, 100, 50, test_bitmap);

```

kisvegabor, 2018-02-16:
Could you solve the issue?

closed 2018-02-23 by KarolAbego

r2com, 2018-06-18:
I just have a quick question, there are 3 main functions as I see, for those who plan use buffering & GPU:
ex_disp_flush
ex_mem_blend
ex_mem_fill

So is the working function which is called all the time is ex_disp_flush? In other words, if user tries to display even a single character or item, the ex_disp_flush called automatically in background, correct?

what about other two functions? when are they typically called? only manually by user?

kisvegabor, 2018-06-19:
Hi,

`flush` is called when the final image is is rendered to "copy" it to the screen

`blend/fill` are used to accelerate rendering. So they are used e.g. to fill a memory area with red (which memory after drawings will be flushed using by `flush` function) 

If you have other questions please open a new issue because it's not related to this topic.