lvgl/lvgl #112: can't display "Celsius Degree" symbol   while UTF8=1

gilloup, 2018-02-14:
In lv_conf.h, I have this line : 
#define LV_TXT_UTF8             1 

In some file , I have written:
lv_label_set_text(label_temperature, " Temperature: 20 °C");

But, the symbol "°" isn't displayed.
Did I make something wrong?

Thanks by advance,

Gilles Peltier

kisvegabor, 2018-02-14:
Hi,

Because this symbol is located among the Latin Supplement characters you need to enable `USE_LV_FONT_DEJAVU_20_SUP` (replace the size with yours)

Or if you're using your own fonts instead of the built-ins be sure this character is included.

gilloup, 2018-02-15:
I have enabled these characters : 
USE_LV_FONT_DEJAVU_40_SUP= 1 and USE_LV_FONT_DEJAVU_60_SUP=1.
But, I can't display the "°" character. I investigate.

kisvegabor, 2018-02-15:
And have you modified font of the text in its style? If not, then what is the default font?

gilloup, 2018-02-15:
Here is the code:

 /////////////// label temperature  ////////////////////////////////////
      /*Create a new style*/
      static lv_style_t style_txt_temp;
      lv_style_copy(&style_txt_temp, &lv_style_plain);
      style_txt_temp.text.font = &lv_font_dejavu_60;
      style_txt_temp.text.letter_space = 2;
      style_txt_temp.text.line_space = 1;
      style_txt_temp.text.color = LV_COLOR_WHITE;

      /*Add a label to the button*/
      lv_obj_t * label_temp = lv_label_create(lv_scr_act(), NULL);
      lv_obj_set_style(label_temp, &style_txt_temp);
      lv_label_set_text(label_temp, " Temp.: 20 ºC");
      lv_obj_align(label_temp, NULL, LV_ALIGN_CENTER, -5, 0);

In  "lv_font.h":
#define LV_FONT_DEFAULT        &lv_font_dejavu_40

kisvegabor, 2018-02-15:
It seems correct.

Are you sure the character coding of your IDE is UTF-8?
Check it with this:
```c 
const uint8_t buf[] = {"20 ºC"};
printf("%02x %02x %02x %02x %02x %02x %02x\n", buf[0], buf[1], buf[2], buf[3], buf[4], buf[5], buf[6], buf[7]);
```

It works for me and I got this result: `32 30 20 c2 ba 43 00`  
The ` c2 ba` is the UTF8 encoded `º` character

gilloup, 2018-02-15:
Bravo!
I use Eclipse in "nominal" mode.
The original encoding is Cp1252. I changed with "UTF8". It works fine, now...

closed 2018-02-15 by gilloup

kisvegabor, 2018-02-15:
Great! :)