lvgl/lvgl #113: Activate long pressed action for button

Masi2017, 2018-02-15:
Hi I need help for using button long pressed action. I created 3 buttons and added them into a group where I added my input device which is 4 hardware keys ( Arrow and OK keys). I defined two actions for each key, a long pressed action and click action. whenever I pressed a key no matter for how long, just button action is run. I couldn't find any example for that, I appreciated it if some one could help me. 
Regards MM

kisvegabor, 2018-02-15:
Hi,

there is no opportunity to trigger the long press action of a button with a hardware button which sends `LV_GROUP_KEY_ENTER`.

However it seems a useful feature. Constructive ideas are welcome.

Masi2017, 2018-02-15:
Thanks Kisvegbor
I was wondering how I can activate this action ? Is there any way to activate the long press action for the button through software?

kisvegabor, 2018-02-15:
You can get and call the long press action:
```c
lv_action_t lpr_action =  lv_btn_get_action(btn, LV_BTN_ACTION_LONG_PR);     /*Get the action*/
lpr_action(btn);   /*Call it*/
```

Masi2017, 2018-02-15:
Thanks so much

kisvegabor, 2018-02-15:
Your welcome! 
I hope it helps until your suggested feature won't be implemented.

Masi2017, 2018-02-15:
sure

kisvegabor, 2018-02-24:
Hi,

I added this feature to `dev-5.1` branch. Can you  test it?

I don't how your `keypad_read` function looks like, but for this feature it should be like:
```c
bool keypad_read(lv_indev_data_t * data)
{
    data->key = last_key;
    data->state = last_state;                           /*LV_INDEV_STATE_PR/REL*/
    
    return false;
}
```

Masi2017, 2018-02-26:
Hi Gabor,

Sure I’ll test that and let you know if it works or not.