lvgl/lvgl #115: Can't have multiple indev drivers of same type?

CapnBry, 2018-02-16:
I'm working on a project that is going to use the Linux uninput layer as their input drivers for keyboard, touchscreen, and/or mouse. At the very least I need both a LV_INDEV_TYPE_POINTER and LV_INDEV_TYPE_KEYPAD both of which call the same base code, my indev/uinput driver I've written. Creating the driver was rather straightforward and works for a single device, but if I wanted to create multiple copies of the device, say one for the keyboard and one for the mouse, then `lv_indev_drv_t.read` callback doesn't know which device file it is reading from or shift states for the keyboard.

I can work around this by copypasting the read code multiple times and having each one operate on a different internal state struct, but I'd have to know in advance how many devices I'd want to support and create separate wrappers for each of them. Is there any chance `lv_indev_drv_t` can get an `void *opaque` that can be passed to the read() callback either as a new parameter or in `lv_indev_data_t` ? That would allow a driver to maintain a state between calls and have multiple devices at the same time running on the same code. For example, a EV_REL device (a mouse) and an EV_ABS device (a touchscreen) without creating a restriction that says I can create exactly one mouse and exactly one touchscreen and exactly one keyboard with read functions like this:
```
bool uinput_read_mouse(lv_indev_data_t *data) { return uinput_read(data, &uinput_mouse_state); }
bool uinput_read_touch(lv_indev_data_t *data) { return uinput_read(data, &uinput_touch_state); }
bool uinput_read_keyboard(lv_indev_data_t *data) { return uinput_read(data, &uinput_keyboard_state); }
```

kisvegabor, 2018-02-17:
Hi,
It's a good idea and it's possible to add a pointer to a user defined data (as you suggested) to `lv_indev_data_t` because it won't change the API.  

I will add it in 5.1. Hopefully it will be released in a few weeks.
Meanwhile you can add this feature yourself very simply. Just add a new field to `lv_indev_data_t` and initialize it.

kisvegabor, 2018-02-17:
As was thinking about it is's a little bit more complicated. The user data should be added to `lv_indev_drv_t` too and set in `lv_indev_data_t` in `indev_proc_task` (lv_indev.c)

CapnBry, 2018-02-19:
Yup thanks, it isn't complicated at all, just four lines of code really. This is all it takes to for my driver to be able to have the same read function for keyboard and mouse, as well as having the driver emit calibrated pointer coordinates.
```
diff --git a/lv_hal/lv_hal_indev.c b/lv_hal/lv_hal_indev.c
index d7a5904..f25e813 100644
--- a/lv_hal/lv_hal_indev.c
+++ b/lv_hal/lv_hal_indev.c
@@ -46,6 +46,7 @@ void lv_indev_drv_init(lv_indev_drv_t *driver)
 {
     driver->read = NULL;
     driver->type = LV_INDEV_TYPE_NONE;
+    driver->priv = NULL;
 }

 /**
@@ -107,6 +108,7 @@ bool lv_indev_read(lv_indev_t * indev, lv_indev_data_t *data)
     bool cont = false;

     if(indev->driver.read) {
+        data->priv = indev->driver.priv;
         cont = indev->driver.read(data);
     } else {
         memset(data, 0, sizeof(lv_indev_data_t));
diff --git a/lv_hal/lv_hal_indev.h b/lv_hal/lv_hal_indev.h
index d8852d7..5485eb0 100644
--- a/lv_hal/lv_hal_indev.h
+++ b/lv_hal/lv_hal_indev.h
@@ -48,12 +48,14 @@ typedef struct {
         uint32_t key;          /*For INDEV_TYPE_KEYPAD*/
     };
     lv_indev_state_t state; /*LV_INDEV_EVENT_REL or LV_INDEV_EVENT_PR*/
+    void *priv;             /*'lv_indev_drv_t.priv' for this driver*/
 }lv_indev_data_t;

 /*Initialized by the user and registered by 'lv_indev_add()'*/
 typedef struct {
     lv_hal_indev_type_t type;                       /*Input device type*/
     bool (*read)(lv_indev_data_t *data);        /*Function pointer to read data. Return 'true' if there is still data to be read (buffered)*/
+    void *priv;                                    /*Opaque pointer for driver's use, passed in 'lv_indev_data_t' on read*/
 }lv_indev_drv_t;

 struct _lv_obj_t;
```

kisvegabor, 2018-02-19:
Thank you!
To merge it may I ask you to create a pull request (for `dev-5.1` branch )?

closed 2018-02-24 by kisvegabor