lvgl/lvgl #117: Drop down list

Masi2018, 2018-02-21:
I have some question and I would appreciate if someone can help me :
1- I need to know what should be done if the LCD's height is not long enough to display all the options in the drop down list ? Now once I scroll among the options I can not see those are in the bottom of my list ( since the lcd height is not long enough but I expected that once I go down the options will move up in the way that user can see the last item which cursor point to)
2- Is there any object to display a list of Items and more than one option can be selected? 
Thanks in advance

upbeat27, 2018-02-21:
1. You make the dropdown height whatever you want it to be, the list inside will scroll to reveal more options. With touch, the scrolling is simply done with your finger. With keyboard, the scrolling did have a bug that I fixed where focus didn't move along with the highlighted entry; it probably isn't merged into master.
2. lv_list can contain a list of toggle-able buttons. I think you can also use checkboxes inside of a page.

Masi2018, 2018-02-22:
Thank you very much, since I am using keyboard that's why I have problem and focus doesn't move along the highlighted entry. May I ask when this bug will be resolved?

kisvegabor, 2018-02-22:
Hi,

The auto-scrolling of a  drop down list  is It is already fixed in the `beta` branch.

The set a fix height for a drop down list you should use: `lv_ddlist_set_fix_height(ddlist, height)

`

closed 2018-02-24 by kisvegabor