lvgl/lvgl #119: Action after clicking an image

Anarakofona, 2018-02-22:
Hi,
I'm new in littlevgl.
Is there a way to make action after clicking in an image, or a container?

I know, it is possible to make it clickable with **lv_obj_set_click(... , ...)** but I don't know how to associate an action.

With **button** or **buttonMatrix**, I can use **lv_btnm_set_action( ..... )** or **lv_btn_set_action( ..... )** , but with an image or a container, what can I do?

Please help me please

Thank you very much

kisvegabor, 2018-02-22:
Hi,
Right now there no simple way to add action to an image. 
`lv_icon` is a planned new object type which will make exactly this.

Now you can create a  button with transparent style and create an image on it.
Now for the image `lv_obj_set_click(img, false)` because the background button should be clicked.

Anarakofona, 2018-02-22:
Thank you for your answer,
 How to add a personalised image to a button? or Should  I just place the image above the transparent button?

kisvegabor, 2018-02-22:
```c
    lv_obj_t * btn1 =  lv_btn_create(lv_scr_act(), NULL);
    lv_btn_set_fit(btn1, true, true);                        /*Automatically increase the size to involve all children (now only the image)*/
    lv_btn_set_style(btn1, LV_BTN_STYLE_REL, &lv_style_transp_tight);   /*Transparent style for released button*/
    lv_btn_set_style(btn1, LV_BTN_STYLE_PR, &lv_style_transp_tight);   /*Transparent style for pressed button*/
    lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, lpr_test);
    lv_obj_t * img = lv_img_create(btn1, NULL);                         /*Create the image on 'btn1'*/
    lv_obj_set_click(img, false);                                  /*Don't let to click to the image (click the background button instead)*/

```

Anarakofona, 2018-02-22:
Oh my God it work,

Thank you very much

I have another question but I wanna create a new issue because may be it help another persons

Thank you, Thank you

kisvegabor, 2018-02-22:
Great! :)
Okay, I will check it!

closed 2018-02-23 by kisvegabor

TheGrovesy, 2018-07-03:
I just had this same issue but already came to this conclusion. What is the reason for allowing an object to be clickable when there is no way to set an action? Seems redundant to me!

kisvegabor, 2018-07-04:
"clickable" is a more general attribute the "actions". For example:
- It is required to scroll an `lv_page`. You need to be able to click on the scrollable part to scroll it.   
- If you have multiple layers (or a pop-up) then the layer's background should be clickable too. It ensures not the objects on the back layer will be clicked.

TheGrovesy, 2018-07-04:
Thanks for the explanation :+1: