lvgl/lvgl #120: [QUESTION] RESIZING FREELY IMAGE

Anarakofona, 2018-02-22:
Hi,

I want to resize an image but : 
   - it is repeated like mosaïque if the new size is big than the original size
   - it is clipped if the new size is little than the original size

Is there a way to resize the image freely (not scale) without these probleme?

Thank you very much

kisvegabor, 2018-02-22:
Hi,

for example you have a 100x100 image. And you want it to occupy 200x200 place but keep the original 100x100 sized image on it?

Anarakofona, 2018-02-22:
No,

For example I have 100 x 100 original image, but I want tu put it on the screen with 75 x 75 or 36 x 36 or 25 x 25 or 144 x 144  or (free sizing it) size. My goal is to resize the image and put it on the screen as resized.

I don't know if it's the same thing that you say in your commen above.

I tried to use lv_obj_set_width(img, 25) and lv_obj_set_height(img, 25) but it will clipp the image.

Thank you

kisvegabor, 2018-02-22:
I see. Resizing the image to arbitrary size (upscale downscale) in real time is not implemented right now.

You should resize your image with an online or offline tool and use the resized image.
http://picresize.com/

Anarakofona, 2018-02-22:
Okay, thank you for your help!!

:)

Thank you

kisvegabor, 2018-02-22:
Your welcome :)

closed 2018-02-22 by kisvegabor

kisvegabor, 2023-06-07:
Yes, you can [scale the images](https://docs.lvgl.io/8.3/widgets/core/img.html) in the latest version of LVGL but note that it's a quite performance hungry operation.