lvgl/lvgl #121: Problem with antialiasing

KarolAbego, 2018-02-22:
Hello.
I have another problem with antialiasing. When antialiasing is off text in objects e.g buttons is bigger that it should (I'm currently testing library with tutorials that You've provided), so I assume that everything is done well. I've tested this issue with two functions put_pixel wich draws graphics pixel by pixel and draw_lcd wich uses DMA2D and copies buffer into the memory. In both cases result is the same.
When I turn on the antialiasing data on screen corrupts (random lines are visible) but text look better. 
I've been trying to fix that by changing VDB_SIZES as well in internal and external memory with FMA. And it doesn't gave any effects. 
The pictures of my problem are attached here -> [Antialias.zip](https://github.com/littlevgl/lvgl/files/1748222/Antialias.zip)

Thank You in advance.

kisvegabor, 2018-02-22:
Hi,

Anti aliasing makes font to half sized. So the size changing is normal. Learn more here: https://littlevgl.com/basics#drawing-and-rendering

Have you tried the anti aliased version with pixel based frame buffer write instead of DMA2D?

KarolAbego, 2018-02-22:
Yes I've tried (also I read this artcile) it but the results are the same as in the pictures that are provided in attachment. Also I've tried to draw simple rectangles etc. and drawing functions works fine. Have you got any idea why it corrupts.

kisvegabor, 2018-02-22:
I see.

Please attache your disp_flush() function and lv_conf.h

By the way, what about your previous issues? Can you give some feedback in the related topics?

KarolAbego, 2018-02-22:
Flickering of screen was a hardware problem. But about corrupting data on screen I can't say anything because I think that could be connected with antialiasing problem. 
disp_flush and lv_conf are here -> [disp_flush.zip](https://github.com/littlevgl/lvgl/files/1748331/disp_flush.zip)

kisvegabor, 2018-02-22:
LV_VDB_SIZE is already in pixel count so you don't need to multiply it by 4

To eliminate some possible issues try this way.

```c
#define LV_VDB_SIZE         (480*10)  /*Size of VDB in pixel count*/
#define LV_VDB_ADR          (0)            /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically  into RAM)*/

/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
 * The flushing should use DMA to write the frame buffer in the background*/
#define LV_VDB_DOUBLE       (0)       /*1: Enable the use of 2 VDBs*/
#define LV_VDB2_ADR         (0)       /*Place VDB2 to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/
```

And can I see the pixel based disp_flush too? Firstly we should try with that.

KarolAbego, 2018-02-22:
It doesn't helped, moreover I can see more straight lines on the objects ->
[image], pixel based disp_flush is the same as in Your's porting tutorial and it's here ->
[put_pixel.zip](https://github.com/littlevgl/lvgl/files/1748400/put_pixel.zip)

kisvegabor, 2018-02-22:
To be honest I have no idea what can cause an issue like that. To examine test 2 things please:
1. Set LV_VDB_SIZE to 480*2 (minimum size) and turn on LV_ANTIALAIS
2. Delete (comment) all `lv_label_create` from the example and check with LV_ANTIALIS and LV_FONT_ANTIALIAS too. Try with LV_VDB_SIZE (480*2) and with larger too (e.g. 480*20). Maybe the letter drawing cause the issue somehow

KarolAbego, 2018-02-23:
I've tried those two test that you suggested, but it doesn't helped at all, also screen width is 480 px and this function works fine (it was tested in previous project), and also it was tested with example that You've provided in previous issue about using DMA2D:

lv_color_t test_bitmap[100 * 50];
int i;
for(i = 0; i < 100* 50; i++) test_bitmap[i] = LV_COLOR_RED;

lcd_draw(50, 80, 100, 50, test_bitmap);

and it drew normal rectangle on the screen without any issues.

kisvegabor, 2018-02-23:
Just to be sure: LCD_WIDTH id 479 right?

I'm working on v5.1 release in which I reworked anti-aliasing too. Please read this post (#125) and update to `dev-v5.1` 

Anyway your issue is very strange. LittlevGL already worked on a lot of STM machine but your driver seems correct.

kisvegabor, 2018-02-23:
If it doesn't help a good way to exclude a lot of possible issues is:
1. Set LV_VDB_SIZE (40 * 480)
2. Load the demo application (lv_examples/lv_apps/demo): `demo_create()`
3. Check if the GUI is wrong or not. If yes:
4. When `disp_flush` is called first time it will refresh the first 40 lines. Print the value of the buffer you got and I will compare it with mine.

KarolAbego, 2018-02-26:
Ok, I've tried with 5.1 version of library and problem still exsists, also I made test according to your tips and in result I've recived buffer which is equal to 0x20013A50 (screen horizontal res 480 vertical 272 it's stm32f7 discovery board). Because I'm confused and I don't really know how the demo should look like I've attached picture of it to this issue -> 
[image] (The letters of keyboard are barely visible).

Thank You for your answer in advance.

kisvegabor, 2018-02-26:
Hi,
The demo should look like this:
[image]

The colors are clearly wrong for you. The color of the "header" should be: 0xff487fb7. (Alpha = 0xFF, Red = 0x48, Green = 0x7F, Blue = 0xB7). At least the top byte should be 0xFF for you to because it is hard coded.

Are you using the unmodified version of the LittlevGL? Isn't there any "hacks"?

KarolAbego, 2018-02-26:
Sorry I made a mistake I assume that I can take this buffer by commenting out the lv_flush_ready() function, am I right? So now the buffer is A:0xFF R:0x01 G:0x3C B:0x20. I'm useing unmodiied version there are no "hacks" in there.

kisvegabor, 2018-02-26:
You should just write a `while(1)` to the end of `disp_flush()` (to be sure it will be called only once and you see the first lines) and print the buffer before it. Or check it with a debugger.

I have an STM32F746 Discovery board but I didn't have tie to try it yet. I think its the same as your board. If you sent your project I would examine it.

KarolAbego, 2018-02-26:
So now get result like that ->
[image]. I didn't modified library at all. The full name of board is STM32F746G-DISCO.

KarolAbego, 2018-02-26:
And project is here -> [ltdc.zip](https://github.com/littlevgl/lvgl/files/1758410/ltdc.zip) 
Thank You very much for Your time and help.

kisvegabor, 2018-02-26:
I have the same board. 
I downloaded the project but its quite "empty". I can't see e.g. LTDC initialization and it halts with WWDG interrupt at start-up. Can you send the project where you tested the disp_flush() function? (Of course delete all proprietary data/code first)

KarolAbego, 2018-02-26:
Hmm I've send you working project I mean this is the code from which pictures/photos comes. Ok here is the project exported via SystemWorkbench for STM32 (Eclipse)->
[ltdcExported.zip](https://github.com/littlevgl/lvgl/files/1758728/ltdcExported.zip). LTDC initialisation is inside lvgl_init() function like dma2d sdram and fmc inits.

kisvegabor, 2018-02-26:
Sorry, it was my bad. I thought lvgl_init() is lv_init(). Now I found the code, thank you.

However for same reason it still not working for me. It "dies" in the start-up code.
[image]

Any idea?

KarolAbego, 2018-02-26:
[ltdc.zip](https://github.com/littlevgl/lvgl/files/1758802/ltdc.zip) to exclude enviroment problems try to load this .elf file

kisvegabor, 2018-02-26:
It works with your elf file.

My compiler flags:
-mcpu=cortex-m7 -mthumb -mfloat-abi=soft -mfpu=fpv5-sp-d16 -DSTM32 -DSTM32F7 -DSTM32F746NGHx -DSTM32F746G_DISCO -DDEBUG -DSTM32F746xx -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/ampire480272" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/n25q512a" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/s5k5cag" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/mfxstm32l152" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/CMSIS/device" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/n25q128a" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/ts3510" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/st7735" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/rk043fn48h" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/HAL_Driver/Inc/Legacy" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/ov9655" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/otm8009a" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/STM32746G-Discovery" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/Common" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/HAL_Driver/Inc" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/stmpe811" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/wm8994" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Fonts" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/ft5336" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/inc" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/ft6x06" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/exc7200" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/st7789h2" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Log" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/adv7533" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/ampire640480" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/CMSIS/core" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc/Utilities/Components/mx25l512" -I"/home/kisvegabor/stm-eclipse-workspace/ltdc" -O0 -g3 -Wall -fmessage-length=0 -ffunction-sections -c

Linker flags:
-mcpu=cortex-m7 -mthumb -mfloat-abi=soft -mfpu=fpv5-sp-d16 -T"/home/kisvegabor/stm-eclipse-workspace/ltdc/LinkerScript.ld" -Wl,-Map=output.map -Wl,--gc-sections -lm

KarolAbego, 2018-02-26:
Did you imported project that I've sent, or made new project and copied files into it?

kisvegabor, 2018-02-26:
I have imported your project. However I needed to change the FPU settings to be able to compile it.

With Floating Point ABI = hard I got a lot of similar errors:

/home/kisvegabor/Ac6/SystemWorkbench/plugins/fr.ac6.mcu.externaltools.arm-none.linux64_1.15.0.201708311556/tools/compiler/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: error: ltdc.elf **uses VFP register arguments**, lvgl/lv_core/lv_group.o does not
/home/kisvegabor/Ac6/SystemWorkbench/plugins/fr.ac6.mcu.externaltools.arm-none.linux64_1.15.0.201708311556/tools/compiler/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file lvgl/lv_core/lv_group.o

kisvegabor, 2018-02-26:
When I let  the original "hard" value I get the errors but the flag is -mfloat-abi=soft in the command line.

KarolAbego, 2018-02-26:
-mcpu=cortex-m7 -mthumb -mfloat-abi=hard -mfpu=fpv5-sp-d16 -T"C:\Projekty\SW-workspace\ltdc\LinkerScript.ld" -Wl,-Map=output.map -Wl,--gc-sections -lm here is my linker config

KarolAbego, 2018-02-26:
-mcpu=cortex-m7 -mthumb -mfloat-abi=hard -mfpu=fpv5-sp-d16 -DSTM32 -DSTM32F7 -DSTM32F746NGHx -DSTM32F746G_DISCO -DDEBUG -DSTM32F746xx -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/ampire480272" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/n25q512a" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/s5k5cag" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/mfxstm32l152" -I"C:\Projekty\SW-workspace\ltdc/CMSIS/device" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/n25q128a" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/ts3510" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/st7735" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/rk043fn48h" -I"C:\Projekty\SW-workspace\ltdc/HAL_Driver/Inc/Legacy" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/ov9655" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/otm8009a" -I"C:\Projekty\SW-workspace\ltdc/Utilities/STM32746G-Discovery" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/Common" -I"C:\Projekty\SW-workspace\ltdc/HAL_Driver/Inc" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/stmpe811" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/wm8994" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Fonts" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/ft5336" -I"C:\Projekty\SW-workspace\ltdc/inc" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/ft6x06" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/exc7200" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/st7789h2" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Log" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/adv7533" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/ampire640480" -I"C:\Projekty\SW-workspace\ltdc/CMSIS/core" -I"C:\Projekty\SW-workspace\ltdc/Utilities/Components/mx25l512" -I"C:\Projekty\SW-workspace\ltdc" -O0 -g3 -Wall -fmessage-length=0 -ffunction-sections -c and here is compilator config

KarolAbego, 2018-02-26:
But they should be the same, are you using System Workbench or eclipse with ARM plugin?

kisvegabor, 2018-02-26:
I'm using System Workbench.

KarolAbego, 2018-02-26:
Ok we can try this way. 
Create new blank project and chose configuration for stm32f746G board. 
Copy files *exclude .PROJECT and .CPROJECT
Add Preprocessor PATHS Project->Properties->C/C++ General->Paths and Symbols->Symbols:

DEBUG
USE_FULL_ASSERT
TRACE
OS_USE_TRACE_SEMIHOSTING_DEBUG
HSE_VALUE=25000000
STM32F746xx

In C/C++ Build->Settings->MCU GCC Compiler->Includes
include all folders
and in MCU GCC Linker->General->linkerScript(-T) Browse... chose LinkerScript.ld

kisvegabor, 2018-02-26:
I did this. Started a new empty project. Now FPU is hard.

But I got `undefined reference to `lv_tick_inc'`. I'm trying to solve it.

KarolAbego, 2018-02-26:
Ok maybe try to rebuild indexer, in this IDE it's common problem, and it occured me so many times.

kisvegabor, 2018-02-26:
I was the first thing I tried :D

kisvegabor, 2018-02-26:
Ohh, no object files are generated from lvgl foder...

kisvegabor, 2018-02-26:
I needed to create a New source folder with eclipse (Right click->New->Source folder) and copy lvgl there.

So I was able to flash the board and it is wrong for me too. Now I can start to examine it. Thank you for help.

KarolAbego, 2018-02-26:
Ok no problem, but I really aprreciate Your help and time as well.

kisvegabor, 2018-02-26:
I pushed an update to `dev-5.1`. Can you test it?

The alpha channel of 24 bit colors was uninitialized in a color creator function. Therefore sometimes fully transparent pixels were in buffer in disp_flush(). And the LTDC controller drawn transparent pixels hence the black background become visible.

KarolAbego, 2018-02-26:
Yes I will. But will dev-5.1 solve this issues ? Can you provide me chaged code ;p?

kisvegabor, 2018-02-26:
In lv_color.h update lv_color_mix()

```c
static inline lv_color_t lv_color_mix(lv_color_t c1, lv_color_t c2, uint8_t mix)
{
    lv_color_t ret;
    ret.red =   (uint16_t)((uint16_t) c1.red * mix + (c2.red * (255 - mix))) >> 8;
    ret.green = (uint16_t)((uint16_t) c1.green * mix + (c2.green * (255 - mix))) >> 8;
    ret.blue =  (uint16_t)((uint16_t) c1.blue * mix + (c2.blue * (255 - mix))) >> 8;
#if LV_COLOR_DEPTH == 24
    ret.alpha = 0xFF;
#endif

    return ret;
}

```

KarolAbego, 2018-02-26:
Oh finally, It looks like it should thank you very much. Curently I'm using 5.1 ver of lvgl library I think we can close this issue finally.
Thank You very much!

closed 2018-02-26 by KarolAbego

kisvegabor, 2018-02-26:
Great! :)
I also have  to say thank you for your persistence!

kisvegabor, 2018-03-04:
I would like to add an STM32F7 Discovery project to the `lv_boards` repo. 
Will you mind if I use  the project you sent as a starting point?

KarolAbego, 2018-03-05:
Yes, of course feel free to use/share it.