lvgl/lvgl #122: Question :  lv_list_focus  : btn parameter

gilloup, 2018-02-22:
Hi,

Considering this list:     
         	lv_obj_t *list = lv_list_create(settings_screen, NULL);
                lv_list_add(list,  SYMBOL_SAVE, 	   "Setting 1", 	list_btn_1_action);
		lv_list_add(list, SYMBOL_SAVE,            "Setting 2", 	list_btn_2_action);
		lv_list_add(list, SYMBOL_SAVE, 	    "Setting 3", 	list_btn_3_action);

And the following function:	 
lv_list_focus (lv_obj_t *btn, bool anim_en);

How to force the focus to the 1st element of the list ( or how to set "btn" parameter) ?

Thanks by Advance
Gilles Peltier

kisvegabor, 2018-02-22:
Hi,

Use the return value of `lv_list_add` which is the create list element (button).

```c
lv_obj_t * btn1 = lv_list_add(list, SYMBOL_SAVE, "Setting 1", list_btn_1_action);
lv_obj_t * btn2 = lv_list_add(list, SYMBOL_SAVE, "Setting 2", list_btn_2_action);
lv_obj_t * btn3 = lv_list_add(list, SYMBOL_SAVE, "Setting 3", list_btn_3_action);

lv_list_focus (btn1, true);
```

gilloup, 2018-02-23:
Thanks Gabor.

Gilles Peltier

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

kisvegabor, 2018-02-23:
Your welcome

closed 2018-02-23 by kisvegabor