lvgl/lvgl #123: getting error in PC simulator

mithunkamat, 2018-02-22:
21:54:18 **** Incremental Build of configuration Debug for project pc_simulator ****
Info: Internal Builder is used for build
gcc -o pc_simulator "lv_drivers\\display\\R61581.o" "lv_drivers\\display\\SSD1963.o" "lv_drivers\\display\\ST7565.o" "lv_drivers\\display\\fbdev.o" "lv_drivers\\display\\monitor.o" "lv_drivers\\indev\\FT5406EE8.o" "lv_drivers\\indev\\XPT2046.o" "lv_drivers\\indev\\keyboard.o" "lv_drivers\\indev\\mouse.o" "lv_examples\\lv_apps\\benchmark\\benchmark.o" "lv_examples\\lv_apps\\benchmark\\img_benchmark_bg.o" "lv_examples\\lv_apps\\demo\\demo.o" "lv_examples\\lv_apps\\demo\\img_bubble_pattern.o" "lv_examples\\lv_apps\\sysmon\\sysmon.o" "lv_examples\\lv_apps\\terminal\\terminal.o" "lv_examples\\lv_apps\\tpcal\\tpcal.o" "lv_examples\\lv_tests\\lv_test_group\\lv_test_group.o" "lv_examples\\lv_tests\\lv_test_obj\\lv_test_obj.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_bar\\lv_test_bar.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_btn\\lv_test_btn.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_btnm\\lv_test_btnm.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_cb\\lv_test_cb.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_chart\\lv_test_chart.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_cont\\lv_test_cont.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_ddlist\\lv_test_ddlist.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_gauge\\lv_test_gauge.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_img\\img_flower_icon.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_img\\lv_test_img.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_kb\\lv_test_kb.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_label\\lv_test_label.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_led\\lv_test_led.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_line\\lv_test_line.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_list\\lv_test_list.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_lmeter\\lv_test_lmeter.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_mbox\\lv_test_mbox.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_page\\lv_test_page.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_roller\\lv_test_roller.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_slider\\lv_test_slider.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_sw\\lv_test_sw.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_ta\\lv_test_ta.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_tabview\\lv_test_tabview.o" "lv_examples\\lv_tests\\lv_test_objx\\lv_test_win\\lv_test_win.o" "lv_examples\\lv_tests\\lv_test_stress\\lv_test_stress.o" "lv_examples\\lv_tests\\lv_test_theme\\lv_test_theme.o" "lv_examples\\lv_tutorial\\0_porting\\lv_tutorial_porting.o" "lv_examples\\lv_tutorial\\10_keyboard\\lv_tutorial_keyboard.o" "lv_examples\\lv_tutorial\\1_hello_world\\lv_tutorial_hello_world.o" "lv_examples\\lv_tutorial\\2_objects\\lv_tutorial_objects.o" "lv_examples\\lv_tutorial\\3_styles\\lv_tutorial_styles.o" "lv_examples\\lv_tutorial\\4_themes\\lv_tutorial_themes.o" "lv_examples\\lv_tutorial\\5_antialiasing\\lv_tutorial_antialiasing.o" "lv_examples\\lv_tutorial\\6_images\\img_red_flower.o" "lv_examples\\lv_tutorial\\6_images\\lv_tutorial_images.o" "lv_examples\\lv_tutorial\\7_fonts\\lv_tutorial_fonts.o" "lv_examples\\lv_tutorial\\7_fonts\\ubuntu_40_ascii.o" "lv_examples\\lv_tutorial\\7_fonts\\ubuntu_40_cyrillic.o" "lv_examples\\lv_tutorial\\8_animations\\lv_tutorial_animations.o" "lv_examples\\lv_tutorial\\9_responsive\\lv_tutorial_responsive.o" "lvgl\\lv_core\\lv_group.o" "lvgl\\lv_core\\lv_indev.o" "lvgl\\lv_core\\lv_obj.o" "lvgl\\lv_core\\lv_refr.o" "lvgl\\lv_core\\lv_style.o" "lvgl\\lv_core\\lv_vdb.o" "lvgl\\lv_draw\\lv_draw.o" "lvgl\\lv_draw\\lv_draw_rbasic.o" "lvgl\\lv_draw\\lv_draw_vbasic.o" "lvgl\\lv_hal\\lv_hal_disp.o" "lvgl\\lv_hal\\lv_hal_indev.o" "lvgl\\lv_hal\\lv_hal_tick.o" "lvgl\\lv_misc\\lv_anim.o" "lvgl\\lv_misc\\lv_area.o" "lvgl\\lv_misc\\lv_circ.o" "lvgl\\lv_misc\\lv_color.o" "lvgl\\lv_misc\\lv_font.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_10.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_10_cyrillic.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_10_latin_ext_a.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_10_latin_ext_b.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_10_sup.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_20.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_20_cyrillic.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_20_latin_ext_a.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_20_latin_ext_b.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_20_sup.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_30.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_30_cyrillic.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_30_latin_ext_a.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_30_latin_ext_b.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_30_sup.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_40.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_40_cyrillic.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_40_latin_ext_a.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_40_latin_ext_b.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_40_sup.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_60.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_60_cyrillic.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_60_latin_ext_a.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_60_latin_ext_b.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_60_sup.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_80.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_80_cyrillic.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_80_latin_ext_a.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_80_latin_ext_b.o" "lvgl\\lv_misc\\lv_fonts\\dejavu_80_sup.o" "lvgl\\lv_misc\\lv_fonts\\symbol_10_basic.o" "lvgl\\lv_misc\\lv_fonts\\symbol_10_feedback.o" "lvgl\\lv_misc\\lv_fonts\\symbol_10_file.o" "lvgl\\lv_misc\\lv_fonts\\symbol_20_basic.o" "lvgl\\lv_misc\\lv_fonts\\symbol_20_feedback.o" "lvgl\\lv_misc\\lv_fonts\\symbol_20_file.o" "lvgl\\lv_misc\\lv_fonts\\symbol_30_basic.o" "lvgl\\lv_misc\\lv_fonts\\symbol_30_feedback.o" "lvgl\\lv_misc\\lv_fonts\\symbol_30_file.o" "lvgl\\lv_misc\\lv_fonts\\symbol_40_basic.o" "lvgl\\lv_misc\\lv_fonts\\symbol_40_feedback.o" "lvgl\\lv_misc\\lv_fonts\\symbol_40_file.o" "lvgl\\lv_misc\\lv_fonts\\symbol_60_basic.o" "lvgl\\lv_misc\\lv_fonts\\symbol_60_feedback.o" "lvgl\\lv_misc\\lv_fonts\\symbol_60_file.o" "lvgl\\lv_misc\\lv_fonts\\symbol_80_basic.o" "lvgl\\lv_misc\\lv_fonts\\symbol_80_feedback.o" "lvgl\\lv_misc\\lv_fonts\\symbol_80_file.o" "lvgl\\lv_misc\\lv_fs.o" "lvgl\\lv_misc\\lv_ll.o" "lvgl\\lv_misc\\lv_mem.o" "lvgl\\lv_misc\\lv_task.o" "lvgl\\lv_misc\\lv_templ.o" "lvgl\\lv_misc\\lv_trigo.o" "lvgl\\lv_misc\\lv_txt.o" "lvgl\\lv_misc\\lv_ufs.o" "lvgl\\lv_objx\\lv_bar.o" "lvgl\\lv_objx\\lv_btn.o" "lvgl\\lv_objx\\lv_btnm.o" "lvgl\\lv_objx\\lv_cb.o" "lvgl\\lv_objx\\lv_chart.o" "lvgl\\lv_objx\\lv_cont.o" "lvgl\\lv_objx\\lv_ddlist.o" "lvgl\\lv_objx\\lv_gauge.o" "lvgl\\lv_objx\\lv_img.o" "lvgl\\lv_objx\\lv_kb.o" "lvgl\\lv_objx\\lv_label.o" "lvgl\\lv_objx\\lv_led.o" "lvgl\\lv_objx\\lv_line.o" "lvgl\\lv_objx\\lv_list.o" "lvgl\\lv_objx\\lv_lmeter.o" "lvgl\\lv_objx\\lv_mbox.o" "lvgl\\lv_objx\\lv_objx_templ.o" "lvgl\\lv_objx\\lv_page.o" "lvgl\\lv_objx\\lv_roller.o" "lvgl\\lv_objx\\lv_slider.o" "lvgl\\lv_objx\\lv_sw.o" "lvgl\\lv_objx\\lv_ta.o" "lvgl\\lv_objx\\lv_tabview.o" "lvgl\\lv_objx\\lv_win.o" "lvgl\\lv_themes\\lv_theme.o" "lvgl\\lv_themes\\lv_theme_alien.o" "lvgl\\lv_themes\\lv_theme_default.o" "lvgl\\lv_themes\\lv_theme_material.o" "lvgl\\lv_themes\\lv_theme_mono.o" "lvgl\\lv_themes\\lv_theme_night.o" "lvgl\\lv_themes\\lv_theme_templ.o" "lvgl\\lv_themes\\lv_theme_zen.o" main.o -lSDL2main -lSDL2 
C:/Program Files (x86)/CodeBlocks/MinGW/bin/../lib/gcc/mingw32/4.9.2/../../../libmingw32.a(main.o):main.c:(.text.startup+0xa7): undefined reference to `WinMain@16'
collect2.exe: error: ld returned 1 exit status

Above is the error I am getting while trying to compile in eclipse. I had done all the steps described on the website.

kisvegabor, 2018-02-22:
Hi,

are you sure you are using 32 bit MingGw and 32 bit SDL (or both 64 bit)?

helezh, 2018-02-22:
hi! happy new year!  this err i have get. you can change "main" to "winmain"

helezh, 2018-02-22:
i have posted the issure,you try to find it

helezh, 2018-02-22:
issue #53

memen45, 2018-03-07:
Ran into the same issue this afternoon: undefined reference to winmain, after a clean install of eclipse and mingw. Finally fixed it by adding 'mingw32' to libs in project properties (there should be three listed under libs and the order matters: mingw32, SDLmain, SDL, is the correct order). Maybe this should be added to the 'getting started' - steps that are provided on the site?

kisvegabor, 2018-03-07:
Than you! I will add it to the website as you suggested.

kisvegabor, 2018-03-07:
Is it required only for 32 it MinGW?

memen45, 2018-03-07:
As far as I can see, I installed the 64 bit version of mingw and I copied the 64bit versions of all the files to the mingw installation directory. I found the fix here: [https://stackoverflow.com/a/48328724](url)

Besides this, I ran into some other problems before I could get it started: 

 - The directories stated on the get started page [https://littlevgl.com/pc-simulator](url) are not completely correct. Steps 3 and 4 are: 
3. Copy ..._mingw32/include/SDL2 folder to C:/MinGW/include
4. Copy ..._mingw32/lib/ content to C:/MinGW/lib
While with my clean install it worked after changing it to "C:\...\mingw-w64\x86_64-7.2.0-posix-seh-rt_v5-rev1\mingw64\x86_64-w64-mingw32\*" (with * for both include and lib directories). It is confusing, since the include and lib directories are also present in "C:\...\mingw-w64\*", but will not work there.

 - Clean build was not possible -> In the .cproject file there is a 'cleanCommand' that is set to 'rm -rf', however this does not work on windows (fix: install coreutils or replace 'rm -rf' with something supported by windows like 'del').

Not sure whether to open a new issue for this or not, but since you are looking at the steps from the site anyways, I thought I'd post it here as well.

kisvegabor, 2018-03-08:
>  I found the fix here: https://stackoverflow.com/a/48328724

I edited the link because is was misformatted: `[https://stackoverflow.com/a/48328724](url)` -> `https://stackoverflow.com/a/48328724`
GitHub recognizes the links as link. If you want to name link you can use: `[my_link](www.something.com)`

Thank you for the explanation. I check again the paths on the Windows machine. I used `C:\MinGW\x86_64-w64-mingw32\*` ('*' also for lib and include)

So it's really not the same as on the webpage but also not the same as yours because the `\x86_64-7.2.0-posix-seh-rt_v5-rev1\mingw64` part is missing for me.

memen45, 2018-03-08:
That is weird, but maybe adding the last directory name "...\x86_64-w64-mingw32\*" would clarify this sufficiently (as that is the same in both our cases.

kisvegabor, 2018-03-09:
Yeah it's really weird. Anyway I added the `\x86_64-w64-mingw32\` part as you suggested.

Thank you very much!

kisvegabor, 2018-03-12:
As this issue seems solved I close it. 
If you find other build related issues on PC please open an issue in the pc_simulator repository instead.

Thank you sharing you experiences! :)

closed 2018-03-12 by kisvegabor