lvgl/lvgl #126: hal: what is the difference of flush and map in driver?

Vonger, 2018-02-24:
Hi,
  The code is beautiful. 👍 
  I find in display driver, flush and map function are almost same code, just flush have a lv_flush_ready() called at end. So what is the different and when should use map/flush?

Thanks.

kisvegabor, 2018-02-24:
Hi,

Thank you! :)

Basically they do the same job. The only difference is `disp_map()` is used with `LV_VDB_SIZE = 0` to  put  images to the display directly. `lv_disp_flush()` is used with `LV_VDB_SIZE != 0` when buffered drawing is enabled to put the buffer's content to the display. 

So both of them put a pixel map to the display. You can ask then why is there two functions to do the same task? The answer is because they have very different purpose and because `lv_flush_ready() ` is required in one case and not in the other. And I wasn't able to give an intuitive name which describes the two functionality at once. :P

If you have a good function name idea for the combined `disp_map` and `disp_flush` I'm yery open for it!

Vonger, 2018-02-25:
I see :) Thank you!
Another question I find yesterday
When I run it on my embed linux(580MHz MIPS), I find it is not that smooth as your youtube video when switch between "Write" and "List" and "Chart" label in the demo. First I thought it might be my device is render slow, I reduce its pixel width 16bits but still no luck..
After I run in my PC linux find same issue, it is 3.5GHz CPU, so I think it should not be the CPU speed problem. What the problem it is possible? I think the video device should be much slower than my device.

MenglongWu, 2018-02-25:
this is lvgl efficiency too [SLOW]. I try lvgl on  linux base on /dev/fb,the effect as you.

fix. don't  use AA （anti aliasing）. edit lv-conf.h .

ps. youtube video demo is do not use AA.

I also try ucgui on linux base on fb 90fps.
but lvgl only 20fps.
cpu 450MHz,ddr2 90MB

I dont try lvgl on stm32 with AA.
flush/map is  different is that flush cpu draw point,map use dma.

if use fb (on use space compile lvgl.so ) MAP mode unuse.it become more SLOW,about 10fps.

if lvgl run on linux driver that you can use map,use DMA draw point.
###################
和你一样我在linux环境基于/dev/fb[效果也很差]。

解决方法，不要用AA

顺带说一句。youtube的演示里就是没使用AA，也没有使用贴图。

我没有在stm32上具体移植过lvgl，不知道AA效果。

fluah和map模式的差别是，一个使用cpu绘图，一个使用dma。
用户空间使用 /dev/fb编译成lvgl.so，没必要使用map模式，效率低下，10fps

如果在驱动层可采样map模式，效果可能会好一些

来自 MEIZU MX 

时间：周日 2月25日 08:59
收件人：littlevgl/lvgl <lvgl@noreply.github.com>
抄送：Subscribed <subscribed@noreply.github.com>

Vonger, 2018-02-25:
It is, after disable AA it is much faster. But it is not beautiful anymore :)

kisvegabor, 2018-02-25:
HI,

Please try the `dev-5.1` branch and read this issue: #125 . I made a lot of update on anti-aliasing. Now it needs only very small extra rendering time. I'm very curious about your test results!

The LV_VDB_SIZE = 0 configuration is:
*  slower 
* lower quality (flickeering)
* not supports some features(e.g. opacity)

So you should use the  buffered (LV_VDB_SIZE != 0) configuration. The unbuffered is useful only in some special cases. 

Increasing the size of VDB also makes rendering faster. Play with it!

kisvegabor, 2018-03-12:
As the speed related thought were discussed about v5.1 in #125 I close this issue.
If you have other speed related questions pleas open a new issue for it!

closed 2018-03-12 by kisvegabor