lvgl/lvgl #127: mouse for embed linux without SDL

Vonger, 2018-02-25:
Hi, 

[mouse_hid.zip](https://github.com/littlevgl/lvgl/files/1755679/mouse_hid.zip)

This is a driver for embed linux without touch screen, using USB mouse(I am).
I have tested it works well, but one problem is, I did not find a way to display the cursor :)

kisvegabor, 2018-02-25:
Hi,
Thank you! :)

You can use the 'lv_indev_set_cursor()' function:
```c
    mouse_init();
    lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);          /*Basic initialization*/
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read = mouse_read;         /*This function will be called periodically (by the library) to get the mouse position and state*/
    lv_indev_t * mouse_indev = lv_indev_drv_register(&indev_drv);

    LV_IMG_DECLARE(img_cursour);
    lv_obj_t * cursor_obj =  lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src(cursor_obj, &img_cursour);

    lv_indev_set_cursor(mouse_indev, cursor_obj);

```

Vonger, 2018-02-25:
Thank you, I checked 5.0 and 5.1 version, but both of them do not have "lv_img_set_src"

kisvegabor, 2018-02-25:
Sorry, I forget to say that the example is from v5.1.
lv_img_set_src in v5.1. Check here: https://github.com/littlevgl/lvgl/blob/dev-5.1/lv_objx/lv_img.c#L115

Have you really pulled the dev-5.1 branch? Wasn't it any git erreor?

Vonger, 2018-02-25:
I have load the cursor by the old way :) 
but the background of the cursor is not transparent, is there anyway to do that?

kisvegabor, 2018-02-26:
On the old way (v5.0) you can use a background color defined in LV_COLOR_TRANSP (lv_conf.h) to make pixels transparent. The default is totally green (#00ff00).

To the image converter tool you need to assign a `-t` flag.

Have you find the `lv_img_set_src` function?

Vonger, 2018-02-26:
Yes, I find lv_img_set_src in 5.1, still trying to compile it :")
When I try the tranparent in 5.0, it does not work, added -t flag and set background to lime.
I am trying to find the problem now

Vonger, 2018-02-26:
find the problem :)
It is in lv_draw_vbasic.c/lv_draw_rbasic.c compare the value of pixel is not matched LV_COLOR_LIME=0xff00ff00 but in the image it is 0x0000ff00...
The is an update broken, I make a quick fix:

lv_color_t transp_color = LV_COLOR_TRANSP;
transp_color.alpha = 0;

then it works!!!
this can close. 👍

kisvegabor, 2018-02-27:
Oh thank, I will fix it in v5.1.

If you can't compile v5.1 feel free to open an issue for it.

closed 2018-02-27 by kisvegabor

kisvegabor, 2018-03-01:
I added your mouse HID device to` lv_drivers` in `dev-5.1` branch.
Thanks you again!