lvgl/lvgl #128: Waterfall drawing

zhilulingguan, 2018-02-26:
Hi
Thank you for offering us an open source GUI library for free.
I found this GUI library can not display images continuously. For example, I need to show a waterfall map, or a GIF animation. I did not expect to realize it in any way. I want to know if there are any easy ways to do this?
Thank you!

kisvegabor, 2018-02-26:
Hi,

Your welcome! I hope you like it! :)

You can create an image object (lv_img) and continuously change its image with a task.
I recommend you to update to `dev-5.1` branch because there are a lot of new stuffs there for images and it will make your work easier. Read more here: #125 

An example with `dev-v5.1`:
```c
    /*Create an image object*/
    lv_obj_t * gif_img = lv_img_create(lv_scr_act(), NULL);
    lv_obj_set_pos(gif_img, 50, 50);

    /* Create a task which will change the images on 'gif_img'
     * The last parameter will be passed to the function */
    lv_task_create(gif_task, 500, LV_TASK_PRIO_MID, gif_img); 

   ...

/* A task to periodically change images of 'img_obj' */
static void gif_task(void * img_obj)
{
    /*Declare the image variables*/
    LV_IMG_DECLARE(my_img1);
    LV_IMG_DECLARE(my_img2);
   
    /*Create an array from the images.*/
    static const lv_img_t * gif_frames[] = {&my_img1, &my_img2, NULL};
    static uint8_t i = 0;
    
    /*Set the current element of the array*/
    lv_img_set_src(img_obj, gif_frames[i]);

    /*Go to the next image*/
    i++;
    if(gif_frames[i] == NULL) i = 0;
}

```

kisvegabor, 2018-03-09:
I close this issue because a simple workaround is presented.

The "gif-like" animations might be useful for others too. In case of more interest is show for this object type it will be implemented in an upcoming release!

closed 2018-03-09 by kisvegabor

arpiatek, 2019-01-24:
GIF support would be very useful. Please comment on future support.

kisvegabor, 2019-01-25:
With this extension of LittlevGL external image libraries can be attached to support various image formats: https://blog.littlevgl.com/2018-10-05/png_converter

I found this simple GIF library: https://github.com/lecram/gifdec But it should be updated to be able to read data from a buffer (array) instead of a file.

Would you like play with it?

arpiatek, 2019-01-25:
Thank you very much! We will look into the extension.