lvgl/lvgl #130: Drop down list

Masi2018, 2018-02-26:
Hi, previously I had asked this question:
" I need to know what should be done if the LCD's height is not long enough to display all the options in the drop down list ? Now once I scroll among the options I can not see those are in the bottom of my list ( since the lcd height is not long enough but I expected that once I go down the options will move up in the way that user can see the last item which cursor point to)"
I have been responded there is a bug there for keyboard which has been fixed in beta branch, I want to know if I should download all the files in beta branch and replace in lvgl folder? I did this but problem still exists and focus didn't move along with the highlighted entry, I would appreciate if someone help me with this.  I should mentioned that I have already set a fix height for a drop down list using lv_ddlist_set_fix_height(ddlist, height)

sirius65535, 2018-02-27:
I have another question,i found it can not call a function when a new option is chosen in a roller,but the ddlist has the API,how can I call a function in roller?

kisvegabor, 2018-02-27:
Hi,
@Masi2018  v5.1 will be released soon. All the new stuffs and fixes will go there. (Read #125 to learn more about v5.1). So please update to `dev-5.1`. Just type in a terminal: `git checkout dev-5.1` or download fro GitHub and replace your lvgl folder with the new version. 
Also please copy-paste your code about ddlist.

About the action: with `lv_ddlist_set_action(ddlist, my_action)` you can specify a callback to call when a new item is chosen. Now the ENTER key opens/closes the list and the **action** will be called on every UP/DOWN or LEDT/RIGHT keys. 

However as I typed it is not so logical :) Call the action on ENTER and forget ddlist open/close functionality of ENTER seems more intuitive. What do you think?

@sirius65535 you can use `lv_roller_set_action` but is works similarly to ddlist. When a new option selefted with UP/DOWN or LEFT/RIGHT the action function will be called.

sirius65535, 2018-02-27:
@kisvegabor Thanks! I found `lv_roller_set_action` in 5.1, I did not find it in 5.0, I'll transfer my work to 5.1 when release.

Masi2018, 2018-02-28:
Hi
@kisvegabor , Thanks for all your time and efforts.
I wanted once user selects an item , automatically drop down list is closed and user is moved back to the previous menu that's why I needed an Enter key callback function .However, I made some changes in lv_ddlist_ext_t struct and added a new action for now. Also in lv_ddlist_signal function I called lv_ddlist_refr_size where  LV_GROUP_KEY_DOWN and LV_GROUP_KEY_UP is checked in the if...else statement. For now it works but I'll be waiting for  v5.1 . Please let me know if you think these changes may leads to problems.

kisvegabor, 2018-02-28:
Hi,

What about the following scenario?
- on focus: the ddlist opens
- on UP/DOW LEFT/RIGHT: move the selection
- on ENTER: call the action and close the list 
- on ENTER on a closed ddlist: open the ddlist
- on defocus: close the list

kisvegabor, 2018-03-01:
I pushed a some updates to `dev-5.1` related to drop down list and monochrome displays .
- implemented the behavior described in my previous post
- fixed some coloring issues with LV_COLOR_DEPTH = 1

Masi2018, 2018-03-06:
Hi Kisvegbor,
Sorry I was off from work in the past week. 
Sure I'll test that.
I was wondering if up and down key is not working in Keyboard to move to the next and previous row? because it's hard to move one by one and find the key in typing.

Masi2018, 2018-03-06:
@kisvegabor Sorry I didn't get you scenario about ddlist action. Actually the problem is now ddlist action is called automatically once I move the selection and what I need is call the action on Enter so I didn't get that what do you mean? you mean I check the key in action function and if it's Enter then close the list and ...?

kisvegabor, 2018-03-07:
Hi Masi2018,
Have you updated to the newest `dev-5.1` branch? The ddlist behavior have been already updated. 
May the IDE cache tricks you. Cleaning the project also worth a try.

There where updates also for LV_COLOR_DEPTH = 1. But it won't look well with the default styles. Try to use the mono theme.

kisvegabor, 2018-03-07:
UP/DOWN navigation on keyboard was really a missing "feature".
I added it to `dev-5.1`

Masi2018, 2018-03-07:
@kisvegabor Thanks very much Kisvegabor ,
I tested the keyboard and ddlist, they were fine but seems that you removed lv_image_create_file, since I am using this to draw images, what should be done now? 
About mono theme, how can I use that? I have set USE_LV_THEME_MENU to 1 , but it seems that I should do something else , would you tell me about it?
I was wondering if you have added long pressed action on buttons for hardware keyboard? Do you want me to test this as well?

kisvegabor, 2018-03-07:
You can use `lv_img_set_src` and generate the image C files with the online image converter tool: https://littlevgl.com/image-to-c-array 

About themes more info here: https://littlevgl.com/basics#themes
"You can force the library to apply the styles from a theme when you create new objects. To do this use lv_theme_set_current(th);"

For long press test you need to set the long press action of the button. You can check the [keyboard driver](https://github.com/littlevgl/lv_drivers/blob/master/indev/keyboard.c) as a reference. It's for PC keyboard using SDL but you should do something similar.

kisvegabor, 2018-03-07:
Closing means is it worked? :)

Masi2018, 2018-03-07:
No I closed by mistake!!!

Masi2018, 2018-03-07:
@kisvegabor I was testing and it was working but I don't know why now I am getting this error message:
/lvgl/lv_misc/lv_task.c:164:12: error: 'new_lv_task' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     return new_lv_task;

kisvegabor, 2018-03-08:
Hi,

It should be only a warning but I fixed it in the final v5.1 release. Now you can use the master branch. Please, update the lv_examples repo too ad test again.

Masi2018, 2018-03-08:
@kisvegabor Actually the compiler I am using takes this an error message and doesn't allow me t go ahead :(

kisvegabor, 2018-03-09:
Maybe just the IDE cache was tricked you.
Thank you for your help in making ddlist better! :)

Masi2017, 2018-03-09:
@kisvegabor Thank you for this amazing library you have created.

closed 2018-03-09 by kisvegabor