lvgl/lvgl #131: Can version 5.1 support UTF16？

fiend520, 2018-02-27:
Can version 5.1 support UTF16? My settings used in Chinese, using UTF8 to store too much space

kisvegabor, 2018-02-27:
Hi,
It is not planned in v5.1.

Using UTF8 for Chinese characters means 1 or 2 extra byte for every character compared to UTF16.
Do you have really so much text that this extra size matters?

fiend520, 2018-02-28:
My design requires multilingual support, at least 10, and needs full support from the UTF16 font library

kisvegabor, 2018-02-28:
I see. Supporting UTF16 and other decoding   is a possible future in v5.2. Now the UTF-8 decoding is not implemented modularly, so it's interface need to be reworked too.

I added this feature to the v5.2 todo list: https://github.com/littlevgl/lvgl/blob/master/docs/TODO_MINOR.md

kisvegabor, 2018-07-25:
I added the character decoder/encoder interface in `dev-5.2` branch. 

With this one can easily change the character decoder functions  because the  functions are pointed with function pointers:   https://github.com/littlevgl/lvgl/blob/dev-5.2/lv_misc/lv_txt.c#L53

It also makes easy to add built-in support for other character encodings because the interface is modular now.

If you or somebody is working on an UTF-16 support for LittelvGL please open a new issue for it.

closed 2018-07-25 by kisvegabor