lvgl/lvgl #133: [Brainstrom]Maybe a firmware can be developed for MCUs

WangZhihua1998, 2018-02-28:
Just like NodeMCU(It's a firmware for a cheap chip called esp8266), a lua based system can be built to use  scripts to control the screen. It's just my idea. I donnot know whether it is possible to do this on small MCUs and whether it deservers to do so.:grin:

kisvegabor, 2018-02-28:
Hi,

Actually I'm working on a Lua interface right now. The basic concept is already working, ust ned time to wrap every C function to Lua. It will be in v5.2. 
I also tested LittlevGL with EP32 and it was surprisingly fast on a 320x240 TFT (ILI9341)

So in my opinion your idea is great! :)

WangZhihua1998, 2018-02-28:
And I think you can just modify the NodeMCU proj and add the graphic lib as modules. By doing this it will be much easier.

kisvegabor, 2018-02-28:
I don't know NodeMCU in details. I will look at it!
Thanks!

closed 2018-03-05 by WangZhihua1998

vlisivka, 2020-05-25:
@kisvegabor Can we look at your Lua bindings for lvgl?

kisvegabor, 2020-05-25:
Finally I used SWIG to generate a binging. 

[example.i.zip](https://github.com/lvgl/lvgl/files/4676815/example.i.zip)

vlisivka, 2020-05-28:
My attempt on Lua bindings is here: https://github.com/vlisivka/lvgl_dash . I'm able to run simple program in Lua: 

```Lua
#!/usr/bin/sblua

lv = require "lvgl";
lv.init();

g = lv.lv_group_create();

lv.init_keyboard(g);

-- Create controls
tv = lv.lv_tabview_create(lv.lv_scr_act(), NULL);
tab_tests = lv.lv_tabview_add_tab(tv, "Tests");
tab_logs = lv.lv_tabview_add_tab(tv, "Logs");

lv.lv_group_add_obj(g, tv);

lv.lv_page_set_scrl_layout(tab_tests, lv.to_lv_layout_t(lv.LV_LAYOUT_COLUMN_MID));

btn = lv.lv_btn_create(tab_tests, NULL);
label =  lv.lv_label_create(btn, NULL);
lv.lv_label_set_text(label, "TODO");
lv.lv_group_add_obj(g, btn);
lv.lv_group_focus_obj(btn);

print "Entering event loop. Press ^C to stop program.";
lv.event_loop();
```
It produces nice looking interface: [image]

But, it cannot do anything useful because SWIG doesn't generate callbacks for me. Did you have solution or direction for me for automatic callback generation using SWIG?

kisvegabor, 2020-05-29:
It looks great! Have you done this manually or do you have a script?

Which callbacks do you mean exactly?

kisvegabor, 2020-06-01:
Awesome! If you can confirm it works well, we can create a repo for the Lua binding.

vlisivka, 2020-06-01:
Lua bindings are working OK for my needs, but they need more polishing: size of lua module can be reduced by banning unnecessary internal functions, more functions to support callbacks should be added. Also, I implemented my own input method for my device keyboard with Up, Down, Enter, and Esc keys only.

I did simple script to enable/disable logs on my device in Lua. See  `logs.lua` in repo. I found many problems in LVGL code:
  * page is not scrolling when driven by keyboard;
  * dropdown is not working properly in edit mode with my keyboard , because it closes popup immediately, because it receives RELEASE signal.
  * list is not working.
  * slider is not working and displayed improperly (too wide for 240x320 screen, maybe because of stlye).
  * menu displayed improperly (too wide, maybe because of style).

I also found no way to change orientation of screen by 90, 180, or 270 degrees.

kisvegabor, 2020-06-03:
Nice!

>  found many problems in LVGL code:

Do you use the latest version of lvgl (v7)?

vlisivka, 2020-06-03:
I use version LVGL v7 from week ago. I will update and check again.