lvgl/lvgl #135: malloc failes when creating sys_layer

curly-brace, 2018-03-01:
malloc returns null pointer on line 94 of lv_obj.c while executing lv_init()
`sys_layer = lv_obj_create(NULL, NULL);`
i'm not sure. but i believe 20k should be enough to display label without using buffers?
i've tried tinkering with linker script heap and stack sizes (making them smaller and bigger) and with LV_MEM_SIZE setting (from 32 to 512) but all this does not help, which is strange.
please point me to right solution i really want to use this lib it is very cool!

kisvegabor, 2018-03-01:
Hi,

It's because LV_MEM_SIZE is too small. The absolute minimum is 1k bu I recommend to use at least 2k.

And As I see you've successfully turned off the buffering with LV_VDB_SIZE  0.

I'm was also planning to port LittlevGL to STM32f103C8 but still haven't time to do it.
Please keep us informed about your process :)

curly-brace, 2018-03-01:
yay! thanks! it was the LV_MEM_SIZE!
but i still can not see the label.
i've implemented fill and map functions and they are working fine, but never called. i'm using the code from "hello world" example, so there are only 3 lines of code that create a label.
there is also SysTick interrupt calling lv_tick_inc(1);
but nothing is rendered and draw functions aren't called at all.

kisvegabor, 2018-03-01:
Please copy-past your code where you add he display driver (initialize the `lv_disp_drv_t` structure)

curly-brace, 2018-03-01:
```
lv_disp_drv_t disp_drv;
lv_disp_drv_init(&disp_drv);

disp_drv.disp_fill = ex_disp_fill;
disp_drv.disp_map = ex_disp_map;

lv_disp_drv_register(&disp_drv);
```

curly-brace, 2018-03-01:
actually i tried tracing lv_tick_inc(1); in debugger and i see that it only increments counter variable and nothing besides this. where are actual rendering functions are called? may be i miss some other function call?

curly-brace, 2018-03-01:
is it supposed to run on bare hardware or does it actually need rtos?

kisvegabor, 2018-03-01:
No OS required.
Do you call `lv_task_handler()` periodically in a while(1)?

kisvegabor, 2018-03-01:
As I see it is not shown in the porting guide. I will update that!

curly-brace, 2018-03-01:
yeah! thank you for great support! now everything is running fine!

kisvegabor, 2018-03-01:
Awesome :)

curly-brace, 2018-03-01:
i confirm that your great lib works on stm32f103c8!

closed 2018-03-01 by curly-brace

kisvegabor, 2018-03-01:
Great!
If you sent your project I would publish it `lv_boards` repo.

curly-brace, 2018-03-01:
yeah, i totally able to send it, but lcd is a separate module (hx8352b) and there is no code concerning stm32f103 intself, i just using SPL library. not sure if it will be helpfull. But hx8352b driver may be will help someone, but i haven't implemented touchscreen driver yet, only graphics is working for now.

kisvegabor, 2018-03-01:
I don't know SPL library and haven't found it. Can send a link to it?

curly-brace, 2018-03-01:
it is Standard Peripheral Library from st itself. it is already considered a legacy and st says that everyone should use HAL, but i don't like it (tho i'm a newbie, lol).
i think it could be downloaded here.
http://www.st.com/en/embedded-software/stm32-standard-peripheral-libraries.html?querycriteria=productId=LN1939
it is also included in most IDEs for stm32 (i'm using ac6 system workbench).

kisvegabor, 2018-03-01:
Ah, I see, thank you.
I thinks it's not problem if you used the SPL. It shows an other option to port LittlevGL :)

gnlcosta, 2018-08-02:
Hi Curly,
can you add/send the example of lvgl for stm32f103c8?
Thanks