lvgl/lvgl #137: Don't have listview  objects

fiend520, 2018-03-02:
like this 
[image]

kisvegabor, 2018-03-02:
Hi,

there isn't a list view object directly like this. I thought an object like this wouldn't be "handy" on a small TFT display. However I see many people use LittlevGL on larger displays with powerful MCUs.

Can you tell what is your usecase?  Where you would like to apply a listview object?

fiend520, 2018-03-05:
My English is not very good, using the translation tool, I do not know if you can understand. I'm using TI's MCU AM3352, not using OS
[image]

kisvegabor, 2018-03-05:
I see. I agree on this object could be useful with larger displays.

I add this object type to the v5.2 [TODO list](https://github.com/littlevgl/lvgl/blob/master/docs/TODO_MINOR.md)

kisvegabor, 2018-10-07:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-10-07 by kisvegabor