lvgl/lvgl #138: lv_indev_get_point strange behaviour

curly-brace, 2018-03-03:
i'm trying to implement calibration screen, but i'm having strange issue.
my touch driver seems to work fine, if i hardcode some offset and scale values inside it.
but when i try to read lv_indev_get_point() function, it just returns [605,0] everytime i execute it.
i'm using the same method, that is used in calibration app - big button and state machine.
i've tried going with my own idea first, without button and using lv_indev_wait_release(), but somehow it was not working at all (may be it is also a bug, i'm not sure, but it was returning without a delay at all).

curly-brace, 2018-03-03:
i've even done this in SysTick handler that fires every ms:
```
	char buf[16];
	lv_indev_t *indev = lv_indev_get_act();
	lv_indev_get_point(indev, &p1);
	sprintf(buf, "%u : %u" , (uint16_t)p1.x, (uint16_t)p1.y );
	lv_label_set_text(label, buf);
```
and it is really returns the same value.
but at the same time i have enabled LV_INDEV_POINT_MARKER and it plots points right.
am i using this method in the wrong way, or its data updates only once?

curly-brace, 2018-03-03:
ok. i know the reason. lv_indev_get_act() returns null pointer. what is the right way to get indev then?

curly-brace, 2018-03-03:
i've got it working in SysTick using lv_indev_next(NULL) function call.
and now i think that somehow before button callback gets executed, the actual coords are already overwritten by another indev read in the backgound and it just returns coords for unpressed screen.
but why?

curly-brace, 2018-03-03:
ok. i fixed this by editing line 260 of lv_indev.c.
it was
`if(i->driver.type == LV_INDEV_TYPE_POINTER)`
and i did
`if(i->driver.type == LV_INDEV_TYPE_POINTER && i->proc.state == LV_INDEV_STATE_PR)`
i'm not sure if it is a poper fix, but i think for touchpad you need to check whether it is pressed, before updating coords. but i'm not very good in LVGL architecture and i'm also a newbie in C too, so this fix can be a wrong way.

curly-brace, 2018-03-03:
ah. not fixed. button now just crashes the MCU. and here i don't know how to debug further.

curly-brace, 2018-03-03:
fixed it in lv_indev.c
```
            if(i->driver.type == LV_INDEV_TYPE_POINTER)
            {// for touchpanel it is better update coords only while it is pressed
                if (i->proc.state == LV_INDEV_STATE_PR) {
					i->proc.act_point.x = data.point.x;
					i->proc.act_point.y = data.point.y;
                }

                /*Process the current point*/
                indev_proc_point(&i->proc);
            }
```
first i've just added a second condition with &&, but it prevented ` indev_proc_point(&i->proc);` to be executed, thus preventing button from calling the callback.
i don't know whether it is a right fix, because i'm newbie in C and a don't know LVGL architecture well.

kisvegabor, 2018-03-04:
Hi,

Thank you for the detailed tracking. 

Here is how *indev* processing works:
1. Get the first input device and set  it as active
2. Read its point (or key)
3. Process the new point
4. Go the next input device (point 1) until have others
5. Set the active indev to NULL if there is no more indev.

In other words you can only see the active indev in action functions (e.g. button click) called during "Process the new point" phase to determine which indev triggered the action. Outside of action functions, (in general somewhere in the code) there is no active (being processed) input device.

Your technique with `lv_indev_get_next(NULL)` is a good idea, it gives you the lastly added indev. However I suggest rather save the return value of `lv_indev_drv_register()`. It's an `lv_indev_t *` variable.

It's dangerous to set a label's text in the sys tick handler because the systick can interrupt `lv_task_handler` in a wrong time. LittlevGL is designed to do all the things in init functions before `lv_task_handler` or action callbacks, lv_tasks etc. It is better to read a value (In your case the touchpad coordinates) in the systick interrupt and create and `lv_task` (which is a periodically function) to refresh a label's text there.  Or read the last coordinates in the `lv_task` too.

If I'm correct then your issue was that you read the touchpad when it was released hence you read dummy values. To solve this you should set  the last x and y coordinates in you `indev_read` function regardless to the state (pressed/released) of touchpad. 
Your fix seems good for touchpad, but it's not good if the POINTER is a mouse. The mouse uses the released coordinates too to position the cursor.

curly-brace, 2018-03-04:
thank you for your great explanation!

closed 2018-03-04 by curly-brace