lvgl/lvgl #139: stm32f746gdiscovery support

JefkeB, 2018-03-03:
hi,

first of all compliments on this very nice library.
been playing with it in the pc simulator
it looks very usable for giving some home automation hobby projects a nice gui

is there any support planned for the stm32f7 discovery board?

if not i'll try to implement it by myself 
following the porting guidelines
and probably with some help from your side 😉

jef

curly-brace, 2018-03-03:
it is very easy btw. you just need to implement a bunch of functions. another side of medal here is that you will need to IMPLEMENT those functions. but this is already not concerned LVGL itself.
i mean all that stuff to drive the lcd through DMA2D, using onboard SRAM and reading the touch controller.

kisvegabor, 2018-03-04:
Hi,

As curly-brace mentioned you need to implement a few functions yourself.

The good news are I already tried STM32F7 Discovery I'm planning to share the source code in `lv_boards` repo. I hope it will happen in a few days.

JefkeB, 2018-03-04:
that would be nice and very helpfull for getting lvgl running on my hardware

thanks

closed 2018-03-04 by JefkeB