lvgl/lvgl #141: A small disp bug

WangZhihua1998, 2018-03-04:
I found that the text in the page container which has round corners may display out of the page. The text is limited in a rectangle box, so the words around the corner may overflow. It is a small disp bug which  may not affect the disp effect. And the bug is hard to notice. But I still want to help you to better the details.

kisvegabor, 2018-03-04:
Hi,
thanks for the report. To solve this the library should apply masking on the corners which would require a lot of computation and extra memory. 

If you would like to apply only vertical scroll then as a workaround I suggest to set `style.body.padding.hor` >= `style.body.radius` on page background (`lv_page_set_style(page, LV_PAGE_STYLE_BG, &page_bg_style)`). This way you will get some margins:

[image]

```c
    static lv_style_t page_bg_style;
    lv_style_copy(&page_bg_style, &lv_style_pretty_color);
    page_bg_style.body.radius = 5;
    page_bg_style.body.padding.hor = 8;

    lv_obj_t* page = lv_page_create(lv_scr_act(), NULL);
    lv_obj_set_size(page, 100, 120);
    lv_obj_set_pos(page, 10, 10);
    lv_page_set_style(page, LV_PAGE_STYLE_BG, &page_bg_style);
    lv_page_set_sb_mode(page, LV_SB_MODE_DRAG);

    lv_obj_t * label = lv_label_create(page, NULL);
    lv_label_set_long_mode(label, LV_LABEL_LONG_BREAK);                         /*Wrap text*/
    lv_obj_set_width(label, lv_page_get_scrl_width(page));                      /*Text width to the page's scrollable width*/
    lv_label_set_text(label, "aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa");
```

classic-gentleman, 2018-09-10:
One option that would facilitate a lot in creating "fake" clippings on scrollable content is to do the same you did to borders: create `lv_radius_part_t`. However to keep backwards compatibility I'd have the empty case mean the same as all corners instead of creating `LV_RADIUS_FULL` for example. Or alias them both to the same operation. What do you think?

classic-gentleman, 2018-09-10:
Just to clarify what I mean, we have a couple options:

For `lv_list_set_style`, when `LV_LIST_STYLE_BTN_REL` or `LV_LIST_STYLE_BTN_PR` contains a nonzero radius, apply them partially as `LV_RADIUS_TOP_LEFT|LV_RADIUS_TOP_RIGHT` on the first list element, and `LV_RADIUS_BOTTOM_LEFT|LV_RADIUS_BOTTOM_RIGHT` on the last.

This is probably too magical for your taste, so the second option would be to use the returned values of `lv_list_add` for the first and last items and configure their partial radii after the fact, probably by cloning their styles with `lv_btn_get_style` and reapplying the modified ones.

Seems to me that both approaches would require a couple extra styles (and their respective static allocations) so I'm wondering if you have a better idea. One way or another, short of a full-blown clipping implementation, partial radii is probably the key to making this happen.

classic-gentleman, 2018-09-10:
Another option to fake clipping is to use the same algorithm you use for borders but unconditionally extend them to the bounding box of the object instead of drawing a line of `body.border.width`.

Yet another option (that would not take a ton of memory to implement and is what iOS8+ does) is real but specialised clipping: to create a square buffer of 2*radius by 2*radius and stash the frame buffer memory that would otherwise be overwritten by the contents of the clipping container, and after the container is drawn then restore the outer colours. Of course this could be reduced to 4r² - πr² (which is a square with a punched hole) pixels but I don't think at this stage it's worth doing this at all.

This of course only applies to rounded rects, but I believe it's super worth it for containers.

Finally you could extend the `lv_area_t` struct to take corner radii. It would be almost trivial to implement roundrect clipping there, and probably worth to rethink what's inlineable there. It's also not difficult to split the drawing area into cold-hot-cold parts to speed up the unaffected rectangular areas.

I'm throwing some ideas at the wall to see what sticks, because I'm interested in helping implementing any of those ideas if you find them worth doing. Just need a "hey this is cool" feedback of sorts. 😄

kisvegabor, 2018-09-10:
Hi,

Wow, thank you for these great ideas! 

1. Native support to round corner clipping 
Buffering the corners sounds the most simple and to implement. The user should define the size of  "CORNER_BUFFER" (?) and proper clipping will be executed if the area fit's into this buffer. 
We should have a bit in styles like `style.body.radious_clip` to enable or disable this feature.

2. In v6.0 I'm planning to add a `stencil` object. It will be a transparent object and will have a `stencil_test` function. While the library draws the children of the stencil it calls the `stencil_test` function for each pixel. Or gives it a pointer to the new content and the background to mix them as needed. It can be used for more complex maskings too but it will be slow. Maybe your suggested solution makes it unnecessary because its main purpose would be to support rounded corners.
  
And thank you for offering your help!
It seems to me the "Native support to round corner clipping" can be added in `lv_draw_rect`. Would you like to do it? :)

classic-gentleman, 2018-09-10:
On your bullet point 2: we're in total agreement. Rounded corners are the extremely common case and lowest hanging fruit to implement. Generalised masking should be done with generalised clipping, whether by rasterisation or by on the fly calculations with Bézier paths. If we implement corner radius rounding we're tackling 99% of the real life use cases for clipping, so I'd go heavily for it!

On bullet 1: I'm still not familiar enough with LittlevGL to just go ahead and do it, so I'd while I'm familiarising myself with it I'm looking at "pain points" and how I'd be able to help.

Fortunately you fixed my greatest pain second to none (byte swapping for 16-bit colours; thank God you use the colour macros everywhere!) and I might as well look for other byte formats for 24-bit (RGB, BGR, ARGB, BGRA...). The main issue there is not that it's difficult to do it, especially now that I have your bswapped 16-bit commit to follow; it is that it's *quite* impractical not to do it at rendering time because of DMA. And 24-bit with alpha as 32-bit is only usable for GPU blending. For all other purposes packed 24-bit colour is quite more useful since most cheap TFTs are 262k colours by discarding the bottom 2 bits per channel in the controller, but they transfer full 3 bytes per pixel.

The general solution of course is to have a configurable pixel format and produce colours under said format perhaps according to some rules, as the cost is paid only once per pixel at rendering time, a cost that you're always paying anyway. (I originally "solved" this by interleaving the byte swapping routines with the DMA transfers, using manually unrolled loops and a bunch of assumptions on the transfer size... Mainly assuming `my_disp_flush` always receives a full frame buffer as configured on `lv_conf.h` and that it was never read from, so I just did away with the const qualifier for the FB pointer and swapped in-place (because, well, I already gave lvgl a ton of DMA-able memory; see below). Thanks for allowing me to delete that bunch of hogwash 😂). I understand you were aiming to solving this with `disp_drv->vdb_wr` because of colour ePaper, but those are slow enough that it's never an issue. For 60fps responsiveness on the other hand (which I managed to achieve "easily", benchmark is down to 14.9ms and I'm not even finished tuning the config yet), this is absolutely impractical.

The other thing that created a ton of trouble is that I _think_ many sensitive code paths are not thread safe at all. I have `my_disp_flush` shuttling messages to the actual flusher thread that lives in a different core than the `lv_task_handler` and boy has it been fun to debug threading issues with experimental development tools... I'm making sure to call `lv_flush_ready` on the same core as `lv_task_handler` but I'm still experiencing corruption at times.

One thing I believe is another quite low hanging fruit is having a version of `LV_MEM_CUSTOM_ALLOC` for the frame buffer only. Some MCUs need specific caps on DMA-able heaps and one in particular benefits quite a lot of 32-bit-aligned accesses. As it stands I'm allocating all lvgl memory as DMA-able 💩

One more thing that you could leave on the back burner is to do frame buffer diffing like https://github.com/juj/fbcp-ili9341. I'm not extremely constrained by neither CPU nor bus speed (especially after enabling large DMA transfers) but I'm a bit memory constrained, so I'm not even looking into that at the moment. But many people have opposite constraints and have a lot of memory with mediocre bus and quite possibly multiple cores, and sacrificing a core for display performance might net a big gain for them.

A different take on this (that doesn't mess with memory constraints) to allow assuming a stateful TFT with onboard GRAM and track dirty areas; instead of flushing full FBs you could just flush the dirty parts.

The final low hanging fruit is logical rotation. Since you have horizontally-sliced frame buffers the worst case scenario for bandwidth is vertical scrolling, since that could potentially make them all dirty (which is one of the biggest reasons for most modern mobile GPUs being tile-oriented). Sufficiently clever but not too difficult optimisation could help with that. Not reinventing any wheels here, just putting them all together.

I'm telling you all this because between pixel formats, thread safety, diffing buffers or tracking dirty areas, rotation (which is already on the 6.0 roadmap), roundrect clipping and FB allocator I'm quite partial to taking the last two... For now. 😄

kisvegabor, 2018-09-12:
**24 bit RGB888**
I agree it would be useful to support "real" 24 bit colors. Due to some updates in the past, it seems a little bit easier than before. 

I can confirm that removing the `const` in `disp_flush` should work to "in place" convert the pixel format to the desired one. 

`disp_drv->vdb_wr` can be useful if you need to choose between memory and performance. And you can sacrifice performance. For high-end GUI's with 60 FPS it no acceptable indeed.

**Allocate DMA-able memory**
I just added an API to do it in `dev-5.2`. Have look at this: https://github.com/littlevgl/lvgl/issues/399#issuecomment-420536509

**Thread safe**
And what causes the issue? Calling `lv_flush_ready` from a different thread (and even a core) mess thing up?

**Frame buffer diffing**
LittelvGL refreshes only those parts which are changed. So if you press a button with 100x100 size only that area will be refreshed. So it keeps track with the dirty areas.

**Logical rotation**
I don't understand. E.g. if you scroll in a 50x50 px "window"  only this 50x50 px will be dirty and will be refreshed. So it doesn't matter how the frame buffer is sliced. Or have I missed something?

I think we should start with "clipping of rounded corners". If you feel that you are ready to work on it, please open a new issue for that. 

I will check how difficult it is to add native 24 bit support.

classic-gentleman, 2018-09-12:
Should we split this ticket into a small bunch so we keep track of the broader themes?

kisvegabor, 2018-09-12:
Yes, creating separate issues for each suggestion and idea would make easier to discuss them

closed 2018-09-12 by kisvegabor

classic-gentleman, 2018-09-12:
> **Thread safe**
> And what causes the issue? Calling `lv_flush_ready` from a different thread (and even a core) mess thing up?

I'm experiencing some weird stuff. Could be related to the recently introduced time keeping that doesn't depend on ticking (manifesting as a layering craze when I call the touch calibration app over the demo app). Could be my inexperience with the lib as well (manifesting as the tabs appearing on the bottom(!) depending on where/when I call the app creators with regards to the lv task). I'm really puzzled at this at the moment so I have more questions than answers right now...
 
> **Frame buffer diffing**
> LittelvGL refreshes only those parts which are changed. So if you press a button with 100x100 size only that area will be refreshed. So it keeps track with the dirty areas.

Maybe this is true for direct mode but per your own comments (https://github.com/littlevgl/lvgl/issues/7) VDB mode flushes the entirety of it if the screen slice corresponding to a given buffer gets dirty. This is consistent with my experiments and logs. The thing is that for antialiasing and other niceties we need VDB, but I don't think it's strictly necessary to flush the entirety of it if we track the parts that actually get drawn on. I'm kind of envisioning a scheme where you send a linked list of buffers to flush with DMA. I'm letting it sink for a moment.

> **Logical rotation**
> I don't understand. E.g. if you scroll in a 50x50 px "window" only this 50x50 px will be dirty and will be refreshed. So it doesn't matter how the frame buffer is sliced. Or have I missed something?

See above. Have I misunderstood it?

> I think we should start with "clipping of rounded corners". If you feel that you are ready to work on it, please open a new issue for that.

Might need a hand later down the road but I'm going for it. 😃

classic-gentleman, 2018-10-08:
Ping!

Sorry for not following up sooner, I'm getting a newer piece of hardware up and running and I'm of course facing the usual issues. Rest assured, you'll hear from me soon.

Congrats on releasing 5.2!