lvgl/lvgl #143: my stm32f103c8 + hx8352b tft support implementation

curly-brace, 2018-03-04:
here is the link to my test project. may be it will be useful.
https://www.dropbox.com/s/7hbxg3vrh6w9ym0/lvgl_test.tar.gz?dl=0

but it won't work out of the box, because by default stm32f103c8 has 64k ROM, which is not enough for LVGL. but it is possible to unlock second flash bank. i needed to edit openocd device script. and of course edit linker script. so LVGL fits fine in 129k ROM and works fine on 20k RAM.

P.S. code is a mess, i know, i'm not very experienced in C yet.

kisvegabor, 2018-03-04:
Thank you! I will check it soon and write!

kisvegabor, 2018-03-04:
I was very curious so I removed the `sscanf`  and done the following changes in your project:  
- updated to lvgl to  v5.1 (from `dev-5.1`) branch
- disabled the complex objects (e.g. Tab view)
- disabled unused modules in general
- switched to O3 optimization  (Project properties -> C/C++ build -> Settings -> Optimization)
- removed the mentioned sscanf
- enabled buffered drawing: LV_VDB_SIZE     (10 * LV_HOR_RES) 
- Implemented TFT_Flush (based on TFT_Map just added lv_flush_ready at the end)
- Registered TFT_Flush in the display driver (disp_flush = TFT_Flush)

Here is the modified project:  [lvgl_test_kisvegabor.zip](https://github.com/littlevgl/lvgl/files/1778667/lvgl_test_kisvegabor.zip)

It compiles for me and results 44kB code. 
Can you test it?

curly-brace, 2018-03-04:
wow! that's great! i was thinking that 'flush' routine was meant only for LCDs with gpu capabilities, like DMA2D in higher stm32 MCUs. will check your project as soon as possible!

curly-brace, 2018-03-05:
i've tested it. it draws the gui fine and it even seems to draw it faster, but i failed to enable touch. and during debug, i see no code. seems like you've disabled symbols for size optimisation, but i don't know how to switch them back, lol.

kisvegabor, 2018-03-05:
Great!

I modified your `TFT_ReadTouch` function to set the last coordinates even in released mode. Hence the lv_indev modification is not required.
```c
bool TFT_ReadTouch(lv_indev_data_t *data) {
	static int16_t last_x = 0;
	static int16_t last_y = 0;

	data->state = readZ() > TOUCH_Z_MIN ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;

	if (data->state == LV_INDEV_STATE_PR) {
		if (xMin == 0) {
			last_x = readX();
			last_y = readY();
		} else {
			last_x = mapX(readX());
			last_y = mapY(readY());
		}
	}

	/*Always set valid points*/
	data->point.x = last_x;
	data->point.y = last_y;

       ...
}
```

curly-brace, 2018-03-05:
aha! now i understood what you've meant bu setting coords everytime. i made it that they were updated every time from ADC, but not updated conditionally using static variables. it really works fine that way.

kisvegabor, 2018-03-05:
Great! :)
Please. keep us informed if you made the optimization working with touchpad.  
I'm very curious about the final code size with -O3.

curly-brace, 2018-03-05:
that is very interesting things happening here. i've added again the debug label, because debug dots were not drawing. and i'm noticing jumping x or y coords while touchpanel is not touched at all. and while i make coord vars volatile or move them to header file, they alter behaviour, in some case jumps x and y is right, in other case jumps y and x is right. rofl. c optimisations are hell for me. but i will try hard to make my code work.

curly-brace, 2018-03-05:
oh! i've fixed it not only by getting those static variables out of TFT_ReadTouch, but also leaving them uninitialised and resetting them to zero in TFT_Init.
here is the result of your variant:
```
Generating binary and Printing size information:
arm-none-eabi-objcopy -O binary "lvgl_test.elf" "lvgl_test.bin"
arm-none-eabi-size "lvgl_test.elf"
   text	   data	    bss	    dec	    hex	filename
  49608	   1184	  10308	  61100	   eeac	lvgl_test.elf
```

curly-brace, 2018-03-05:
but yeah, buffering looks much better! while i was trying to solve optimisation issue, i was using my project with no buffering, and now i run your variant with buffering and it just feels better!

kisvegabor, 2018-03-05:
Yes, optimization can make strange things. The `voaltile` prefix usually helps.
So the code is < 50kB which leavs you 16 kB space for the application. Which is not too much but might be enough for a simple GUI and the related measure/control .

kisvegabor, 2018-03-05:
And thank you for telling you experiences :)

curly-brace, 2018-03-05:
while this 'blue pill" stm32f103c8 board actually has 128k ROM (but need unlocking), this makes your library suitable even for those "1 dollar" boards! and even without config hacking it works with default 64k! and even with buffering running on just 20k RAM! that's is very very great!

closed 2018-03-05 by curly-brace