lvgl/lvgl #144: Button object addition idea

ramonschepers, 2018-03-05:
Would be awesome to have a button with background image (or animation like a battery graph)

It's a much wanted feature for me, as I am planning to use these kind of icons for a GUI.

I think it's best to create a separate object for this, like lv_obj_btn_anim.

What do you think about this?

kisvegabor, 2018-03-05:
Hi,

**lv_icon** is a planned object which will acts as button but draws an image

Until lv_icon arrives you can create an image on a button:
```c
lv_obj_t * btn = lv_btn_create(lv_scr_act(), NULL);
lv_btn_set_style(btn, LV_BTN_STYLE_REL, &lv_style_transp_tight);
lv_btn_set_style(btn, LV_BTN_STYLE_PR, &lv_style_transp_tight);
lv_btn_set_action(btn, LV_BTN_ACTION_CLICK, my_action_func);

lv_obj_t * img = lv_img_create(btn, NULL);
lv_img_set_click(img, false);
lv_img_set_src(img, &some_image);
```

closed 2018-03-09 by kisvegabor

PetrenkoIgor, 2020-03-10:
> Hi,
>
> **lv_icon** is a planned object which will acts as button but draws an image
>
> Until lv_icon arrives you can create an image on a button:
>
> ```c
> lv_obj_t * btn = lv_btn_create(lv_scr_act(), [...]

lv_img_set_click(img, false);

embeddedt, 2020-03-10:
You should use [`lv_obj_set_click`](https://docs.littlevgl.com/en/html/object-types/obj.html#_CPPv416lv_obj_set_clickP8lv_obj_tb), which works on all object types.