lvgl/lvgl #145: freertos examples?

curly-brace, 2018-03-05:
are there any freertos (or other rtos) examples?

kisvegabor, 2018-03-05:
I have test, but have no time to publish them.
I tried LittlevGL on the following boards and RTOS:
- ESP32 which running FreeRTOS inside
- STM32 Discovery with NuttX (LittlevGL already integrated to Nuttx)

curly-brace, 2018-03-05:
wait, nuttx's graphic system uses LVGL?
actually i was tinkering with nuttx, but it appeared too complex for me, so i decided to go with freertos and some standalone gui library and found your cool project.

kisvegabor, 2018-03-05:
I had a cooperation with NuttX and added LVGL to it. It is an optional application you can enable. The touchpad calibration and the demo application is available in Nuttx.

I agree, NuttX is more complex then FreeRTOS + LittlevGL but have a lot of advantages. Like the extremely good driver and board support + POSIX compatible environment.

Masi2017, 2018-04-02:
@kisvegabor I was wondering if I may have the ESP32 example?

kisvegabor, 2018-04-03:
I will update the ESP32 example to lv_boards.

kisvegabor, 2018-04-04:
@Masi2017  I uploaded the ESP32 project using ILI9341: https://github.com/littlevgl/lv_boards

Please read the README because you need to copy two files manually.

brightproject, 2025-07-20:
> I uploaded the ESP32 project using ILI9341: https://github.com/littlevgl/lv_boards

Please tell me where I can now look at examples of `freeRTOS` for `ESP32` and `LVGL`?🙄

uLipe, 2025-07-20:
@brightproject since official Espressif framework, the IDF is built on top of freertos you can simply use the LVGL port for esp32  board reference project:

https://github.com/lvgl/lv_port_esp32

brightproject, 2025-07-21:
> since official Espressif framework, the IDF is built on top of freertos you can simply use the LVGL port for esp32 board.

Great link, thank you😀
Unfortunately I don't work with `ESP-IDF`, my tools are:

> framework-arduinoespressif32 @ 3.2.1

However, I tried the example code and was disappointed with it.

```c++
// This file was modified by e.vlad.engineer
// LVGL version: 9.3
// Arduino_GFX version: 1.6

#include "Arduino.h"
#include <lvgl.h>
#include <Arduino_GFX_Library.h>

// #define NO_RTOS
// #define ANIM

#define SERIAL_BAUD 115200

// https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration#esp32s3-21-tp
Arduino_DataBus *bus = new Arduino_ESP32QSPI(
    10 /* CS */, 9 /* SCK */, 11 /* D0 - mosi */, 12 /* D1 - miso */, 13 /* D2 - quadwp */, 14 /* D3 - quadhd */);
Arduino_GFX *gfx = new Arduino_ST77916(bus, 47 /* RST */, 0 /* rotation */, true /* IPS */, 360 /* width */, 360 /* height */);

#define LV_TICK_PERIOD_MS 5

// Backlight PIN
#define GFX_BL 15

uint32_t screenWidth;
uint32_t screenHeight;
uint32_t drawBufSize;
lv_display_t *disp;

lv_obj_t * background;
lv_obj_t * horizon_line;

static int32_t roll_deg = 0;

/* Creates a semaphore to handle concurrent call to lvgl stuff
 * If you wish to call *any* lvgl function from other threads/tasks
 * you should lock on the very same semaphore! */
SemaphoreHandle_t xGuiSemaphore;

#ifdef NO_RTOS
    uint32_t millis_cb(void)
    {
    return millis();
    }
#endif

void my_disp_flush(lv_display_t *disp, const lv_area_t *area, uint8_t *px_map) // LVGL 9.3
{
  uint32_t w = lv_area_get_width(area);
  uint32_t h = lv_area_get_height(area);

  // gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h); // LVGL 8.3

  // gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t *)px_map, w, h); // LVGL 9.3
  gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)px_map, w, h); // LVGL 9.3

  // lv_timer_set_period(disp->refr_timer, 10);

  /*Call it to tell LVGL you are ready*/
  lv_disp_flush_ready(disp);
}
[... 161 lines cut ...]
          lv_obj_set_style_transform_rotation(horizon_line, -roll_deg * 10, 0);
        #endif
        
        lv_task_handler();
        // vTaskDelay(5 / portTICK_PERIOD_MS);
        vTaskDelay(LV_TICK_PERIOD_MS / portTICK_PERIOD_MS);
    #endif
}
``` 

The code with the help of defile `NO_RTOS` can work in two modes.
Without `RTOS` without animation, I get `195 FPS`

[image]

With animation `15-25 FPS`.

https://github.com/user-attachments/assets/c0df3ab1-9724-440d-add2-3297bb597809

When I run the code with `RTOS`, without animation about `95 FPS`, and I don’t know yet how to implement animation on `RTOS`.

[image]

I would be grateful if you could tell me how to implement the task on R`TOS.`
Now the code works faster and better without `RTOS` than with it🙄