lvgl/lvgl #146: Drawing Polygon and Arc

bhavin03, 2018-03-05:
It would be nice, if drawing filled polygon and arc were possible. This would allow to create some custom widgets..!!

kisvegabor, 2018-03-05:
Hi,
thank you for raising the idea. True, better drawing support could add a lot of new opportunities to LittlevGL. 

Let's collect the missing and required drawing algorithms:
- polygon border
- filled polygon 
- arc
- pie

Have I missed something relevant?

bhavin03, 2018-03-05:
Hi,

As a reference I compared Littlevgl with wxWidgets basic shapes. Most useful could be :
- ...
- Ellipse
- EllipticArc
- Rounded Rectangles
- ..

Adding rotation parameter to all basic shape will give a new dimension to create custom shapes.

Then I think you are well equipped with basic shapes to create amazing Embedded GUI.

kisvegabor, 2018-03-05:
Then the the full list is:
- Polygon border
- Filled polygon
- Arc
- Pie
- Ellipse
- EllipticArc

Rounded rectangle is already supported.

The most challenging thing is to implement the drawing functions in a memory and CPU effective way at the same time:
- Not use extra layer for rendering
- Don't draw a pixel twice to support opacity
- Support anti-aliasing 

It would be motivating to collect some images about GUIs where these higher level drawing functions were used to make the graphics really engaging. 

As workaround my plan was to add image rotation feature to v5.2 release which can replace drawing complex shapes. It is much more simple to implement and need to be done only once (for image drawing) and also supports more sophisticated shapes. Of course at the cost of ROM usage.

bhavin03, 2018-03-10:
Hi @kisvegabor ,
 as I was trying to draw polygon using line object, I found that polygon object edges are trimmed out  when it was drawn with thicker lines. In the image below, I've drawn polygons with 5, 10 and 15px respectively.

[image]

Would you suggest me a good solution for it ?  

As a feature for lines/polygon, line-caps ( square/round/butt ) can be implemented.

kisvegabor, 2018-03-12:
Hi,

There are two things here:
1. If you draw on *x = 0* coordinate then the left part will be trimmed because it is out of the object. I have an idea to fix it. I will do it in the next bugfix release. Until that please avoid drawing on  < style.line.width coordinates.
2. There is no proper line ending now. Only horizontal and vertical line endings are supported. To draw polygons with thicker lines in my opinion rounded line endings should be used.

Once I played 3D rendering and implemented the triangle drawing. It is still in the library and can be enabled in `lv_draw.h` by `USE_LV_TRIANGLE`  

However it's not easy to "just draw something" (because LittlevGL manages the drawing) with the following code you can also play with triangle drawing:
```c

static bool tri_design(lv_obj_t * obj, const lv_area_t * mask, lv_design_mode_t mode)
{
    if(mode == LV_DESIGN_COVER_CHK) {
        return false;
    }
    else if(mode == LV_DESIGN_DRAW_MAIN) {
        lv_point_t p[] = {{100, 5},{20, 70},{150, 90}};
        lv_draw_triangle(p, mask, LV_COLOR_RED);
    }

    return true;
}

...

    lv_obj_t * tri = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_size(tri, 200, 200);
    lv_obj_set_design_func(tri, tri_design);

```

TheGrovesy, 2018-05-22:
I was going to ask about basic shapes but it looks like you are already planning for it. Do you have a rough date when you hope to complete this work?

kisvegabor, 2018-05-22:
We started work on two improvements for shapes:
- Perpendicular line ending (however the line is not really a shape): #252
- Arc drawing: #255 

As you can see there is some process in line drawing but we are still thinking about the proper arc drawing algorithm. Any ideas are welcome in these topic. Please use threads above for Line ad Arc related thoughts.

kisvegabor, 2018-07-14:
Arc drawing and proper line drawing is implemented in `dev-v5.2`.

Polygon and ellipse drawing is not implemented yet and they are not planned in the near future. If you feel important or you want to implement it/them, please open a dedicated issue for it. 

Thank you for your interest and contribution!

closed 2018-07-14 by kisvegabor