lvgl/lvgl #147: Add ability to change TabView Buttons Location

BenArtes, 2018-03-06:
For applications where the user's hands are located near the bottom of the screen, having the TabView buttons on the bottom of the screen would make it easier for the user to select a new tab without moving their hands.

I'm willing to implement this feature myself, but could use some guidance on how you'd best like it implemented, if at all.

I think the most useful options, and easiest to implement, are TOP and BOTTOM locations, but some users might also want LEFT and RIGHT.

Depending on how this is implemented, it would likely change the API and thus only be able to be changed in v6; correct?

kisvegabor, 2018-03-07:
Hi,
It's a good idea which is absolutely possible to implement. (it was already mentioned in #80 issue)

With a new function - e.g. `lv_tabview_set_btns_pos(tv1, LV_TABVIEW_BTNS_BOTTOM)` - the API can be extended.

I would like to release v5.1 today so this update can part of v5.2!

BenArtes, 2018-03-07:
I added TOP and BOTTOM positions in the btns_pos branch of my lvgl fork: https://github.com/BenArtes/lvgl/commit/2d9056aa78eb18a4029f13c32d5c88cdf30152e6.  Adding LEFT and RIGHT is going to have to wait until I have time to further understand animations and the changes that are required.

I can create a pull request for just TOP and BOTTOM if you'd like, or I can wait until I have more time to try to tackle LEFT and RIGHT (though it could be a while).

kisvegabor, 2018-03-08:
Awesome, thank you very much! It's a useful update.
As v5.1 is released now I will create a `dev-5.2` branch soon and we can merge your updates there.

kisvegabor, 2018-07-14:
Earlier I merged the TOP an BOTTOM position update to `dev-5.2`. I finally tested it and it worked well!
Thank you very much!

LEFT/RIGHT seems more complicated because now there is no way to specify the width of the button area. However, there are some great ideas to simplify the adjusting these sizes of "internal" objects. See #302

I think it will be better to deal with LEFT/RIGHT when these updates will be released.
I close this thread now but feel free to comment here or re-open if you have any remarks.

closed 2018-07-14 by kisvegabor